OpenKinect/libfreenect2 #439: enumerateDevices() must be called before doing anything

xionglingfeng, 2015-10-22:
`freenect2.enumerateDevices()` must be called before doing anything with libfreenect2, even `freenect2.getDefaultDeviceSerialNumber()`. Otherwise, the following functions just failed silently without any messages.

This behaviour is very confusing. Is it possible to
- get ride of this limitation, or
- show an meaningful error message, or
- document this behaviour?

xlz, 2015-10-22:
* document this behaviour

If you don't enumerate devices, how do you know the devices exist? getDeviceSerialNumber() can use a range check though.

xionglingfeng, 2015-10-22:
@xlz When only one Kinect 2 connected, just go with `getDefaultDeviceSerialNumber()` after created `Freenect2` object is expected to work I think...

xlz, 2015-10-22:
You want to put enumerateDevices() inside getDefaultDeviceSerialNumber() so you can save one line of code? I'm not sure. Create a PR and see how it goes.

xionglingfeng, 2015-10-25:
@xlz Sorry for the confusion. I am not mean to save one or two lines of code.
By reading the code (libfreenect2.cpp), it seems that `Freenect2::openDevice()` depends on `getNumDevices()` and `enumerated_devices_`. These two variables are filled by `Freenect2::enumerateDevices()`. Thus, it introduce a sequential requirement that `Freenect2::enumerateDevices()` must be called before `Freenect2::openDevice()`.

In most cases, only one `Kinect 2` is connected to the computer. My instinct is to invoke `openDefaultDevice()` immediately after `Freenect2 freenect2` because there are only one device available, there are no need to enumerate devices and choose which one to open.

An more confusing behaviour here is that when I invode `openDefaultDevice()` without `enumerateDevices()`, the function will simplely failed in silence, without any error messages.

Based on the current design, I think it is better to introduce an comment line in `example/Protonect.cpp` to remind the user invoking `enumerateDevices()` immediately after creating `Freenect2` object.

xlz, 2015-10-25:
No, even if you have only one Kinect, the program can detect zero Kinect. That is why you must call enumerateDevices() before opening any devices.

It does not make any sense to open a device if you/the program don't even know if it exists.

xionglingfeng, 2015-10-25:
@xlz I agree with your opinion as I stated above and I also read the code and agree with current design. The main concern here is that calling other functions without `enumerateDevices()` make them silently failed without error messages. My suggestion is to put a comment line in the example project (`example/Protonect.cpp`) to remind users call `enumerateDevices()` before any further actions.

xlz, 2015-10-25:
* document this behaviour

Right.

xlz, 2015-10-25:
Add this to #287.

xlz, 2015-12-02:
I think it has been included in the documentation.

closed 2015-12-03 by floe