OpenKinect/libfreenect2 #442: data recording and playback + skeleton tracking

davudadiguezel, 2015-10-22:
Hello,
has someone ever tried to do skeleton tracking on recorded data (on linux) and can help me to get it running? I tried with openni2 and nite2, but experienced a bug in nite2 when reading in recorded (.oni) files, that seems impossible to workaround. 
Can anyone help?
Greetings 
Davud

birgersp, 2016-06-23:
I'm also very interested to know how to get skeleton tracking using the Kinect V2 sensor and libfreenect2. Openni2 is not for Kinect V2, apparently. And I'd really hate to have to ask my industrial partner to install Windows 8/10 to get skeletal tracking with Kinect V2...

BrendanMesters, 2025-03-20:
Is there any new info on this point?

Cause I kindof need to do skeletal tracking via kinect, and I hope to be able to do this with the open source driver