OpenKinect/libfreenect2 #443: resource.inc.h can't allocate an array with constant size of 0

Edwinzero, 2015-10-22:
[image]
Hey guys, I try to compile the newest source to update my libfreenect2 library. 
I didn't do any change and I use vs2010 vc100 x64. It didn't support c+11 so I change isnan() to _isnan() in registration.cpp file line 273
[image]
But I find it always generate an empty array in resource.inc.h during compiling.
[image]
One more thing is when I config cmake it says I didn't have a PKG_CONFIG_EXECUTABLE... Will this cause a problem?
[image]

Could you tell me is there anything wrong during my configuration?
@xlz  @floe @larshg

xlz, 2015-10-22:
`isnan` is C99. Old MSVC does not support C99. That has nothing to do with C++11. Maybe add a `#ifdef` here.

`PKG_CONFIG_EXECUTABLE not found` is harmless. Maybe I should use `find_packet(pkgconfig quiet)`. At least two people have asked about this.

I am assuming you disabled both OpenGL and OpenCL so their shader/kernel files are not included in resources, then it is expected that there is no resource to generate. Maybe we can make it easier for old MVSC by adding a dummy entry.

If you enable OpenGL or OpenCL and resource.inc.h is still empty, then there is something wrong more seriously.

Edwinzero, 2015-10-23:
I fix the isnan by using following codes in registration.cpp
[image]
Thank you very much @xlz :+1:  : )

xlz, 2015-10-23:
@Edwinzero 

My question is: did you disable OpenGL and OpenCL and then get the error of "resource.inc.h can't allocate an array with constant size of 0"?

Edwinzero, 2015-10-27:
Hi @xlz, I disable OpenGL and OpenCL then get the error. After I enable them, there is no more error

BaptisteTheoriz, 2016-02-11:
Hi,

I use libfreenect2 in an openFrameworks addon for a project on Windows 10 built with Visual Studio 2015, and I encounter a "can't allocate an array with constant size of 0" on line 48 of resource.cpp file.
I ca

xlz, 2016-02-11:
I don't know openFrameworks. How did you build it? Link?

BaptisteTheoriz, 2016-02-11:
The addon is the following one : [https://github.com/ofTheo/ofxKinectV2](https://github.com/ofTheo/ofxKinectV2)
Basically, it's just a wrapper which uses libfreenect2 src files and a modified version of protonect using openFrameworks classes. 

Perhaps I misspoke, so I will try to clarify : I can build with no problem libfreenect2 from your repository, the openKinect one, and use Protonect for example. Because the macro Resources_INC is defined so the line causing error isn't reached.
In my openFrameworks context, with libfreenect wrapped in the addon (src files of libfreenect hasn't been modified), this macro is not defined, so my compiler try and fail on this line : `static ResourceDescriptor resource_descriptors[] = {};`

In any case, even outside my specific context, if you remove the `#ifdef RESOURCES_INC`, the line 48 cause a "can't allocate an array with constant size of 0" error.

xlz, 2016-02-11:
It looks like they include their own copy of libfreenect2 so I can't really do anything here to help you.

BaptisteTheoriz, 2016-02-11:
Okay, forget about openFrameworks.
With your libfreenect, if you comment following lines : 

```
//#ifdef RESOURCES_INC
//#include "resources.inc.h"
//#else
static ResourceDescriptor resource_descriptors[] = {};
static int resource_descriptors_length = 0;
//#endif
```

it cause an error.
What's the meaning of this ifdef/else if the else case doesn't work ?

ofTheo, 2016-02-11:
@BaptisteTheoriz I believe a fix for this would be:

static ResourceDescriptor * resource_descriptors;
static int resource_descriptors_length = 0;

I believe Visual Studio doesn't like you trying to create an array with a size of 0. 
An array with a size of 0, could also be defined as pointer to a ResourceDescriptor. 
As resource_descriptors_length is 0, the pointer/array will never get accessed. 

This should be a safe fix for all platforms.

xlz, 2016-02-11:
This zero length array is non-standard. I'll create a fix later.

BaptisteTheoriz, 2016-02-11:
Solution proposed by @ofTheo works for me. Thanks !

xlz, 2016-02-11:
I would fix it in this way
```diff
-static ResourceDescriptor resource_descriptors[] = {};
+static ResourceDescriptor resource_descriptors[] = {{NULL, NULL, 0}};
```

`static ResourceDescriptor * resource_descriptors;` makes it a NULL pointer and dereferencing NULL pointers is undefined behavior. Even though it's after the check, I'm not very comfortable with the potential optimization the compiler could do.

xlz, 2016-02-14:
Fixed in d987d2c5dd8c6761a50aba1d569bda10f0d897fb

closed 2016-02-14 by xlz