OpenKinect/libfreenect2 #444: [Bug fix] for avoiding error or exception during open kinect devices

Edwinzero, 2015-10-22:
My friend and I find the deallocate method in command_transaction.cpp will cause potential error during open multiple devices. Currently we implement 4 Kinect2 devices on one desktop and make it run in real-time(30fps+)
The modification shows below.
[image]
If you don't add "data = NULL" then it means when call deallocate() will delete data variable once. And when the lifecirle end, destructor will delete data variable again. This potentially will cause LIBUSB_ERROR = -7 and -6, if you run your program frequently and you have to unplug all the Kinects and re plug-in.
Not a big deal but I think in this way this lib will more stable. 
@floe @xlz

xlz, 2015-10-22:
Valid concern. Should fix it with `delete[] data; data = NULL`. `if (data != NULL)` is unnecessary.

But the behavior of a double free is program crash, not libusb errors. Your LIBUSB_ERROR_BUSY (-6) _TIMEOUT (-7) are caused by something else.

Edwinzero, 2015-11-02:
Hello @xlz ,

While I am implementing multiple Kinect v2 by using thread, there always happen a -1 LIBUSB_ERROR_IO.
The log shows in the console is like following:
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:30 with serial 008427750247
[Freenect2Impl] found valid Kinect v2 @2:33 with serial 506904442542
[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e
2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e
2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e
802e2e2e2e2e
0x0040:  31 33 00 00 00 03 11 0f 47 4d 53 33 35 32 2e 31    31332e2e2e2e2e2e474d
533335322e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e
2e2e

[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e
2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e
2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e
802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0e 13 47 4b 4b 30 31 38 2e 31    31332e2e2e2e2e2e474b
4b3031382e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e
2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  6f 26 00 00                                        6f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  6f 26 00 00                                        6f262e2e

4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[CommandTransa[cFtrieoenn:e:crte2cDeeivviec]e Ibmupllk]  tsrtaanrstfeedr
 failed! libusb error -1: LIBUSB_ERROR_IO
[CommandTransaction::execute] missing response complete!
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TurboJpegRgbPacketProcessor::doProcess] Failed to decompress rgb imag[eF!r eTeu
[... 90 lines cut ...]
			// Run Kinect
			mocaThread.resize(devices.size());
			for (int i = 0; i < devices.size(); i++) {
				// run thread
				mocaThread[i] = (HANDLE)_beginthreadex(							// return the HANDLE of thread
					NULL,				// security
					0,					// stack_size "0" = default (1MB)
					RunFreenect2,		// address of function
					&freeArgt[i],
					0,					// 0 indicates manipulate immediately 
					NULL);				// return thread ID, NULL means no return
			}
			
		}
```
I know this is not the newest version, but I think it may causes by multiple tread access OpenDevice at the same time cause the crush problem? Since it gives crush when thread calls start() function for Kinect.
Another thing is all the device info are from the same ```libfreenect2::Freenect2:: freenect2 ``` object. I saw some method says create multiple object for each device. But after I create freenect2 for each device the error is still exist...
```
dbgdel.cpp
```
[image]
Error shown in the call stack.
[image]

Thank you for concerning this matter.

Edwinzero, 2015-11-03:
I add CRITICAL_SECTION to make sure all the devices start first, then proceed to process the captured frames. But it gives the ERROR: LIBUSB_ERROR_TIMEOUT -7 after execute   ```command_tx_.execute(ReadP0TablesCommand(nextCommandSeq()), result);``` in libfreenect2.cpp
[image]
THEN execute ``` command_tx_.execute(ReadRgbCameraParametersCommand(nextCommandSeq()), result);``` also gives -7 ERROR_TIMEOUT, 
Then found ```ReadData0x14 response 0 bytes of raw data... ```
The console log shows below:
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 17 usb devices connected
[Freenect2Impl** S] try to access new usb devices...
[Freenect2Impl] found valid Kinect v2 @2:10 with serial 026791351247
[Freenect2Impl** S] try to access new usb devices...
[Freenect2Impl] found valid Kinect v2 @2:7 with serial 503474541942
[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e
2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e
2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e
802e2e2e2e2e
0x0040:  31 33 00 00 00 05 03 15 47 4d 42 30 30 34 2e 31    31332e2e2e2e2e2e474d
423030342e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e
2e2e

[CommandTransaction::send] bulk transfer failed! libusb error -7: LIBUSB_ERROR_T
IMEOUT
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 20 got: 0
[Transaction::send] bulk transfer failed! libusb error -7: LIBUSB_ERROR_TIMEOUT
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 20 got: 0
[Freenect2DeviceImpl] ReadData0x14 response
0 bytes of raw data
```
Boot kinect part code shows below:
[image]

Please help me....

closed 2015-11-06 by Edwinzero