OpenKinect/libfreenect2 #445: Accuracy and precision of libfreenect2 ?

moonsh, 2015-10-23:
M.  Bueno, L.  Díaz-Vilariño, J.  Martínez-Sánchez, H.  González-Jorge, H.  Lorenzo and P.  Arias, 'Metrological evaluation of KinectFusion and its comparison with Microsoft Kinect sensor', Measurement, vol. 73, pp. 137-145, 2015.

This paper compared Kinect fusion with Kinect depth sensor especially for precision and accuracy. 
 I think libfreenect2 is using kinect depth sensor so... my question is here.

Can I confirm libfeenect2 accuracy is same as Kinect depth sensor ? Cause I guess libfreenect2 is exporting data from the sensor.

floe, 2015-10-23:
I don't quite understand your question. Are you talking about the Kinect v1, or the Kinect v2 (also known as Kinect One)? libfreenect2 delivers data from the Kinect _v2_, while that paper discusses the Kinect _v1_. Also, there may be subtle differences between using the official Kinect v2 SDK, and libfreenect2, but we currently don't have any exact measurements.

moonsh, 2015-10-23:
This paper dosn't mention clearly which kinect was tested but I guess Kinect V2 was used . I am doing project which is important to get high accuracy so I am curious about point accuracy of libfreenect 2 comparing with Kinect v2 SDK.

floe, 2015-10-23:
Had a look at the article myself - on page 4, they clearly show and talk about a Kinect v1. Maybe this paper (found by @xlz) can be helpful: http://designfankhauser.ch/files/fankhauser_2014_kinect_v2.pdf (although it's based on an older version of libfreenect).

HenningJ, 2015-10-23:
There is also [Sarbolandi, H., Lefloch, D., & Kolb, A. (2015): "Kinect Range Sensing: Structured-Light versus Time-of-Flight Kinect"](http://arxiv.org/pdf/1505.05459.pdf), comparing Kinect v1, Kinect v2 Microsoft SDK and Kinect v2 libfreenect2 according to various different characteristics. 
I think the use a libfreenect2 version from January 2015

floe, 2015-10-23:
Interesting, thanks for the paper - once we have release 0.1 (and it can't be long now :-) it might be worth writing to the authors of these papers and ask if they would be interested in re-running their tests with the release.

moonsh, 2015-10-25:
Thanks a lot.  My question is solved . I will close this issue.

closed 2015-10-25 by moonsh