OpenKinect/libfreenect2 #448: failed to claim interface with IrInterfaceId(=1)! on win10

ShineSong, 2015-10-25:
Hey guys,
     I want to connect multiple kinect v2, but offical sdk cannot offer this feature.I'm working on windows now (my linux pc don't have usb3.0).the libusbK 3.0.6.0 has been install as kinect driver for both NUI 0 and 2 ( I tried for 0 alone,but also not work).There are some difference on USBID  between README (045E, PID 02C4) and I (045E 02D8).
    When i run protonect i got this message.
====================================
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:16 with serial 500329242742
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=4 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2] failed to open Kinect v2: @2:16
failure opening device!
===============================
then set LIBUSB_DEBUG=4 , print below message:
================================
[Info] [Freenect2Impl] found 1 devices
[ 0.366607] [000031cc] libusb: debug [handle_events] poll() returned 0
[ 0.366607] [000031cc] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.366607] [000031cc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.367089] [000031cc] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.378110] [000032ec] libusb: debug [libusb_open] open 2.16
[ 0.378110] [000032ec] libusb: debug [libusb_reset_device]
[Info] [Freenect2DeviceImpl] opening...
[ 0.380599] [000032ec] libusb: debug [libusb_get_configuration]
[ 0.381099] [000032ec] libusb: debug [libusb_get_configuration] active config 1
[ 0.381099] [000032ec] libusb: debug [libusb_claim_interface] interface 0
[ 0.381099] [000032ec] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.381601] [000032ec] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.381601] [000032ec] libusb: debug [parse_configuration] skipping descriptor 0xb
[ 0.382104] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.382104] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.382104] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.382600] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.382600] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.382600] [000032ec] libusb: debug [parse_endpoint] skipping descriptor b
[ 0.383100] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 30
[ 0.383100] [000032ec] libusb: debug [parse_endpoint] skipping descriptor 25
[ 0.383100] [000032ec] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.383421] [000032ec] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 0.383421] [000032ec] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 83 to interface 0
[ 0.383421] [000032ec] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 82 to interface 0
[ 0.383421] [000032ec] libusb: debug [libusb_claim_interface] interface 1
[ 0.383421] [000032ec] libusb: debug [unsupported_claim_interface] unsupported API call for 'claim_interface' (unrecognized device driver)
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=4 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[ 0.397933] [000032ec] libusb: debug [libusb_close]
[ 0.397933] [000031cc] libusb: debug [handle_events] poll() returned 1
[ 0.398433] [000031cc] libusb: debug [handle_events] caught a fish on the event pipe
[ 0.398433] [000031cc] libusb: debug [handle_events] someone is closing a device
[ 0.400934] [000031cc] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[Info] [Freenect2DeviceImpl] closed
[ 0.400934] [000031cc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.401936] [000031cc] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[Error] [Freenect2] failed to open Kinect v2: @2:16
failure opening device!
[ 0.502600] [000031cc] libusb: debug [handle_events] poll() returned 0
[ 0.502600] [000032ec] libusb: debug [libusb_exit]
[ 0.503101] [000032ec] libusb: debug [libusb_exit] destroying default context
[ 0.503101] [000032ec] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 0.503101] [000032ec] libusb: warning [libusb_exit] application left some devices open
[ 0.503101] [000032ec] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.503602] [00001c60] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
=============================

MS SDK could connect to kinect perfect,and ...i have no idea of my usb3.0 chipset,In my device manager - universal serial bus controller  have "superspeed" and "inter usb 3.0 expandiable host controller" and in system device ,there are "Inter(R) Series/C220 Series PCI Express Root Port".
My laptop is Alienware 17 R5.

xlz, 2015-10-25:
The latest version of libusbK is v3.0.7.0. Have you update it?

ShineSong, 2015-10-25:
@xlz yeah, i have tried 3.0.7.0.It report same error,and i noticed that readme document denote 3.0.6.0 so i try the history version of usbK.But still not work..

xlz, 2015-10-25:
Can you post the full log with LIBUSB_DEBUG on gist.github.com? I need earlier log about driver setup.

ShineSong, 2015-10-25:
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00002344] libusb: debug [libusb_init] created default context
[ 0.000000] [00002344] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000000] [00002344] libusb: debug [windows_init] Windows 8 64-bit
[ 0.000000] [00002344] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.001000] [00002344] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.001000] [00002344] libusb: debug [winusbx_init] libusbK version: 3.0.6.0
[ 0.001987] [00002344] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.001987] [00002344] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.001987] [00002344] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.001987] [00002344] libusb: debug [windows_init] hires timer available (Frequency: 2435776 Hz)
[ 0.001987] [00002344] libusb: debug [windows_init] timer thread will run on core #0
[ 0.002990] [00002344] libusb: debug [htab_create] using 1021 entries hash table
[ 0.002990] [00002344] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[Info] [Freenect2Impl] enumerating devices...
[ 0.002990] [00002344] libusb: debug [libusb_get_device_list] 
[ 0.002990] [00002df8] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.002990] [00002df8] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.002990] [00002df8] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.002990] [00002df8] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.002990] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [5E]
[ 0.002990] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [266]
[ 0.002990] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [219]
[ 0.004002] [00002344] libusb: debug [get_api_type] driver(s): usbhub
[ 0.004002] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.004002] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [2C4]
[ 0.004002] [00002344] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.004002] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.004002] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [341]
[ 0.004002] [00002344] libusb: debug [get_api_type] driver(s): usbhub
[ 0.004002] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.004002] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [1D6]
[ 0.004002] [00002344] libusb: debug [get_api_type] driver(s): usbhub
[ 0.004002] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.004002] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [3B8]
[ 0.004002] [00002344] libusb: warning [windows_get_device_list] could not retrieve port number for device '\\.\USB#VID_045E&PID_02D9#5&2F2F3B60&0&21', skipping: [13] 数据无效。
[ 0.005003] [00002344] libusb: debug [get_api_type] driver(s): usbhub
[ 0.005003] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005003] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [38F]
[ 0.005003] [00002344] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005003] [00002344] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005003] [00002344] libusb: debug [windows_get_device_list] allocating new device for session [236]
[ 0.005003] [00002344] libusb: debug [windows_get_device_list] found existing device for session [2C4] (0.0)
[ 0.005003] [00002344] libusb: debug [init_device] got bus number from ancestor #2
[ 0.101951] [00002344] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.101951] [00002344] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.101951] [00002344] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 1): '\\.\USB#VID_8087&PID_8008#5&1E5119AE&1&1'
[ 0.103070] [00002df8] libusb: debug [handle_events] poll() returned 0
[ 0.103070] [00002df8] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.103070] [00002df8] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.103070] [00002df8] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.103070] [00002344] libusb: debug [windows_get_device_list] found existing device for session [236] (0.0)
[ 0.103070] [00002344] libusb: debug [init_device] (bus: 2, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&1D7A8555&0&0'
[ 0.103070] [00002344] libusb: debug [windows_get_device_list] found existing device for session [1D6] (0.0)
[ 0.103070] [00002344] libusb: debug [init_device] got bus number from ancestor #2
[ 0.103070] [00002344] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.103070] [00002344] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.103070] [00002344] libusb: debug [init_device] (bus: 3, addr: 5, depth: 1, port: 1): '\\.\USB#VID_8087&PID_8000#5&392DF00B&1&1'
[... cut ...]
[ 0.203293] [00002df8] libusb: debug [handle_events] poll() returned 0
[ 0.203293] [00002344] libusb: debug [libusb_exit] 
[ 0.203293] [00002344] libusb: debug [libusb_exit] destroying default context
[ 0.203293] [00002344] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.203293] [00002d70] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

@xlz  I can't visit gist.github.com now,I paste output of LIBUSB_DEBUG=4 above.not contained [ERROR] message (I can't redirect all output stream to file now).

ShineSong, 2015-10-25:
We have different device v2 for win(045E 02D8) and v2 for xbox (045E, PID 02C4) ,but both of them stopped for same error " failed to claim interface with IrInterfaceId(=1)! "

xlz, 2015-10-25:
The log says no driver is installed for Kinect. In fact, Kinect provides 2 hub devices, and one composite device. You need to install driver only for the composite device. 

> libusbK 3.0.6.0 has been install as kinect driver for both NUI 0 and 2 ( I tried for 0 alone,but also not work)

What does this mean, what are "NUI 0 and 2", can you post a screenshot of the list of "NUI 0 and 2"?

ShineSong, 2015-10-26:
okay...I made a stupid mistake,careless reading.Now it work prety good.Appreciate for this project,I will try to combine multi-kinect now. :+1:

ShineSong, 2015-10-28:
I have another problem now.
I installed usbK 3.0.7.0 correctly,and i can read data frome kinect v2 use another opensource project rtab(https://github.com/introlab/rtabmap) which choose freenect2 as data-source backend.
[image]
But , when use Protonect,I can't read any data from sensor,error message are the same:

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[ 0.320789] [00002bac] libusb: debug [unsupported_submit_iso_transfer] unsupported API call for 'submit_iso_transfer' (unrecognized device driver)

libfreenect2 was built with vs2013,I only modify FindTurboJPEG.cmake frome "FIND_PACKAGE_HANDLE_STANDARD_ARGS(TurboJPEG DEFAULT_MSG TurboJPEG_LIBRARIES TurboJPEG_INCLUDE_DIRS TURBOJPEG_WORKS)" to "FIND_PACKAGE_HANDLE_STANDARD_ARGS(TurboJPEG DEFAULT_MSG TurboJPEG_LIBRARIES TurboJPEG_INCLUDE_DIRS)"

ShineSong, 2015-10-28:
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00002bac] libusb: debug [libusb_init] created default context
[ 0.001001] [00002bac] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.001001] [00002bac] libusb: debug [windows_init] Windows 8 64-bit
[ 0.001001] [00002bac] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.002002] [00002bac] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.002002] [00002bac] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.002002] [00002bac] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.002002] [00002bac] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.003002] [00002bac] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.003002] [00002bac] libusb: debug [windows_init] hires timer available (Frequency: 2435777 Hz)
[ 0.003002] [00002bac] libusb: debug [windows_init] timer thread will run on core #0
[ 0.003002] [00002bac] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003002] [00002bac] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.003002] [0000125c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.003002] [0000125c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003002] [0000125c] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.003002] [0000125c] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[Info] [Freenect2Impl] enumerating devices...
[ 0.003002] [00002bac] libusb: debug [libusb_get_device_list] 
[ 0.004003] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [5E]
[ 0.004003] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [266]
[ 0.004003] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [219]
[ 0.004003] [00002bac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.004003] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.004003] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [2C4]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [1D6]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [3B8]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): usbhub
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [38F]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [27A]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [236]
[ 0.005004] [00002bac] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.005004] [00002bac] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.005004] [00002bac] libusb: debug [windows_get_device_list] allocating new device for session [342]
[ 0.006004] [00002bac] libusb: debug [windows_get_device_list] found existing device for session [2C4] (0.0)
[ 0.006004] [00002bac] libusb: debug [init_device] got bus number from ancestor #2
[ 0.103490] [0000125c] libusb: debug [handle_events] poll() returned 0
[ 0.103490] [0000125c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.103490] [0000125c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.103490] [0000125c] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.104491] [00002bac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.104491] [00002bac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.104491] [00002bac] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 1): '\\.\USB#VID_8087&PID_8008#5&1E5119AE&1&1'
[ 0.105589] [00002bac] libusb: debug [windows_get_device_list] found existing device for session [236] (0.0)
[ 0.105589] [00002bac] libusb: debug [init_device] (bus: 2, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&1D7A8555&0&0'
[ 0.105589] [00002bac] libusb: debug [windows_get_device_list] found existing device for session [1D6] (0.0)
[ 0.105589] [00002bac] libusb: debug [init_device] got bus number from ancestor #2
[ 0.105589] [00002bac] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.105589] [00002bac] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.105589] [00002bac] libusb: debug [init_device] (bus: 3, addr: 5, depth: 1, port: 1): '\\.\USB#VID_8087&PID_8000#5&392DF00B&1&1'
[... cut ...]
[ 1.477747] [0000125c] libusb: debug [usbi_handle_transfer_completion] transfer 000000AFAFA8EC30 has callback 00007FFAB5C5BA50
[ 1.477747] [0000125c] libusb: debug [libusb_submit_transfer] transfer 000000AFAFA8EC30
[ 1.477747] [0000125c] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[ 1.477747] [0000125c] libusb: debug [winusbx_submit_bulk_transfer] reading 16384 bytes
[ 1.477747] [0000125c] libusb: debug [usbi_add_pollfd] add fd 13 events 1
[ 1.477747] [0000125c] libusb: debug [libusb_get_next_timeout] next timeout in 0.999589s
[ 1.477747] [0000125c] libus

ShineSong, 2015-10-28:
I tried tracking where throw this error,and here ,in libusb, I found transfer->type going to case LIBUSB_TRANSFER_TYPE_ISOCHRONOUS branch and return the error code -12 .AND this is from r = priv->apib->submit_iso_transfer(SUB_API_NOTSET, itransfer); that the function ptr member of 'apib' is an function that print "UNSUPPORTED_API".I have no idea about how libusb working and how to solve this error.BUILD INCORRECT? SYSTEM PROBLEM? CHIPSET HARDWARE?

static int windows_submit_transfer(struct usbi_transfer *itransfer)
{
	struct libusb_transfer *transfer = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);

	switch (transfer->type) {
	case LIBUSB_TRANSFER_TYPE_CONTROL:
		return submit_control_transfer(itransfer);
	case LIBUSB_TRANSFER_TYPE_BULK:
	case LIBUSB_TRANSFER_TYPE_INTERRUPT:
		if (IS_XFEROUT(transfer) &&
		    transfer->flags & LIBUSB_TRANSFER_ADD_ZERO_PACKET)
			return LIBUSB_ERROR_NOT_SUPPORTED;
		return submit_bulk_transfer(itransfer);
	case LIBUSB_TRANSFER_TYPE_ISOCHRONOUS:
		return submit_iso_transfer(itransfer);
	case LIBUSB_TRANSFER_TYPE_BULK_STREAM:
		return LIBUSB_ERROR_NOT_SUPPORTED;
	default:
		usbi_err(TRANSFER_CTX(transfer), "unknown endpoint type %d", transfer->type);
		return LIBUSB_ERROR_INVALID_PARAM;
	}
}

xlz, 2015-10-28:
How did you build libusb?

ShineSong, 2015-10-28:
@xlz Download source code from sourceforge,libusb project.And build with msvc/libusb_2013.sln,x64 release/debug

xlz, 2015-10-28:
Wrong. Please see README.

ShineSong, 2015-10-28:
Got it,I'll test it tomorrow,thanks a lot.

floe, 2015-11-06:
Should be fixed by #429, please re-test with latest master and re-open if necessary.

closed 2015-11-06 by floe