OpenKinect/libfreenect2 #451: Frame rate control

vaheta, 2015-10-27:
Is there a way to decrease FPS to be able to process the data on slow PCs? Right now I am getting a lot of "skipped rgb/depth", I think this is due to the slow pc, which is not able to process the frames on Kinect frame rate.

ShineSong, 2015-11-02:
I meet same problem recently，maybe caused by low bus speed。 I have 2 pcie2usb3 card (nec chip),and an usb3 controller on the motherboard.when use external card,it's work well while loss frame on motherboard port.

Edwinzero, 2015-11-03:
Each Kinect needs at least 3 Gbps bandwidth. The slot on mother board is not parallel, which means the speed will be slow. Just put all the Kinect devices on usb3 card.

xlz, 2015-11-03:
Why, you seeing "skipped rgb/depth" is exactly how frame rate control looks like. If processing is slower than the source rate, then output rate is slower.

vaheta, 2015-11-04:
@xlz, got it, thanks! Am I correct, that the processing is done mostly on GPU in the default mode? So to be able to process faster I need better GPU?

xlz, 2015-11-04:
In general yes.

closed 2015-11-05 by vaheta