OpenKinect/libfreenect2 #453: Configureation of OpenCL dependency in Odroid xu4 installed 14.04

CrazyPopLin, 2015-11-03:
Hi, Could you give me some suggestions about how to configure OpenCL in Arm environment with **mali** GPU. I've done some research about lots issues in Arm Ubuntu 14.04. 
Any suggestions will be thankful.

xlz, 2015-11-03:
See #377

closed 2015-11-04 by floe

CrazyPopLin, 2015-11-19:
@xlz, thanks for your reply. i think i set up the environment nice without any issues with your advices (at least no issues in build).
But i got access denied like below:
>odroid@odroid:~/libfreenect2/build/bin$ ../bin/Protonect
>Environment variables: LOGFILE=<protonect.log>
Usage: ../bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Error] [Freenect2Impl] failed to open Kinect v2: @4:4 LIBUSB_ERROR_ACCESS Access denied (insufficient permissions)
>[Info] [Freenect2Impl] found 0 devices
>no device connected!

And I think i got correct usb like:   **libusb-1.0.so.0 => /home/odroid/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0xb6cc4000)**
Could you give me some suggestions, thanks.

xlz, 2015-11-19:
Have you set up udev rules

CrazyPopLin, 2015-11-20:
thx @xlz , i run `./Protonect cl` and got some error like below:
>odroid@odroid:~/libfreenect2/build/bin$ ./Protonect cl
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 501169643042
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenCLDepthPacketProcessorImpl] cl::Platform::get failed: -1001
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 08 01 04 47 4d 56 35 34 37 2e 31    31332e2e2e2e2e2e474d563534372e31
>0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e
>
>[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e
>
[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e
>
>[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 501169643042
device firmware: 2.3.3912.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
libGL error: MESA-LOADER: malformed or no PCI ID
libGL error: dlopen /usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen ${ORIGIN}/dri/exynos_dri.so failed (${ORIGIN}/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen /usr/lib/dri/exynos_dri.so failed (/usr/lib/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
GLFW error 65543 The requested client API version is unavailable
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
>Failed to create opengl window.
 
Is that because I miss some OpenCL files? Also when I run `./Protonect cl` is that supported to process kinect data just with OpenCL. I knew odroid doesn't work with OpenGL, and I see #254 you mentioned just enabled OpenCL in build. I am newcomer with Cmake and could you tell which one is that. Because there are so many Cmakelists. 
Also there are outputs with **cmake..**
>odroid@odroid:~/libfreenect2/build$ cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
[... 90 lines cut ...]
[sudo] password for odroid: 
[  3%] Built target generate_resources_tool
[ 88%] Built target freenect2
[100%] Built target Protonect
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/lib/libfreenect2.so
-- Set runtime path of "/usr/local/lib/libfreenect2.so" to "/home/odroid/libfreenect2/depends/libusb/lib"
-- Installing: /usr/local/include/libfreenect2
-- Installing: /usr/local/include/libfreenect2/registration.h
-- Installing: /usr/local/include/libfreenect2/frame_listener.hpp
-- Installing: /usr/local/include/libfreenect2/rgb_packet_processor.h
-- Installing: /usr/local/include/libfreenect2/packet_processor.h
-- Installing: /usr/local/include/libfreenect2/depth_packet_processor.h
-- Installing: /usr/local/include/libfreenect2/logger.h
-- Installing: /usr/local/include/libfreenect2/frame_listener_impl.h
-- Installing: /usr/local/include/libfreenect2/libfreenect2.hpp
-- Installing: /usr/local/include/libfreenect2/packet_pipeline.h
-- Installing: /usr/local/include/libfreenect2
-- Installing: /usr/local/include/libfreenect2/config.h
-- Installing: /usr/local/include/libfreenect2/export.h
-- Installing: /usr/local/lib/cmake/freenect2/freenect2Config.cmake
>-- Installing: /usr/local/lib/pkgconfig/freenect2.pc

thanks for your help again. That's a lot stuff to read. really appreciate :)

xlz, 2015-11-20:
So you no longer have LIBUSB_ERROR_ACCESS?

Can you check the additional instructions for Mali GPU? https://github.com/OpenKinect/libfreenect2/pull/472/files

CrazyPopLin, 2015-11-21:
thx, LIBUSB_ERROR_ACCRSS is solved, i will check this instructions.

CrazyPopLin, 2015-12-25:
Hi @xlz , I've been doing other stuff for a while. the odroid board was broken and I reinstall **libfreenect2** on a new odroid xu4 installed ubuntu 14.04.
According to odroid did not support OpenGL so I did not install **GLFW3** this time but setup the last properly I think. 
>
odroid@odroid:~/Dependency/libfreenect2/build/bin$ ./Protonect cl -noviewer
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:7 with serial 501169643042
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 501169643042
device firmware: 2.3.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 160
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] not all subsequences received 64
[Debug] [DepthPacketStreamParser] not all subsequences received 1020
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
....
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [DepthPacketStreamParser] 20 packets were lost
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
....
[Info] [Freenect2DeviceImpl] stopping...
[... 49 lines cut ...]
	libxcb-render.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-render.so.0 (0xb5bcd000)
	libxcb-dri2.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-dri2.so.0 (0xb5bc2000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabihf/libXext.so.6 (0xb5bb0000)
	libXau.so.6 => /usr/lib/arm-linux-gnueabihf/libXau.so.6 (0xb5ba5000)
	libXfixes.so.3 => /usr/lib/arm-linux-gnueabihf/libXfixes.so.3 (0xb5b99000)
	libXdmcp.so.6 => /usr/lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb5b8d000)
	libXcomposite.so.1 => /usr/lib/arm-linux-gnueabihf/libXcomposite.so.1 (0xb5b82000)
	libXrender.so.1 => /usr/lib/arm-linux-gnueabihf/libXrender.so.1 (0xb5b73000)
	libXrandr.so.2 => /usr/lib/arm-linux-gnueabihf/libXrandr.so.2 (0xb5b65000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb5b57000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb5b4c000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb5aa2000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb5a36000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5a15000)
>	/lib/ld-linux-armhf.so.3 (0xb6f14000)

**dpkg -S libOpenCL.so**
>ocl-icd-libopencl1:armhf: /usr/lib/arm-linux-gnueabihf/libOpenCL.so.1
ocl-icd-libopencl1:armhf: /usr/share/man/man7/libOpenCL.so.7.gz
ocl-icd-libopencl1:armhf: /usr/lib/arm-linux-gnueabihf/libOpenCL.so
>ocl-icd-libopencl1:armhf: /usr/lib/arm-linux-gnueabihf/libOpenCL.so.1.0.0

Also I try `echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd` and `echo /usr/lib/arm-linux-gnueabihf/mali-egl/libmali.so >/etc/OpenCL/vendors/mali-arm7hf.icd`, they seems had same results as shown above. 

Thanks for you help and Merry Xmas!

xlz, 2015-12-25:
I think it's running alright. Use `cmake -DENABLE_CXX11=ON` to show performance information again. The timing function either comes out of GLFW or C++11 stdlib.

CrazyPopLin, 2015-12-25:
thx, I got it. which means I can use **iai_kinect2** the ros driver to get depth data, right?

CrazyPopLin, 2016-01-07:
Hi @xlz, I was running `libfreenect2` and `iai_kinect2` perfectly well on odroid board as you commanded.
But today I just can't.

**run following command with sudo had same result**
>odroid@odroid:~/Dependency/libfreenect2/build/bin$ ./Protonect cl -noviewer
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found 0 devices
>no device connected!

**lsusb**
>Bus 003 Device 012: ID 045e:02d9 Microsoft Corp. 

**dmesg**
>[ 6235.211038] [c0] usb 3-1.2: new high-speed USB device number 12 using xhci-hcd
[ 6235.251855] [c0] hub 3-1.2:1.0: USB hub found
>[ 6235.252666] [c0] hub 3-1.2:1.0: 1 port detected

Also I double check the rule is in the `/etc/udev/rules.d` and the usb:
**dpkg -l libusb-1.0-0-dev**
>ii  libusb-1.0-0-dev:armhf            2:1.0.20-1ppa1~trusty armhf                 userspace USB programming library development files

 **ldd ./Protonect | grep usb**
>libusb-1.0.so.0 => /home/odroid/Dependency/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0xb6dce000)

The setup should be all right. Did that kinect2 has some probles?
Thanks so much.

xlz, 2016-01-07:
Seriously, when did you create the libusb copy in `libfreenect2/depends`? Nowhere in the README are you instructed to do anything there. I'm utterly confounded. It seems everyone has a libusb copy. How do they find out the way to do that? Maybe I should add a `rm rf ~/*` in `install_libusb.sh` so people would stop.

Jokes aside, the thing with it being working yesterday and not today just needs you to look at what has changed. `02d9` is the adaptor not the sensor. I think your cable is loose.

CrazyPopLin, 2016-01-07:
haha :joy: I think in problem 377 you said that.
>This is not the correct libusb.
>
>Have you run install_libusb.sh? You need the locally built libusb with proper USB3 support. ldd output >should point to something like /home/oskr1709/src/libfreenect2/depends/libusb/lib/libusb-1.0.so.0.

xlz, 2016-01-07:
OK I'll try to remove references to that.

02C4 or 02D8 is the product id of the sensor.

xlz, 2016-01-08:
What's the result? Is it working yet?

CrazyPopLin, 2016-01-08:
As you said just the cable loose. Thanks.

CrazyPopLin, 2016-01-08:
hi, I just wondering if I can disable the RGB registration to increase depth process. Based on arm process of RGB and depth, the rate of pointcloud I got just about 6 hz which is about the rostopic about `kinect2/sd/point`. Or it's better to ask on `iai_kinect2`. 
Please give me some suggestions, thanks.

xlz, 2016-01-08:
Colored point cloud must have registration. What are you willing to sacrifice?

CrazyPopLin, 2016-01-08:
I just want depth data to get `X,Y,Z` for obstacle avoidance, no color data needed. 
It's seems process too slow in arm board for both color and depth data at same time.
**First, I just want to make sure if it will increase depth process after disable colored registration. (Maybe disable in ROS level)**
Please correct me if I did not explain clearly.

xlz, 2016-01-08:
1. I don't think you need ROS for this. Cut out the middleman.
2. libfreenect2 right now does not provide a way to not process JPEG. But it's not hard to disable that: comment out `rgb_transfer_pool_.submit(20);` in libfreenect2.cpp, and do not subscribe to `Frame::Color` in Protonect.cpp.
3. The depth processing can be sped up by skipping some filtering. Freenect2Device::setConfiguration() https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Freenect2Device.html#ae45b1eead7e5b2e1c8e2259ae55663ff can set EnableBilateralFilter and EnableEdgeAwareFilter to false.
4. Some aggressive math optimization flags e.g. `-cl-fast-relaxed-math` can be added to generateOptions() in opencl_depth_packet_processor.cpp.
5. Do not use registration routines and no registration will be performed.

CrazyPopLin, 2016-01-09:
thanks, I will try it.

CrazyPopLin, 2016-01-25:
Hi @xlz ,still the "cable loose problem". I think it's not physical connection problem. Because I connected kinect with another odroid which didn't  install `libfreenect2` and `iai_kinect2`, I can found **Bus 004 Device 012: ID 045e:02d8 Microsoft Corp**. But when I connect kinect with the odroid installed everything, I could't find **02d8** sometimes. 
Could you give me some suggestions to find what's wrong?
Thanks.

xlz, 2016-01-25:
Try to replug it in, otherwise I dont know.