OpenKinect/libfreenect2 #454: Using Kinect One + libfreenect2 for commercial applications

vaheta, 2015-11-03:
Is it legal to use Kinect One with libfreenect2 for commercial applications? As I understand, there are no problems with using libfreenect2, but not so sure about Kinect itself and couldn't find anything on Microsoft website. Does anyone know anything about it?
Thanks.

xlz, 2015-11-03:
IANAL but Kinect 2 is hardware. You own the hardware. You don't need any license or anyone's approval to do whatever you want with the hardware. The worse that could happen is voiding the warranty. That is my understanding.

floe, 2015-11-04:
The whole point of libfreenect2 is giving you the option to do with the hardware whatever you please, as long as you contribute back to the community as per the open-source license terms.

closed 2015-11-05 by vaheta