OpenKinect/libfreenect2 #455: [WIN Question] NEW and OLD Kinect one for windows has different port number!!!!

Edwinzero, 2015-11-03:
Hi,

I currently have one old Kinect one for windows and two Kinect one + windows adapters. When I use the program to install libUSBK on my computer, I found that Old version shows ```USB ID is VID 045E, PID 02C4```.
But when I try to install libfreenect2 on a new computer and plug with new Kinect one with adapter, I found that New version shows Sensor ```USB ID is VID 045E, PID 02D8``` and Adapter ```USB ID is VID 045E, PID 02D9```.I just choose Sensor ID and PID to install libUSBK. The version I install is libUSBK 3.0.7. Is this version able to support libfreenect2? Do I need to change the version back to libUSBK 3.0.6 as the README said?
Is this new&old devices gonna cause a LIBUSB error that the ```new_serial_``` got in ```start() of libfreenect2.cpp``` will not equal to the serial number which get by ```enumerateDevice() ```? The problem of multithreading Kinect2 is shown in #444 [Bug fix] for avoiding error or exception during open kinect devices.
If anyone know how to solve this please help.

Edwinzero, 2015-11-03:
I am trying to use windows multi-thread to make 4 Kinect v2 run simultaneously. But here I meet some problem with LIBUSB_TIMEOUT ERROR -7.

[image]
I am using multithread to start devices at the same time, but it seems libfreenect2 doesn't recognize the right serial number.  Detailed log shown below:
```
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:19 with serial 008427750247
[Freenect2Impl] found valid Kinect v2 @2:20 with serial 501778742542
[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:19 with serial 008427750247
[Freenect2Impl] found valid Kinect v2 @2:20 with serial 501778742542
[Freenect2Impl] found 2 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
Fi[nFdr eFeinresctt 2KDienveiccte
Impl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e
2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e
2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e
802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0e 13 47 4b 4b 30 31 38 2e 31    31332e2e2e2e2e2e474b
4b3031382e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e
2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TurboJpegRgbPacketProcessor::doProcess] Failed to decompress rgb image! TurboJP
[... 200 lines cut ...]
		{
			listener2.waitForNewFrame(frames2);

			//libfreenect2::Frame *rgb2 = frames2[libfreenect2::Frame::Color];
			libfreenect2::Frame *ir2 = frames2[libfreenect2::Frame::Ir];
			libfreenect2::Frame *depth2 = frames2[libfreenect2::Frame::Depth];

			//cv::imshow("rgb2", cv::Mat(rgb2->height, rgb2->width, CV_8UC3, rgb2->data));
			cv::imshow("ir2", cv::Mat(ir2->height, ir2->width, CV_32FC1, ir2->data) / 20000.0f);
			cv::imshow("depth2", cv::Mat(depth2->height, depth2->width, CV_32FC1, depth2->data) / 4500.0f);

			int key2 = cv::waitKey(1);

			protonect_shutdown = protonect_shutdown || (key2 > 0 && ((key2 & 0xFF) == 27)); // shutdown on escape
			listener2.release(frames2);
			
		}
		dev_2->stop();
		dev_2->close();

		return 0;
}
```

Many thanks....

xlz, 2015-11-03:
Wait a moment. You have multiple issues, please do not mix them. You should open one issue for each. I'm not looking at these closely before you organize them.

02C4 and 02D8 are two versions of Kinects. Just install libUSBK 3.0.7, I don't see any problem. Supposedly you should see multiple Kinects in "Device Manager" under "libusbK USB Devices" tree.

libfreenect2 detects both no problem.

You should be using Windows 10 (at least not Windows 7) and newest USB 3.0 driver for this. And always post debug log with LIBUSB_DEBUG=4 environment variable.

closed 2015-11-04 by Edwinzero