OpenKinect/libfreenect2 #456: [LIBUSB_ERROR_TIMEOUT -7] Multi-thread multiple Kinect error when reach device start() function

Edwinzero, 2015-11-04:
Sorry for bring you inconvenient @xlz, and here is the issue that I organized. 
When I implement multi-thread multiple Kinect capture, I always have LIBUSB_ERROR_TIMEOUT -7 error. Here is the console output... 
[image]

I keep digging into this ```[Freenect2DeviceImpl] serial number reported by libusb 503474541942 differs from serial number 铪铪铪铪铪铪 in device protocol!``` information and find that 
```
 command_tx_.execute(ReadSerialNumberCommand(nextCommandSeq()), serial_result);
  std::string new_serial = SerialNumberResponse(serial_result.data, serial_result.length).toString();
```
in libfreenect2.cpp generates the new_serial which can't match the serial_ member variable in Freenect2DeviceImpl class.
For better debugging, I also print out the libusb log as you said
```
nother thread is doing event handling
[ 1.895869] [00002ac8] libusbx: debug [libusb_get_next_timeout] next timeout in
0.388885s
[ 1.895869] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed] a
nother thread is doing event handling
[ 1.998451] [00002ac8] libusbx: debug [libusb_get_next_timeout] next timeout in
0.286248s
[ 1.999402] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed] a
nother thread is doing event handling
[ 2.100682] [00002ac8] libusbx: debug [libusb_get_next_timeout] next timeout in
0.183890s
[ 2.100682] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed] a
nother thread is doing event handling
[ 2.202975] [00002ac8] libusbx: debug [libusb_get_next_timeout] next timeout in
0.081681s
[ 2.202975] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed] a
nother thread is doing event handling
[ 2.286477] [00002ac8] libusbx: debug [libusb_cancel_transfer]
[ 2.286477] [00000d00] libusbx: debug [handle_events] poll() returned 0
[ 2.286477] [00002ac8] libusbx: debug [winusbx_abort_transfers] will use interfa
ce 0
[ 2.287468] [00002ac8] libusbx: debug [libusb_get_next_timeout] no URB with time
out or all handled by OS; no timeout!
[ 2.287468] [00000d00] libusbx: debug [libusb_get_next_timeout] no URB with time
out or all handled by OS; no timeout!
[ 2.288458] [00000d00] libusbx: debug [libusb_handle_events_timeout_completed] a
nother thread is doing event handling
[ 2.288458] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed] d
oing our own event handling
[ 2.290458] [00002ac8] libusbx: debug [handle_events] poll() 3 fds with timeout
in 100ms
[ 2.291470] [00002ac8] libusbx: debug [handle_events] poll() returned 1
[ 2.291470] [00002ac8] libusbx: debug [windows_handle_events] checking fd 0 with
 revents = 0000
[ 2.292471] [00002ac8] libusbx: debug [windows_handle_events] checking fd 1 with
 revents = 0000
[ 2.292471] [00002ac8] libusbx: debug [windows_handle_events] checking fd 2 with
 revents = 0004
[ 2.293460] [00002ac8] libusbx: debug [usbi_remove_pollfd] remove fd 2
[ 2.293460] [00002ac8] libusbx: debug [windows_transfer_callback] handling I/O c
ompletion with errcode 995, size 0
[ 2.294461] [00002ac8] libusbx: debug [windows_transfer_callback] detected timeo
ut
[ 2.294461] [00002ac8] libusbx: debug [usbi_handle_transfer_completion] transfer
 0000000006724E20 has callback 00007FFFD62DDEF0
[ 2.294461] [00002ac8] libusbx: debug [sync_transfer_cb] actual_length=0
[ 2.295462] [00002ac8] libusbx: debug [libusb_get_next_timeout] no URBs, no time
out!
[[ 2.295462] [00002ac8] libusbx: debug [libusb_handle_events_timeout_completed]
doing our own event handling
C[ 2.296463] [00002ac8] libusbx: debug [handle_events] poll() 2 fds with timeout
 in 100ms
ommandTransaction::send] bulk transfer failed! libusb error -7: LIBUSB_ERROR_TIM
EOUT
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 20 got: 0
[ 2.335491] [00000d00] libusbx: debug [winusbx_submit_bulk_transfer] matched end
point 02 with interface 0
[... 440 lines cut ...]
					0,					// 0 indicates manipulate immediately 
					NULL);				// return thread ID, NULL means no return
			}
			
		}

		void Wait(){
			WaitForMultipleObjects((int)freeArgt.size(), mocaThread.data(), TRUE, INFINITE);
		}
		void Release() {
			WaitForMultipleObjects((int)freeArgt.size(), mocaThread.data(), TRUE, INFINITE);
			for (int i = 0; i < mocaThread.size(); i++)
			{
				CloseHandle(mocaThread[i]);
			}
			mocaThread.clear();
			freeArgt.clear();
			DeleteCriticalSection(&m_cs);
		}
	};

	libfreenect2::Freenect2 KThreadBundle::freenect2;
``` 
Again,
Thank you very much for concerning this matter.

xlz, 2015-11-04:
The problem is literally LIBUSB_ERROR_TIMEOUT, that's all. That is, the bulk transfers to query the device serial number timed out, which is expected if it takes too long. It looks like your USB 3.0 transfers were completely stalled. And the timeout errors are correct behaviors. This does not look like a bug in libfreenect2 or libusb. It is probably related to system configuration.

So in the next step the serial numbers were surely invalid. 铪 is 0xEE 0xFE in GBK charset, so that is obviously uninitialized memory. This is not something important to look at.

A nice-to-have bug fix: libfreenect2 probably should fail gracefully when the serial number query fails.

> This performance can handle 4 Kinects run at the same time with interactive frame rate(28fps) with brute-force implementation. 

Can you elaborate on this? What is your brute-force implementation?

Your log also reads that only 2 Kinects are detected.

xlz, 2015-11-04:
One simple test is to run 4 instances of Protonect.exe <serial> and see how it works.

xlz, 2015-11-06:
Closed? Is this solved?

Edwinzero, 2015-11-06:
Yes, 
From hardware perspective, The USB driver causes the problem. You have to use Gerneral USB hub for Kinect. Microsoft default set the USB controller is Renesas USB 3.0 eExtensible host controller driver, only keep the Microsoft version.
Also from code perspective, libfreenect2 capture procedure I convert it to pure C style and declare the freenect2 and devices as global variables then multi-thread processing will be stable and no more error.

xlz, 2016-03-07:
I believe there are workarounds https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#multiple-kinects

There can even be saner defaults, but use the workarounds for now.

closed 2016-03-07 by xlz