OpenKinect/libfreenect2 #458: I met a problem when i install libfreenect2 step by step. please help me

peterlin300, 2015-11-07:
I follow the explanation install libfreenect2 on ubuntu14.04
In step6: Build the actual protonect executable
when i cmake .. in ~/libfreenect2/build
It said:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/joe/libfreenect2

-- Configuring incomplete, errors occurred!
See also "/home/joe/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/joe/libfreenect2/build/CMakeFiles/CMakeError.log".
What should I do???

kforkarthik, 2015-11-08:
I also encountered the same problem. I don't know what to do. If you find any solution, please share with me. Thanks !

floe, 2015-11-08:
I'm pretty sure you did not install libusb-1.0.20. What does `dpkg -l libusb-1.0-*` show?

xlz, 2015-11-08:
The README says

> Install libusb. sudo apt-get install libusb-1.0-0-dev. **The version must be >=1.0.20.**

Installing libusb 1.0.19 or having old libusb 1.0.19 lingering in `depends` can cause this.

kforkarthik, 2015-11-09:
How to install libusb1.0.20 ? Can you please provide the steps. dpkg -l libusb-1.0-* shows as follows (but I am sorry, it is in German)

Gewünscht=Unbekannt/Installieren/R=Entfernen/P=Vollständig Löschen/Halten
| Status=Nicht/Installiert/Config/U=Entpackt/halb konFiguriert/
         Halb installiert/Trigger erWartet/Trigger anhängig
|/ Fehler?=(kein)/R=Neuinstallation notwendig (Status, Fehler: GROSS=schlecht)
||/ Name           Version        Beschreibung
+++-==============-==============-============================================
ii  libusb-1.0-0   2:1.0.9~rc3-2u userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.9~rc3-2u userspace USB programming library developmen

floe, 2015-11-09:
libusb-1.0.9 seems to be very old, this is the version from Ubuntu 12.04. Before you try anything else, upgrade to Ubuntu 14.04 at least, otherwise, a lot of other things will also be to old for libfreenect2.

floe, 2015-11-09:
Once you're on 14.04, run `sudo apt-add-repository ppa:floe/libusb && sudo apt-get update && sudo apt-get upgrade` and try again.

peterlin300, 2015-11-09:
HI !! thank you very much. i'm so glad that i run cmake and sudo make install succesful
[  4%] Built target generate_resources_tool
[ 88%] Built target freenect2
[100%] Built target Protonect

but,when i run  sudo ./bin/Protonect it show

Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 027602444747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000: bla bla bla 0x0010:bla bla bla  0x0020: bla bla bla  0x0030:  bla bla bla 0x0040: bla bla bla  0x0050:  bla bla bla 
at finally:
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
what'happen and what should i do??

xlz, 2015-11-09:
> When you report USB issues, please attach relevant debug log from running the program with environment variable LIBUSB_DEBUG=4, and relevant log from dmesg.

peterlin300, 2015-11-09:
i so sad ,i'm very poor in english and i'm the new to ububtu
i really cant uderstand and i just know i need  to attached some information when i running the program.
would you please tell me how should i do

xlz, 2015-11-09:
Run `LIBUSB_DEBUG=4 ./Protonect`.

peterlin300, 2015-11-09:
like this??
joe@joe:~/libfreenect2/build$ sudo LIBUSB_DEBUG=4 ./bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000002] [000013ef] libusb: debug [libusb_init] created default context
[ 0.000023] [000013ef] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000046] [000013ef] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000058] [000013ef] libusb: debug [op_init] bulk continuation flag supported
[ 0.000064] [000013ef] libusb: debug [op_init] zero length packet flag supported
[ 0.000075] [000013ef] libusb: debug [op_init] sysfs can relate devices
[ 0.000080] [000013ef] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000288] [000013f0] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.000523] [000013ef] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.000534] [000013ef] libusb: debug [linux_get_device_address] scan usb1
[ 0.000579] [000013ef] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.000588] [000013ef] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.000594] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.000676] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 1-2 detached: 0
[ 0.000684] [000013ef] libusb: debug [linux_get_device_address] scan 1-2
[ 0.000720] [000013ef] libusb: debug [linux_get_device_address] bus=1 dev=8
[ 0.000727] [000013ef] libusb: debug [linux_enumerate_device] busnum 1 devaddr 8 session_id 264
[ 0.000733] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 1/8 (session 264)
[ 0.000762] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6c9700 (1-2) has parent 0x6c9640 (usb1) port 2
[ 0.000841] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 1-5 detached: 0
[ 0.000848] [000013ef] libusb: debug [linux_get_device_address] scan 1-5
[ 0.000883] [000013ef] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.000890] [000013ef] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.000896] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.000921] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6c97c0 (1-5) has parent 0x6c9640 (usb1) port 5
[ 0.000985] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 1-6 detached: 0
[ 0.000992] [000013ef] libusb: debug [linux_get_device_address] scan 1-6
[ 0.001018] [000013ef] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.001024] [000013ef] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.001030] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.001055] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6ca1c0 (1-6) has parent 0x6c9640 (usb1) port 6
[ 0.001151] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 1-8 detached: 0
[ 0.001159] [000013ef] libusb: debug [linux_get_device_address] scan 1-8
[ 0.001186] [000013ef] libusb: debug [linux_get_device_address] bus=1 dev=4
[ 0.001193] [000013ef] libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
[ 0.001198] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
[ 0.001221] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6ca280 (1-8) has parent 0x6c9640 (usb1) port 8
[ 0.001269] [000013ef] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001276] [000013ef] libusb: debug [linux_get_device_address] scan usb2
[ 0.001302] [000013ef] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001309] [000013ef] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001314] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001373] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.001380] [000013ef] libusb: debug [linux_get_device_address] scan 2-2
[ 0.001405] [000013ef] libusb: debug [linux_get_device_address] bus=2 dev=6
[ 0.001410] [000013ef] libusb: debug [linux_enumerate_device] busnum 2 devaddr 6 session_id 518
[ 0.001415] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 2/6 (session 518)
[ 0.001438] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6cb890 (2-2) has parent 0x6ca340 (usb2) port 2
[ 0.001466] [000013ef] libusb: debug [linux_get_device_address] getting address for device: 2-2.1 detached: 0
[ 0.001472] [000013ef] libusb: debug [linux_get_device_address] scan 2-2.1
[ 0.001498] [000013ef] libusb: debug [linux_get_device_address] bus=2 dev=7
[ 0.001503] [000013ef] libusb: debug [linux_enumerate_device] busnum 2 devaddr 7 session_id 519
[ 0.001508] [000013ef] libusb: debug [linux_enumerate_device] allocating new device for 2/7 (session 519)
[ 0.001532] [000013ef] libusb: debug [linux_get_parent_info] Dev 0x6cbd60 (2-2.1) has parent 0x6cb890 (2-2) port 1
[ 0.001644] [000013ef] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[... cut ...]
[ 0.270036] [000013ef] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 1835008
[ 0.270039] [000013ef] libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
[ 0.270041] [000013ef] libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
[ 0.270043] [000013ef] libusb: debug [disarm_timerfd] 
[ 0.270045] [000013ef] libusb: debug [libusb_free_transfer] transfer 0xb66138
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error

xlz, 2015-11-09:
And dmesg?

peterlin300, 2015-11-09:
i'm so sorry .i forgot
i run sudo dmesg ./Protonect

[    0.676552] efifb: scrolling: redraw
[    0.676554] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    0.676654] Console: switching to colour frame buffer device 170x48
[    0.676672] fb0: EFI VGA frame buffer device
[    0.676682] intel_idle: MWAIT substates: 0x11142120
[    0.676684] intel_idle: v0.4 model 0x45
[    0.676685] intel_idle: lapic_timer_reliable_states 0xffffffff
[    0.677081] ACPI: AC Adapter [AC0] (on-line)
[    0.677238] input: Lid Switch as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input0
[    0.679186] ACPI: Lid Switch [LID]
[    0.679234] input: Sleep Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input1
[    0.679238] ACPI: Sleep Button [SLPB]
[    0.679290] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
[    0.679292] ACPI: Power Button [PWRF]
[    0.682152] thermal LNXTHERM:00: registered as thermal_zone0
[    0.682155] ACPI: Thermal Zone [THRM] (45 C)
[    0.682204] GHES: HEST is not enabled!
[    0.682355] Serial: 8250/16550 driver, 32 ports, IRQ sharing enabled
[    0.684412] Linux agpgart interface v0.103
[    0.686071] brd: module loaded
[    0.686863] loop: module loaded
[    0.687109] libphy: Fixed MDIO Bus: probed
[    0.687115] tun: Universal TUN/TAP device driver, 1.6
[    0.687116] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[    0.687167] PPP generic driver version 2.4.2
[    0.687372] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.687380] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 1
[    0.687492] xhci_hcd 0000:00:14.0: cache line size of 64 is not supported
[    0.687586] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.687588] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.687590] usb usb1: Product: xHCI Host Controller
[    0.687592] usb usb1: Manufacturer: Linux 3.19.0-25-generic xhci-hcd
[    0.687594] usb usb1: SerialNumber: 0000:00:14.0
[    0.687733] hub 1-0:1.0: USB hub found
[    0.687752] hub 1-0:1.0: 9 ports detected
[    0.690950] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.690955] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[    0.690990] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[    0.690992] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.690994] usb usb2: Product: xHCI Host Controller
[    0.690996] usb usb2: Manufacturer: Linux 3.19.0-25-generic xhci-hcd
[    0.690998] usb usb2: SerialNumber: 0000:00:14.0
[    0.691112] hub 2-0:1.0: USB hub found
[    0.691120] hub 2-0:1.0: 4 ports detected
[    0.692327] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.692336] ehci-pci: EHCI PCI platform driver
[    0.692451] ehci-pci 0000:00:1d.0: EHCI Host Controller
[    0.692458] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 3
[    0.692471] ehci-pci 0000:00:1d.0: debug port 2
[    0.696362] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
[    0.696380] ehci-pci 0000:00:1d.0: irq 23, io mem 0xf7a23000
[    0.705881] ACPI: Battery Slot [BAT0] (battery present)
[    0.707326] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
[    0.707372] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002
[    0.707374] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.707377] usb usb3: Product: EHCI Host Controller
[    0.707379] usb usb3: Manufacturer: Linux 3.19.0-25-generic ehci_hcd
[    0.707380] usb usb3: SerialNumber: 0000:00:1d.0
[    0.707526] hub 3-0:1.0: USB hub found
[    0.707532] hub 3-0:1.0: 2 ports detected
[    0.707676] ehci-platform: EHCI generic platform driver
[    0.707690] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.707698] ohci-pci: OHCI PCI platform driver
[    0.707713] ohci-platform: OHCI generic platform driver
[    0.707724] uhci_hcd: USB Universal Host Controller Interface driver
[    0.707803] i8042: PNP: PS/2 Controller [PNP030b:PS2K,PNP0f03:PS2M] at 0x60,0x64 irq 1,12
[    0.710537] i8042: Detected active multiplexing controller, rev 1.1
[... 385 lines cut ...]
[ 3215.221238] usb 2-2.1: reset SuperSpeed USB device number 7 using xhci_hcd
[ 3215.236555] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 3215.236560] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 3215.236562] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 3215.236564] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 3215.290347] Protonect[4824]: segfault at 7ff0a8379010 ip 00007ff0b998abc2 sp 00007ffe202a6000 error 4 in libfreenect2.so[7ff0b9970000+3b000]
[ 4808.219805] usb 2-2.1: reset SuperSpeed USB device number 7 using xhci_hcd
[ 4808.234755] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 4808.234760] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 4808.234762] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 4808.234764] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 4808.253627] Protonect[4973]: segfault at 7f49c99c7010 ip 00007f49e069fbc2 sp 00007ffc1c179450 error 4 in libfreenect2.so[7f49e0685000+3b000]
[... cut ...]
[ 5033.705083] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 5033.705089] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 5033.705091] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 5033.705093] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 5033.723610] Protonect[5103]: segfault at 7f9a4251b010 ip 00007f9a59338bc2 sp 00007ffcfd3aaf90 error 4 in libfreenect2.so[7f9a5931e000+3b000]

xlz, 2015-11-09:
Which part of dmesg is relevant to the log you provided above? The dmesg shows your Protonect has crashed with segfault, which I do not see.

peterlin300, 2015-11-09:
i try to find google how to use dmesg and
i try run this command: dmesg | tail -20  after i run ./bin/Protonect in another Terminal
if anything wrong please teach me, I am sorry  this has caused you  inconvenience.

[ 4808.253627] Protonect[4973]: segfault at 7f49c99c7010 ip 00007f49e069fbc2 sp 00007ffc1c179450 error 4 in libfreenect2.so[7f49e0685000+3b000]
[ 4844.670962] systemd-hostnamed[5063]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 4923.673022] usb 2-2.1: reset SuperSpeed USB device number 7 using xhci_hcd
[ 4923.688001] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 4923.688006] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 4923.688008] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 4923.688010] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 4923.704570] Protonect[5070]: segfault at 7f0ed8c55010 ip 00007f0eefa66bc2 sp 00007ffdf25f8960 error 4 in libfreenect2.so[7f0eefa4c000+3b000]
[ 5033.689773] usb 2-2.1: reset SuperSpeed USB device number 7 using xhci_hcd
[ 5033.705083] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 5033.705089] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 5033.705091] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 5033.705093] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 5033.723610] Protonect[5103]: segfault at 7f9a4251b010 ip 00007f9a59338bc2 sp 00007ffcfd3aaf90 error 4 in libfreenect2.so[7f9a5931e000+3b000]
[ 7411.081492] usb 2-2.1: reset SuperSpeed USB device number 7 using xhci_hcd
[ 7411.097062] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b200
[ 7411.097067] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b248
[ 7411.097069] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b2d8
[ 7411.097071] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801cc71b290
[ 7411.113591] Protonect[5469]: segfault at 7f82bd3cc010 ip 00007f82d4146bc2 sp 00007ffc2e95dfa0 error 4 in libfreenect2.so[7f82d412c000+3b000]

xlz, 2015-11-09:
Try `sudo sh -c 'echo 64 > /sys/module/usbcore/parameters/usbfs_memory_mb'`

peterlin300, 2015-11-09:
Its work!! thank you very very much!! Thank for your instruction and advice
By the way,can you give me some advice about ROS? i  see your backgroud is focuses on visual-inertial navigation and PR2 
i just a newcomer in linux and ros,about 1Month.
if i want to using the Kinect One in ROS and learn ROS . how should i do? 
follow this web :https://github.com/code-iai/iai_kinect2 and a book "ros by example indigo By R. Patrick Goebel" step by step???

xlz, 2015-11-09:
`error [submit_bulk_transfer] submiturb failed error -1 errno=12` 12 is ENOMEM. If ENOMEM, get more memory: `echo 64 > /sys/module/usbcore/parameters/usbfs_memory_mb`.

Right, iai_kinect2 to the ROS counterpart for Kinect 2.

floe, 2015-11-09:
@xlz interesting, this is apparently sometimes necessary for a single Kinect, too?

xlz, 2015-11-09:
It should not. There is no change in USB buffer sizes. Another report is needed.

madadam, 2015-11-12:
Hello there,

I seem to have encountered this same problem. It was fixed by the above usbfs_memory_mb trick. Weird thing is it used to work fine until today... If you guys need any more info to help debug this, let me know.

kforkarthik, 2015-11-12:
@madadam I still have this problem. Can you please help me solve this ?

madadam, 2015-11-12:
@kforkarthik I only did the thing @xlz mentioned couple of posts above:

`sudo sh -c 'echo 64 > /sys/module/usbcore/parameters/usbfs_memory_mb'`

It worked for me. If it doesn't work for you, I have no idea how to help you.

xlz, 2015-11-13:
@madadam 
You have it working yesterday. So what has changed? libfreenect2 or libusb? Do you know a version that works and a version that doesn't work?

madadam, 2015-11-13:
That's the weird thing. Nothing changed as far as I know. I mean, obviously, something had to change, but don't know what. I know that neither libfreenct2, nor libusb changed. Also, today when I booted my computer up, I saw that the `usbfs_memory_mb` got reset back to 16, but Protonect seems to be working fine again.

madadam, 2015-11-13:
Ok, I have a bit more info, not sure how helpful:

When I boot up my computer, the usbfs_memory_mb setting has its default value of 16. Then when I run Protonect, everything seems to work fine. But when I run my own app (which utilizes libfreenect2)*, it immediately crashes. When I run Protonect afterwards, it crashes too. Increasing the memory thing fixes both apps (Protonect and my own). So obviously it is something I'm doing in my app that is causing this. Will try to investigate and report back, if/when I found time for it.

*The situation is a bit convoluted: It uses libfreenect2, then a custom c mini-wrapper on top of it, then a C# library which uses that c wrapper, and all of that is imported into a Unity3D game.

xlz, 2015-11-13:
@madadam 
About the crash, can you provide a stack trace? (`ulimit -c unlimited; crash; gdb Protonect core; bt`) At least it should fail nicely.

madadam, 2015-11-16:
Here you go:

```
GNU gdb (Ubuntu 7.10-1ubuntu2) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/bin/Protonect...done.
[New LWP 28125]
[New LWP 28126]
[New LWP 28127]
[New LWP 28128]
[New LWP 28129]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `build/bin/Protonect'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  libfreenect2::Freenect2DeviceImpl::start (this=0x25face0)
    at /home/adam/projects/freenect2/src/libfreenect2.cpp:685
685	  ir_camera_params_.fx = ir_p->fx;
[Current thread is 1 (Thread 0x7f923d5c7780 (LWP 28125))]
(gdb) bt
#0  libfreenect2::Freenect2DeviceImpl::start (this=0x25face0)
    at /home/adam/projects/freenect2/src/libfreenect2.cpp:685
#1  0x00000000004044bd in main (argc=<optimized out>, argv=<optimized out>)
    at /home/adam/projects/freenect2/examples/Protonect.cpp:187
```

floe, 2015-12-23:
@madadam is this still an issue? Otherwise, please close.

madadam, 2015-12-23:
Sorry @floe , I don't have a Kinect around anymore so can't test this. I'll close it then. If someone encounters this issue again, they can reopen.

madadam, 2015-12-23:
Actually, I can't close it, as I'm not the original author of this issue. Someone who can, please close this.

closed 2015-12-23 by floe

xlz, 2015-12-30:
The issue of `submiturb failed error -1 errno=12` can be followed in #516.