OpenKinect/libfreenect2 #459: no depth or rgb windows open after running Protonect on Odroid XU4

brant203, 2015-11-08:
I am having a problem where the data windows are not opening to show the images received from the Kinect. I have an Odroid XU4. Here is the output I'm getting: 

```
odroid@odroid:~/test/libfreenect2/libfreenect2/build/bin$ ./Protonect cl
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 006092451747
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0e 03 47 4d 46 31 37 30 2e 31    31332e2e2e2e2e2e474d463137302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  19 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  1b 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 006092451747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] not all subsequences received 513
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Warning] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 17 of 30 packets were lost
```
 No windows open showing the Kinect images.
Thank you

xlz, 2015-11-09:
Can you try again with the latest code?

At least this
> [Warning] [RgbPacketStreamParser] skipping rgb packet!

is a debug message in the latest code.

brant203, 2015-11-09:
I downloaded and installed the code today and after running it I am having the problem I described.

xlz, 2015-11-09:
The latest code does not possibly print the messages you provided.

> [Info] [DepthPacketStreamParser] 17 of 30 packets were lost

This message is "xx packets were lost", there is no longer the "of 30" part.

This is crucial in determining which part is functional. Please post output from the latest code.

brant203, 2015-11-09:
ok, my bad i was pulling from an older version that someone has hosted. now I'm getting errors installing.  when i run cmake .. in my libfreenect2/build/ directory it is saying LibUSB_LIBRARY is NOTFOUND.

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/odroid/test/libfreenect2
```
I have used ```sudo apt-get install libusb-1.0-0-dev``` as well as ```sh install_libusb.sh``` in my /libfreenect2/depends/ directory.

xlz, 2015-11-09:
> `3`. Install libusb. sudo apt-get install libusb-1.0-0-dev. **The version must be >=1.0.20.**

Do you have libusb 1.0.20?

> You may use sudo apt-add-repository ppa:floe/libusb if version 1.0.20 is not available.

brant203, 2015-11-09:
how can i check what version i have? when i do apt-get install libusb-1.0-0-dev it says it is the most recent version

xlz, 2015-11-09:
`dpkg -l libusb-1.0-0-dev`

Also, if you have a local libusb in `depends`, this copy takes precedence.

brant203, 2015-11-09:
```dpkg -l libusb-1.0-0-dev``` says that libusb-1.0-0-d is version 1.0.19. Trying to figure out what version is in my depends folder.
Also I have used: ```sudo apt-add-repository ppa:floe/libusb``` but when i do the install it still says that I have the newest version.

xlz, 2015-11-09:
Have you `apt-get update`?

You probably want to remove `depends/libusb` too

brant203, 2015-11-09:
...i'm dumb. That's going right now I will see what happens. Should I remove whats in my depends after the update?

xlz, 2015-11-09:
Yes you should.

brant203, 2015-11-09:
ok that helped. I was able to build protonect. Here is the output I am getting:
no args:

```
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 006092451747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0e 03 47 4d 46 31 37 30 2e 31    31332e2e2e2e2e2e474d463137302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 006092451747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
odroid@odroid:~/test/libfreenect2/build$ ./bin/Protonect   
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 006092451747
[Info] [Freenect2Impl] found 1 devices
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```
[... 86 lines cut ...]
[Info] [Freenect2DeviceImpl] started
device serial: 006092451747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
```

xlz, 2015-11-09:
OK, first Odroid XU4 does not meet the minimum OpenGL requirement of 3.1. So there is no OpenGL depth processing and no OpenGL viewer too. The following error messages are just OpenGL errors and can be ignored:

```
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
```

The only way you can test it is `./Protonect cl -noviewer`, or `./Protonect cpu -noviewer` and observe framerate information.

This
```
[Error] [OpenCLDepthPacketProcessorImpl] cl::Platform::get failed: -1001
```
looks like a bug.

brant203, 2015-11-09:
ok so it looks like its running when I run: ```./Protonect cl -noviewer``` but I'm getting a lot of this error: ```[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor``` What exactly does this error mean?

xlz, 2015-11-09:
How did you install OpenCL? I see many different instructions on odroid forum.

libfreenect2 has got a new cl.hpp from khronos.org. Where did you get your cl.hpp before?

brant203, 2015-11-09:
I have tried a lot of different things and eventually broke xorg and stuff and needed to reinstall. I was able to find this: http://forum.odroid.com/viewtopic.php?f=95&t=16149.
There is an image that I used and it had it all on there already. 
There is a command guide for going from scratch. It looks like he does this:

```
sudo apt-get install opencl-headers
sudo rm /usr/include/CL/cl.hpp
sudo cp odroid-development/misc/cl.hpp /usr/include/CL
```
so I think it is in a local file already.
how can i get it from khronos.org?

xlz, 2015-11-09:
Please post the output of `find /usr | grep OpenCL`. I suspect it is linking problem.

xlz, 2015-11-09:
Also post the output of `dpkg -l | grep opencl`.

xlz, 2015-11-09:
Now this is uncharted water. There is no really good documentation about setting up OpenCL on Odroid XU4. Let's figure out this properly

Also the output of 
```
ls -l /usr/lib/arm-linux-gnueabihf/libOpenCL.so*
ls -l /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so
ldd /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so
```

The correct method is this
```
echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd
```
And have `ocl-icd-opencl-dev` installed and let it load this for you.

I'm trying to determine whether `/usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so` is the actual ICD so the above queries for you are necessary.

brant203, 2015-11-09:
```find /usr | grep OpenCL```
output:

```
/usr/lib/arm-linux-gnueabihf/libOpenCL.so.1.0.0
/usr/lib/arm-linux-gnueabihf/pkgconfig/OpenCL.pc
/usr/lib/arm-linux-gnueabihf/libOpenCL.so.1
/usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so.1
/usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so.1.1
/usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so
/usr/lib/arm-linux-gnueabihf/libOpenCL.so
/usr/share/doc/ocl-icd-libopencl1/html/libOpenCL.html
/usr/share/man/man7/libOpenCL.7.gz
/usr/share/man/man7/libOpenCL.so.7.gz
```
```dpkg -l | grep opencl```
output:

```
ii  ocl-icd-libopencl1:armhf               2.2.3-1                                    armhf        Generic OpenCL ICD Loader
ii  ocl-icd-opencl-dev:armhf               2.2.3-1                                    armhf        OpenCL development files
ii  opencl-headers                         1.2-svn26009-1                             all          OpenCL (Open Computing Language) header files
```

```ls -l /usr/lib/arm-linux-gnueabihf/libOpenCL.so*```
output:

```
lrwxrwxrwx 1 root root    18 Oct 25  2014 /usr/lib/arm-linux-gnueabihf/libOpenCL.so -> libOpenCL.so.1.0.0
lrwxrwxrwx 1 root root    18 Oct 25  2014 /usr/lib/arm-linux-gnueabihf/libOpenCL.so.1 -> libOpenCL.so.1.0.0
-rw-r--r-- 1 root root 22128 Oct 25  2014 /usr/lib/arm-linux-gnueabihf/libOpenCL.so.1.0.0
```

```ls -l /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so```
output:

```
-rwxr-xr-x 1 root root 2800 May  8  2015 /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so
```

```ldd /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so```
output:

```
	libmali.so => /usr/lib/arm-linux-gnueabihf/mali-egl/libmali.so (0xb5e2d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb5e19000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb5d2b000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb5d15000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb5d02000)
	libX11.so.6 => /usr/lib/arm-linux-gnueabihf/libX11.so.6 (0xb5c14000)
	libX11-xcb.so.1 => /usr/lib/arm-linux-gnueabihf/libX11-xcb.so.1 (0xb5c02000)
	libdrm.so.2 => /usr/lib/arm-linux-gnueabihf/libdrm.so.2 (0xb5be9000)
	libxcb.so.1 => /usr/lib/arm-linux-gnueabihf/libxcb.so.1 (0xb5bcf000)
	libxcb-dri2.so.0 => /usr/lib/arm-linux-gnueabihf/libxcb-dri2.so.0 (0xb5bc4000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb5b0a000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb5a95000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5a6c000)
	/lib/ld-linux-armhf.so.3 (0xb6f8c000)
	libXau.so.6 => /usr/lib/arm-linux-gnueabihf/libXau.so.6 (0xb5a61000)
	libXdmcp.so.6 => /usr/lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb5a55000)
```

xlz, 2015-11-09:
`echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd`? Does it work?

Or `echo /usr/lib/arm-linux-gnueabihf/mali-egl/libmali.so >/etc/OpenCL/vendors/mali-arm7hf.icd`?

brant203, 2015-11-09:
no /etc/OpenCL is not a directory

xlz, 2015-11-09:
create it

brant203, 2015-11-09:
Ok I made the directory /OpenCL/vendors/ and then did the command ```echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd```
Will I need to rebuild Protonect before it works?

xlz, 2015-11-09:
no

brant203, 2015-11-09:
ok, here is the output that I am getting after running ```echo /usr/lib/arm-linux-gnueabihf/mali-egl/libmali.so >/etc/OpenCL/vendors/mali-arm7hf.icd```

```
odroid@odroid:~/test/libfreenect2/build$ ./bin/Protonect cl
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:6 with serial 006092451747
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0e 03 47 4d 46 31 37 30 2e 31    31332e2e2e2e2e2e474d463137302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 006092451747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
```

 Should I be using ```./Protonect cl -noviewer```

xlz, 2015-11-09:
> The only way you can test it is ./Protonect cl -noviewer, or ./Protonect cpu -noviewer and observe framerate information.

xlz, 2015-11-09:
If you can see framerate information, it is programmatically correct. Using ROS image viewer or your custom OpenCV viewer outside of libfreenect2 with the processed images should visualize it.

xlz, 2015-11-09:
Can you also provide the output using `echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd`? I want to figure out what is libOpenCL.so exactly.

brant203, 2015-11-09:
```./Protonect cl -noviewer```
output:

```
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:8 with serial 006092451747
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0e 03 47 4d 46 31 37 30 2e 31    31332e2e2e2e2e2e474d463137302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 006092451747
device firmware: 4.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] not all subsequences received 897
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
MESA-LOADER: malformed or no PCI ID
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 19 packets were lost
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
^C[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```
there is no output for ```echo /usr/lib/arm-linux-gnueabihf/mali-egl/libOpenCL.so >/etc/OpenCL/vendors/mali-arm7hf.icd```

xlz, 2015-11-09:
It seems to be working? You can run it a while longer.

brant203, 2015-11-09:
how can I tell if the depth data is working?
the frame data does not say if its rgb or depth.

```
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 71.0396ms -> ~14.0767Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [DepthPacketStreamParser] 31 packets were lost
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[... 421 lines cut ...]
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
odroid@odroid:~/test/libfreenect2/build$ ^C
```

xlz, 2015-11-09:
```
[Info] [TurboJpegRgbPacketProcessor] avg. time: 71.0396ms -> ~14.0767Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 70.2682ms -> ~14.2312Hz
```

brant203, 2015-11-10:
OK, so this is working correctly?! I have some not issue related questions that I would like to ask you. Is there a good way for me to contact you about them?

xlz, 2015-11-10:
Can you post it here.

zaccaro, 2015-11-23:
I'm not really sure but maybe he could improve the framerate by skipping some not needed registration.
For example if only needed depth, change

 ``` libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);```

to 
  ```libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Depth);```

floe, 2015-12-23:
Still an issue? Otherwise, please close.

CrazyPopLin, 2015-12-28:
Hi @xlz ,
>If you can see framerate information, it is programmatically correct. Using ROS image viewer or your >custom OpenCV viewer outside of libfreenect2 with the processed images should visualize it.
As you said, if the framerate correct ,can I use iai_kinect2 with opencl to see the data?
Thanks.

xlz, 2015-12-28:
Yes

closed 2016-01-21 by xlz