OpenKinect/libfreenect2 #464: Is there anyway to get the registered RGB image without black areas showed in Protonect by default?

ZhilanZhou, 2015-11-09:
I just started to use libfreenect2 and am kind of wondering how to get two separate RGB and depth images which are registered. Is the undistorted in Registration::apply the registered depth image?

Thanks.

floe, 2015-11-09:
No, `undistorted` is just the rectified depth image. `registered` is the color image mapped onto depth. Optionally, if you supply a `Frame(1920,1080+2,4)` for the `bigdepth` pointer, you can also get depth mapped onto color.

ZhilanZhou, 2015-11-09:
Thanks! What I get via Protonect for registered is a color image with some areas black(same black areas in depth image I suppose), how can I get those areas showing just in normal colors like raw color image? (not sure if my expression is clear enough though)

floe, 2015-11-10:
Well, if there is no depth data in some areas, then it is also not possible to provide a color mapping, because you need to know the distance for each pixel for the mapping to work. It kind of depends on your use case, sometimes the `bigdepth` image (basically depth upscaled to color) can be more suitable.

ZhilanZhou, 2015-11-16:
Thanks for the explanation! Sorry forgot to close this

closed 2015-11-16 by ZhilanZhou