OpenKinect/libfreenect2 #468: Problem compilation on OSX 10.11

supermeichel, 2015-11-15:
Hi, i try to install libfreenect2 on OSX 10.11
The kinect 2 power ON and after the sytem crash :(
Do you have an issue for this problem ?
Thanks !
Rey

----------

MacBook-Pro-de-Supermichel:build REY$ cmake ..
-- using tinythread as threading library
-- RPATH set to /opt/local/lib
-- Linking with these libraries: 
 /opt/local/lib/libusb-1.0.dylib
 /usr/local/opt/jpeg-turbo/lib/libturbojpeg.dylib
 pthread
 /usr/local/Cellar/glfw3/3.1.2/lib/libglfw3.dylib
 /System/Library/Frameworks/OpenGL.framework
 /System/Library/Frameworks/OpenCL.framework
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   freenect2

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/REY/libfreenect2/libfreenect2/build
MacBook-Pro-de-Supermichel:build REY$ make
[  6%] Built target generate_resources_tool
[ 86%] Built target freenect2
[100%] Built target Protonect
MacBook-Pro-de-Supermichel:build REY$ make install
[  6%] Built target generate_resources_tool
[ 86%] Built target freenect2
[100%] Built target Protonect
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/lib/libfreenect2.dylib
-- Up-to-date: /usr/local/include/libfreenect2
-- Up-to-date: /usr/local/include/libfreenect2/depth_packet_processor.h
-- Up-to-date: /usr/local/include/libfreenect2/frame_listener.hpp
-- Up-to-date: /usr/local/include/libfreenect2/frame_listener_impl.h
-- Up-to-date: /usr/local/include/libfreenect2/libfreenect2.hpp
-- Up-to-date: /usr/local/include/libfreenect2/logger.h
-- Up-to-date: /usr/local/include/libfreenect2/packet_pipeline.h
-- Up-to-date: /usr/local/include/libfreenect2/packet_processor.h
-- Up-to-date: /usr/local/include/libfreenect2/registration.h
-- Up-to-date: /usr/local/include/libfreenect2/rgb_packet_processor.h
-- Up-to-date: /usr/local/include/libfreenect2
-- Up-to-date: /usr/local/include/libfreenect2/config.h
-- Up-to-date: /usr/local/include/libfreenect2/export.h
-- Up-to-date: /usr/local/lib/cmake/freenect2/freenect2Config.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/freenect2.pc
MacBook-Pro-de-Supermichel:build REY$ ./bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:10 with serial 018326145047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 11 14 47 4b 53 38 39 35 2e 31    31332e2e2e2e2e2e474b533839352e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 018326145047
device firmware: 2.3.3913.0.7
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 1016
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.1488ms -> ~76.0523Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.2297ms -> ~75.5874Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.0746ms -> ~76.4843Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.2516ms -> ~75.4628Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 418 packets were lost
Segmentation fault: 11

supermichel, 2015-11-15:
When i use this command it's look good :
./bin/Protonect cl

floe, 2015-11-15:
Do you have a laptop with discrete NVidia graphics card? If yes, then this is a known bug, the crash happens inside the video driver and can't be fixed. You can either use OpenCL or force the internal Intel GPU as workarounds.

xlz, 2015-11-16:
According to your provided log, it is not a problem with compilation, and it did not cause your system to crash.

In the case of the segmentation fault, you can provide a stack trace by running Protonect under a debugger, and then the problem can be narrowed down. After you get the stack trace, post it here or compare it with #205.

closed 2015-12-23 by floe