OpenKinect/libfreenect2 #471: libfreenect2 , running Protonect on windows10 errors LIBUSB_ERROR_NOT_SUPPORTED

Byeayum, 2015-11-17:
I am running  Protonect on windows10,but,there are many errors .   could  you give me some advise?

Here is running output:

Environment variables: LOGFILE=<protonect.log>

Usage: C:\Users\user\Desktop\libfreenect2-master\examples\bin\RelWithDebInfo\Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]

[Info] [Freenect2Impl] enumerating devices...

[Info] [Freenect2Impl] 10 usb devices connected

[Info] [Freenect2Impl] found valid Kinect v2 @3:5 with serial 506417542542

[Info] [Freenect2Impl] found 1 devices

[Info] [Freenect2DeviceImpl] opening...

[Info] [Freenect2DeviceImpl] opened

[Info] [Freenect2DeviceImpl] starting...

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...

[Info] [Freenect2DeviceImpl] submitting usb transfers...

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[... 66 lines cut ...]
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

[Error] [usb::TransferPool] all submissions failed. Try debugging with environment variable: LIBUSB_DEBUG=4.

[Info] [Freenect2DeviceImpl] started

device serial: 506417542542

device firmware: 4.3.3914.0.7

[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.9522ms -> ~50.1197Hz

[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.9182ms -> ~50.2053Hz

[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.228ms -> ~49.4364Hz

[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.9035ms -> ~50.2425Hz

floe, 2015-11-17:
Please do what the error message suggests:

    [Error] [usb::TransferPool] all submissions failed. Try debugging with environment variable: LIBUSB_DEBUG=4

Byeayum, 2015-11-18:
Environment variables: LOGFILE=<protonect.log>
Usage: Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000016b4] libusb: debug [libusb_init] created default context
[ 0.000000] [000016b4] libusb: debug [libusb_init] libusb v1.0.20.11014
[ 0.000000] [000016b4] libusb: debug [windows_init] Windows 8 64-bit
[ 0.000000] [000016b4] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.000000] [000016b4] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.000000] [000016b4] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.000000] [000016b4] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [000016b4] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [000016b4] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [000016b4] libusb: debug [windows_init] hires timer available (Frequency: 3330090 Hz)
[ 0.000000] [000016b4] libusb: debug [windows_init] timer thread will run on core #0
[ 0.000000] [000016b4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [000016b4] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [000016b4] libusb: debug [libusb_get_device_list] 
[ 0.000000] [000025f0] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.000000] [000025f0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.000000] [000025f0] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.000000] [000025f0] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [29D]
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [E2]
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [2EA]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [337]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): usbhub
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [3C5]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): usbhub
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [2E8]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): usbhub
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [165]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [159]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [333]
[ 0.009981] [000016b4] libusb: debug [get_api_type] driver(s): usbhub
[ 0.009981] [000016b4] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [3FB]
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [383]
[ 0.009981] [000016b4] libusb: debug [init_device] got bus number from ancestor #3
[ 0.009981] [000016b4] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009981] [000016b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.009981] [000016b4] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 7): '\\.\USB#VID_046D&PID_C31D#6&182A12DA&0&7'
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] extra GUID: {8F74E4B3-271A-4409-99BA-14E5E85D9E9E}
[ 0.009981] [000016b4] libusb: debug [windows_get_device_list] allocating new device for session [58]
[ 0.009981] [000016b4] libusb: debug [init_device] got bus number from ancestor #3
[ 0.009981] [000016b4] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.009981] [000016b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 214 bytes)
[ 0.009981] [000016b4] libusb: debug [init_device] (bus: 3, addr: 14, depth: 1, port: 1): '\\.\USB#VID_045E&PID_02D8#506417542542'
[ 0.020097] [000016b4] libusb: debug [windows_get_device_list] found existing device for session [2E8] (0.0)
[ 0.020097] [000016b4] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB20#4&14207F40&0'
[ 0.021095] [000016b4] libusb: debug [windows_get_device_list] found existing device for session [165] (0.0)
[... cut ...]
[ 0.383415] [000025f0] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 0.383415] [000016b4] libusb: debug [libusb_free_transfer] transfer 000000CA6BB3A330
[ 0.383415] [000025f0] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.383415] [000025f0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.383415] [000025f0] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.383415] [000025f0] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms

Byeayum, 2015-11-18:
The output never be stop!  and Kinect's indicator is not on. the window never shows anything。

xlz, 2015-11-18:
The above log does not include any errors reported here.

If you build libusb correctly, you will see "debug [winusbx_submit_iso_transfer] matched endp
oint" in the output.

If you do not see the message in the debug output, you failed to build libusb correctly and this should be closed as duplicate of #448.

If you do see the message in the debut output, we can continue.

xlz, 2015-11-18:
I built some binary here https://github.com/xlz/libusb/releases/tag/v1.0.20-winiso since you were unable to build it yourself.

xlz, 2015-11-18:
The user confirmed in private email (why?) that he failed to correctly build libusb and the binary build provided above solved his problem.

Close this.

floe, 2015-11-18:
Sigh. Had the same experience recently, we should make it policy to not respond to non-Github inquiries.

closed 2015-11-18 by floe

JoshBlake, 2015-11-18:
I receive a lot of email inquiries on libfreenect2 (probably since my email is listed in the readme). Maybe 75% of them are basic things, and the rest are more intricate things I'm not as familiar with as you guys, but I universally try to recommend they post a GitHub issue, after responding to the immediate simple query.

huatson, 2016-04-21:
@xlz @Byeayum that happens because is not applied the Windows isochronous modifications <Joshua Blake> need to apply this one https://github.com/JoshBlake/libusbx/tree/winiso

cuisinart, 2016-12-29:
Hello, I am getting this same error with the latest sources from github.  Libusb from here, which appears to be the official, and actively maintained release:
https://github.com/libusb/libusb
builds out of the box (amazing!)
But protonect gives:
`[code]D:\projects\SourceCodingLibraries\libfreenect2\MSVCBuild\bin\Debug>protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 21 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @9:4 with serial 103673135247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Error] [usb::TransferPool] all submissions failed. Try debugging with environment variable: LIBUSB_DEBUG=3.
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
[libusb: error [windows_handle_events] could not find a matching transfer for fd 2
Ilibusb: error [handle_events] backend handle_events failed with error -5
nlibusb: error [windows_handle_events] could not find a matching transfer for fd 2
flibusb: error [handle_events] backend handle_events failed with error -5
olibusb: error [windows_handle_events] could not find a matching transfer for fd 2
]libusb: error [handle_events] backend handle_events failed with error -5
 libusb: error [windows_handle_events] could not find a matching transfer for fd 2
[libusb: error [handle_events] backend handle_events failed with error -5
Flibusb: error [windows_handle_events] could not find a matching transfer for fd 2
rlibusb: error [handle_events] backend handle_events failed with error -5
elibusb: error [windows_handle_events] could not find a matching transfer for fd 2
elibusb: error [handle_events] backend handle_events failed with error -5
nlibusb: error [windows_handle_events] could not find a matching transfer for fd 2
elibusb: error [handle_events] backend handle_events failed with error -5
clibusb: error [windows_handle_events] could not find a matching transfer for fd 2
tlibusb: error [handle_events] backend handle_events failed with error -5
2libusb: error [windows_handle_events] could not find a matching transfer for fd 2
Dlibusb: error [handle_events] backend handle_events failed with error -5
elibusb: error [windows_handle_events] could not find a matching transfer for fd 2
vlibusb: error [handle_events] backend handle_events failed with error -5
ilibusb: error [windows_handle_events] could not find a matching transfer for fd 2
clibusb: error [handle_events] backend handle_events failed with error -5
elibusb: error [windows_handle_events] could not find a matching transfer for fd 2
Ilibusb: error [handle_events] backend handle_events failed with error -5
mlibusb: error [windows_handle_events] could not find a matching transfer for fd 2
plibusb: error [handle_events] backend handle_events failed with error -5
llibusb: error [windows_handle_events] could not find a matching transfer for fd 2
[... cut ...]
D:\projects\SourceCodingLibraries\libfreenect2\MSVCBuild\bin\Debug>%LIBUSB_DEBUG%
'3' is not recognized as an internal or external command,
operable program or batch file.

D:\projects\SourceCodingLibraries\libfreenect2\MSVCBuild\bin\Debug>
[/code]`

You can see that my environment variable LIBUSB_DEBUG is set to 3 at the end.  It seems to me that libfreenect2 should be working with the maintained main-line libusb code base as linked above, and if it's not, it needs to be noted what should be done in the use and building description of libfreenect2.

cuisinart, 2016-12-29:
debug level 4:
`D:\projects\SourceCodingLibraries\libfreenect2\MSVCBuild\bin\Debug>protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00003904] libusb: debug [libusb_init] created default context
[ 0.001000] [00003904] libusb: debug [libusb_init] libusb v1.0.21.11157
[ 0.001499] [00003904] libusb: debug [windows_init] Windows 8 32 bit
[ 0.002000] [00003904] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.003000] [00003904] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.003000] [00003904] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.004000] [00003904] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.004000] [00003904] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.004500] [00003904] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.006000] [00003904] libusb: debug [windows_init_clock] hires timer available (Frequency: 3201536 Hz)
[ 0.007500] [00003904] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.008000] [00003904] libusb: debug [htab_create] using 1021 entries hash table
[ 0.008000] [00003904] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[Info[ 0.009001] [000030f0] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
][ 0.009001] [000030f0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
 [ 0.009500] [000030f0] libusb: debug [handle_events] poll fds modified, reallocating
[[ 0.009500] [000030f0] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
Freenect2Impl] enumerating devices...
[ 0.011001] [00003904] libusb: debug [libusb_get_device_list]
[ 0.011501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [286]
[ 0.011501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [1E1]
[ 0.012001] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [3BF]
[ 0.012501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [35B]
[ 0.012501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [66]
[ 0.013001] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [6F]
[ 0.013501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [16E]
[ 0.013501] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [177]
[ 0.014001] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [2AA]
[ 0.015001] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.017502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.018002] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [105]
[ 0.018002] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.018502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.018502] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [336]
[ 0.019002] [00003904] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.019502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.019502] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [263]
[ 0.020002] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.020502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.020502] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [156]
[ 0.021002] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.021502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.022003] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [217]
[ 0.022003] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.022503] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.023002] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [3CD]
[ 0.023002] [00003904] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.023502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.024002] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [90]
[ 0.025502] [00003904] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.026502] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.027003] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [25F]
[ 0.027503] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[ 0.027503] [00003904] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.028003] [00003904] libusb: debug [windows_get_device_list] allocating new device for session [2EC]
[ 0.028503] [00003904] libusb: debug [get_api_type] driver(s): usbhub
[... cut ...]
[ 1.498689] [000030f0] libusb: error [handle_events] backend handle_events failed with error -5
[ 1.499189] [00003904] libusb: debug [libusb_exit]
[ 1.499689] [00003904] libusb: debug [libusb_exit] destroying default context
[ 1.499689] [00003904] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 1.500189] [00003904] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 1.500689] [00003b6c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

D:\projects\SourceCodingLibraries\libfreenect2\MSVCBuild\bin\Debug>`

fred-dev, 2017-12-19:
Did you ever find a solution to this? I am facing similar problems.

cuisinart, 2017-12-19:
I went down the rabbit hole of looking into different libusb builds which support different USB transfer modes, but never came to a working solution.  I ended up using the microsoft kinect2 driver and it's working well, with very low CPU resources required.