OpenKinect/libfreenect2 #475: I can not run two kinect at  the same time on windows 10

Byeayum, 2015-11-19:
is there something necessary for it ?  It seems to only the first kinect works correctly.

xlz, 2015-11-19:
Your bug report does not contain any information to work with. Take a look at this https://issues.apache.org/bugwritinghelp.html

Byeayum, 2015-11-20:
I connected two Kinect on the computer, when I run the program but only a Kinect image will appear.

here is output:

Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\user\Desktop\libfreenect2\examples\bin\RelWithDebInfo\Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:13 with serial 004897751147
[Info] [Freenect2Impl] found valid Kinect v2 @3:12 with serial 506417542542
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 004897751147
device firmware: 2.3.3913.0.7
[Info] [DepthPacketStreamParser] 9 packets were lost
[Info] [DepthPacketStreamParser] 6 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 26.1124ms -> ~38.296Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 21.9416ms -> ~45.5755Hz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.7235ms -> ~42.1523Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 16.2979ms -> ~61.3577Hz
[Info] [DepthPacketStreamParser] 75 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.6919ms -> ~48.328Hz
[Info] [DepthPacketStreamParser] 3 packets were lost
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 5 packets were lost

xlz, 2015-11-20:
Works as intended. Not a bug.

Byeayum, 2015-11-20:
yeah... but why i can only see one kinect  works?

xlz, 2015-11-20:
Because you used only one.

Byeayum, 2015-11-20:
NO，i have conected two kinect on the computer ，the output also hava found two kinect and opened.

[Info] [Freenect2Impl] found valid Kinect v2 @3:13 with serial 004897751147
[Info] [Freenect2Impl] found valid Kinect v2 @3:12 with serial 506417542542
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started

xlz, 2015-11-20:
Protonect is never designed to use two Kinects at once.

Byeayum, 2015-11-20:
so what should do for running two kinect on one computer at the same time?

xlz, 2015-11-20:
Have you tried to run two Protonect's?

Byeayum, 2015-11-20:
yeah  but when i  do  it , there is  same  errors and the two kinect both can not running .
 the output is:
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\user\Desktop\libfreenect2\examples\bin\RelWithDebInfo\Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:13 with serial 004897751147
[Info] [Freenect2Impl] found valid Kinect v2 @3:12 with serial 506417542542
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 004897751147
device firmware: 2.3.3913.0.7
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 32 packets were lost
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 32 packets were lost
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 32 packets were lost
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 50 packets were lost
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!

xlz, 2015-11-20:
Have you specified serial number for Protonect?

Byeayum, 2015-11-20:
NO ,how to do that?

xlz, 2015-11-20:
Have you read the first two lines of the output of Protonect?

Byeayum, 2015-11-20:
Yes， i have  read it , but I don't know what it means.

Byeayum, 2015-11-20:
and what shoud i do to let two kinect work well?

xlz, 2015-11-20:
It says `Usage: Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]`. How do I explain the meaning of this.

Perhaps you should use `Protonect.exe 004897751147` and `Protonect.exe 506417542542`?

Byeayum, 2015-11-20:
yeah it works，thanks a lot .

closed 2015-11-21 by floe