OpenKinect/libfreenect2 #478: link of GLFW3 static libraries in cmake

a-cvision, 2015-11-20:
In the current version, the glfw3 static libraries are not linked.. this could cost some time for a lot of people as "make" will fail as such.

Perhaps someone could correct that in the cmake file.

Thanks

xlz, 2015-11-20:
Not linked to libfreenect2.so or Protonect?

a-cvision, 2015-11-22:
I don't have a very good know-how with the whole cmake -make toolchain. But I had to add $GLFW3_STATIC_LIBRARIES in the append libraries part of the cmake file in order for "make" to work. Otherwise it resulted in a bunch of unknown references to XRR_..

I apologize if it is a particular case for me. Though I can't imagine why that would happen just for me.

xlz, 2015-11-22:
That means you didn't install OpenGL dependencies. mesa-common-dev freeglut3-dev libxrandr-dev libxi-dev

a-cvision, 2015-11-23:
I checked that and they are installed, otherwise I wouldn't be able to link to the static libraries. But certainly this seems to be a strange particular case, thanks for the comments.

closed 2015-11-23 by a-cvision