OpenKinect/libfreenect2 #480: Cannot get Protonect Viewer running on the Tegra X1

bigcmos, 2015-11-25:
I'm trying to get libfreenect2 working on the Nvidia Tegra X1 and I'm having issues displaying anything from the viewer.  The main viewer window opens, but it is blank.  The log output seems to indicate that the TurboJpegRgbPacketProcessor is running at 20Hz.

Should I expect that this should work on the tegra X1 (or K1)?  let me know if there is any other output I can give you to help diagnose.

**Protonect Output**
```
$ ./bin/Protonect 
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @10:4 with serial 068176734547
[Info] [Freenect2Impl] found 1 devices
freenect2.openDevice.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 03 11 17 47 48 57 31 32 35 2e 31    31332e2e2e2e2e2e4748573132352e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 068176734547
device firmware: 1.1.3326.0.7
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[... 101 lines cut ...]
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!

```

**Host Details**
```
$ uname -a
Linux tegra-ubuntu 3.10.67-g3a5c467 #1 SMP PREEMPT Fri Oct 30 14:26:17 PDT 2015 aarch64 aarch64 aarch64 GNU/Linux
$ head -1 /etc/nv_tegra_release 
# R23 (release), REVISION: 1.1, GCID: 6220086, BOARD: t210ref, EABI: hard, DATE: Fri Oct 30 21:33:12 UTC 2015
```

xlz, 2015-11-25:
Have you enabled USB 3.0 and disabled USB autosuspend?

Also, `LIBUSB_DEBUG=4 ./Protonect` and `dmesg`.

xlz, 2015-11-25:
May be related: #481

Things to do:

1. `git checkout v0.1-rc1`, rebuild, and run it
2. `gdb ./Protonect`, `run`, Ctrl-C, `thread apply all backtrace` and `bt full`

bigcmos, 2015-11-30:
[jetson_x1_dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/47735/jetson_x1_dmesg.txt)
[jetson_x1_protonect.txt](https://github.com/OpenKinect/libfreenect2/files/47736/jetson_x1_protonect.txt)

yes.

usb_port_owner_info=2
usbcore.autosuspend=-1
```
$ cat /boot/extlinux/extlinux.conf | egrep "usb_port_owner_info|usbcore.autosuspend"
      APPEND fbcon=map:0 console=tty0 console=ttyS0,115200n8 androidboot.modem=none androidboot.serialno=P2180A00P00940c003fd androidboot.security=non-secure tegraid=21.1.2.0.0 ddr_die=2048M@2048M ddr_die=2048M@4096M section=256M memtype=0 vpr_resize usb_port_owner_info=2 lane_owner_info=0 emc_max_dvfs=0 touch_id=0@63 video=tegrafb no_console_suspend=1 debug_uartport=lsport,0 earlyprintk=uart8250-32bit,0x70006000 maxcpus=4 usbcore.old_scheme_first=1 usbcore.autosuspend=-1 lp0_vec=${lp0_vec} nvdumper_reserved=${nvdumper_reserved} core_edp_mv=1125 core_edp_ma=4000 gpt android.kerneltype=normal androidboot.touch_vendor_id=0 androidboot.touch_panel_id=63 androidboot.touch_feature=0 androidboot.bootreason=pmc:software_reset,pmic:0x0 root=/dev/mmcblk0p1 rw rootwait

$ grep . /sys/bus/usb/devices/*/power/autosuspend
/sys/bus/usb/devices/10-1/power/autosuspend:-1
/sys/bus/usb/devices/10-2.1/power/autosuspend:-1
/sys/bus/usb/devices/10-2/power/autosuspend:0
/sys/bus/usb/devices/7-1.5/power/autosuspend:-1
/sys/bus/usb/devices/7-1.6/power/autosuspend:-1
/sys/bus/usb/devices/7-1/power/autosuspend:0
/sys/bus/usb/devices/9-3/power/autosuspend:0
/sys/bus/usb/devices/usb10/power/autosuspend:0
/sys/bus/usb/devices/usb1/power/autosuspend:0
/sys/bus/usb/devices/usb2/power/autosuspend:0
/sys/bus/usb/devices/usb3/power/autosuspend:0
/sys/bus/usb/devices/usb4/power/autosuspend:0
/sys/bus/usb/devices/usb5/power/autosuspend:0
/sys/bus/usb/devices/usb6/power/autosuspend:0
/sys/bus/usb/devices/usb7/power/autosuspend:0
/sys/bus/usb/devices/usb8/power/autosuspend:0
/sys/bus/usb/devices/usb9/power/autosuspend:0
```

bigcmos, 2015-11-30:
```
$ ldd ./Protonect | grep libusb
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xf70d1000)
```

```
$ gdb ./Protonect 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Protonect...done.
(gdb) run
Starting program: /home/acme/dev/archive/libfreenect2/build/bin/Protonect 
Failed to read a valid object file image from memory.
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Environment variables: LOGFILE=<protonect.log>
Usage: /home/acme/dev/archive/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[New Thread 0xf5c2d440 (LWP 3945)]
[New Thread 0xf542d440 (LWP 3946)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @10:6 with serial 068176734547
[Info] [Freenect2Impl] found 1 devices
[New Thread 0xefc18440 (LWP 3947)]
[New Thread 0xef418440 (LWP 3948)]
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 03 11 17 47 48 57 31 32 35 2e 31    31332e2e2e2e2e2e4748573132352e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 068176734547
device firmware: 1.1.3326.0.7
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[... 218 lines cut ...]
        viewer = {<WithOpenGLBindings> = {_vptr.WithOpenGLBindings = 0xebd0 <vtable for Viewer+8>, bindings = 0x2efb10}, 
          shouldStop = false, window = 0x1e7ba0, triangle_vbo = 873984, triangle_vao = 4, renderShader = {<WithOpenGLBindings> = {
              _vptr.WithOpenGLBindings = 0xe9c8 <vtable for ShaderProgram+8>, bindings = 0x2efb10}, program = 3, vertex_shader = 1, 
            fragment_shader = 2, 
---Type <return> to continue, or q <return> to quit---
            error_buffer = "\005\000\000\000\002\000\000\000\004V\r\000\004V\r\000l2\000\000\004\000\000\000\000\000\000\000\004\000\000\000\004\000\000\000\352\000\000\000\001\000\000\000\003\004\000\000\314\026\016\000̖\r\000\b", '\000' <repeats 11 times>, "\004\000\000\000\000\000\000\000/home/acme/dev/archive/libfreenect2/build/bin/Protonect\000Ԗ\r\000x", '\000' <repeats 11 times>, "\004\000\000\000\000\000\000\000\t\001\000\000\001\000\000\000\003\000\000\000L\027\016\000L\227\r\000\004\000\000\000!\336\376\377\270\346\376\377)\336\376\377\a\000\000\000A4\000\000\000a"...}, renderGrayShader = {<WithOpenGLBindings> = {
              _vptr.WithOpenGLBindings = 0xe9c8 <vtable for ShaderProgram+8>, bindings = 0x2efb10}, program = 6, vertex_shader = 4, 
            fragment_shader = 5, 
            error_buffer = "?\353\376\377\\\345\376\377`\345\376\377P\000\177\367\003\000\000\000\360\352\376\377\200\351\376\377 (\303\365\240\000\000\000\330\062\306\365@\345\376\377\067\261}\367", '\000' <repeats 174 times>...}, shader_folder = {
            static npos = <optimized out>, 
            _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
              _M_p = 0x2282b4 "src/shader/"}}, frames = {_M_t = {
              _M_impl = {<std::allocator<std::_Rb_tree_node<std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, libfreenect2::Frame*> > >> = {<__gnu_cxx::new_allocator [... cut ...]
          ir = {<WithOpenGLBindings> = {
              _vptr.WithOpenGLBindings = 0xe9f8 <vtable for Texture<ImageFormat<4u, 33326u, 6403u, 5126u> >+8>, 
              bindings = 0x2efb10}, bytes_per_pixel = 4, height = 0, width = 0, texture = 0, data = 0x0, size = 0}}
        binpath = {static npos = <optimized out>, 
          _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
            _M_p = 0x1a3e4 "/home/acme/dev/archive/libfreenect2/build/bin/"}}
```

bigcmos, 2015-11-30:
Added some logging and let it run for a while, looks like I'm not getting the correct footer in the DepthPacketStreamParser (see below).

[Info] [DepthPacketStreamParser] [zdaa] DepthPacketStreamParser::onDataReceived footer in_length: 28312, footer_found: 0, wb.length: 0, wb.capacity: 298496

[Info] [DepthPacketStreamParser] [zdaa] DepthPacketStreamParser::onDataReceived memcpy in_length: 28312, footer_found: 0
6

```
$ git diff src/
diff --git a/src/depth_packet_stream_parser.cpp b/src/depth_packet_stream_parser.cpp
index 4b7e9e7..4c51da6 100644
--- a/src/depth_packet_stream_parser.cpp
+++ b/src/depth_packet_stream_parser.cpp
@@ -79,6 +79,12 @@ void DepthPacketStreamParser::onDataReceived(unsigned char* buffer, size_t in_le
       footer = reinterpret_cast<DepthSubPacketFooter *>(&buffer[in_length]);
       footer_found = true;
     }
+    LOG_INFO << "[zdaa] DepthPacketStreamParser::onDataReceived footer " <<
+        "in_length: " << in_length <<
+        ", footer_found: " << footer_found <<
+        ", wb.length: " << wb.length <<
+        ", wb.capacity: " << wb.capacity <<
+    std::endl;
 
     if(wb.length + in_length > wb.capacity)
     {
@@ -87,6 +93,10 @@ void DepthPacketStreamParser::onDataReceived(unsigned char* buffer, size_t in_le
       return;
     }
 
+    LOG_INFO << "[zdaa] DepthPacketStreamParser::onDataReceived memcpy " <<
+        "in_length: " << in_length <<
+        ", footer_found: " << footer_found <<
+    std::endl;
     memcpy(wb.data + wb.length, buffer, in_length);
     wb.length += in_length;
 
@@ -98,6 +108,12 @@ void DepthPacketStreamParser::onDataReceived(unsigned char* buffer, size_t in_le
       }
       else
       {
+        LOG_INFO << "[zdaa] DepthPacketStreamParser::onDataReceived " <<
+            "in_length: " << in_length <<
+            ", current_sequence_: " << current_sequence_ <<
+            ", footer->sequence: " << footer->sequence <<
+            ", seq eq: (" << (current_sequence_ != footer->sequence) << ")" <<
+            "processor_->ready(): " << processor_->ready() << std::endl;
         if(current_sequence_ != footer->sequence)
         {
           if(current_subsequence_ == 0x3ff)
@@ -112,7 +128,9 @@ void DepthPacketStreamParser::onDataReceived(unsigned char* buffer, size_t in_le
               packet.buffer = buffer_.back().data;
               packet.buffer_length = buffer_.back().length;
 
+              LOG_INFO << "[zdaa] Pre process Length: " << in_length << std::endl;
               processor_->process(packet);
+              LOG_INFO << "[zdaa] post process Length: " << in_length << std::endl;
 
               processed_packets_++;
               if (processed_packets_ == 0)
```

xlz, 2015-12-01:
28312 is strange. It's far from a round number.  There are a lot of these `tegra-xhci tegra-xhci: WARN Event TRB for slot 4 ep 8 with no TDs queued?` in dmesg, this is also not normal. Also, libusb has `[handle_iso_completion] low-level USB error -18`. 18 is EXDEV, meaning iso transfers are partially completed or something.

Things to do:
* Post `lspci -nn`.
* Verify if libusb is 1.0.20. If you know it is, no action is needed.
* Try another way to set autosuspend: `sudo sh -c 'for dev in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$dev; done'`
* Try maximizing CPU and GPU performance http://elinux.org/Jetson/Performance (I don't know if this applies to X1). Maybe it was too slow to handle the rate of data.
* Try newer kernel: http://elinux.org/Jetson_TX1. Maybe it's a driver issue. This takes a lot of time to figure out so you don't have to do it.

floe, 2015-12-23:
Still an issue? Otherwise, please close.

soulslicer, 2015-12-23:
Can I know if anyone got libfreenect2 to work with tegra?

xlz, 2015-12-24:
It works on Tegra K1. I don't have a Tegra X1 so no idea about it.

soulslicer, 2015-12-25:
Does it work with OpenCL or does it run on the CPU?

xlz, 2015-12-25:
There is no OpenCL support for Tegra K1. It works with OpenGL.

bigcmos, 2015-12-26:
Still an issue.  

> Things to do:
> 
> - Post lspci -nn.

Will send this when I'm back in the office in Jan.

> Verify if libusb is 1.0.20. If you know it is, no action is needed.

Yes, using 1.0.20.

> - Try another way to set autosuspend: sudo sh -c 'for dev in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$dev; done'

Yes, I tried this the first time, but had to fix it every time I plugged/unplugged the USB.  Is there a good way to test if autosuspend is working?

> - Try maximizing CPU and GPU performance http://elinux.org/Jetson/Performance (I don't know if this applies to X1). Maybe it was too slow to handle the rate of data.

I don't think it is too slow.  I was able to run 3 separate USB cameras @ 40 Hz with no problem.

> - Try newer kernel: http://elinux.org/Jetson_TX1. Maybe it's a driver issue. This takes a lot of time to figure out so you don't have to do it.

I'm at the newest supplied by Nvidia.  I would have to build a new Kernel.  Do you have an idea of what kernel people are using on the TK1 that is succeeding?

> 
> @xlz

hanyazou, 2016-01-31:
I bought my TK1 and setup today. And Protonect prints many 'skipping rgb packet!' message without any image.

>[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!

I found that ./bin/Protonect uses /usr/lib/arm-linux-gnueabihf/mesa/libGL.so instread of tegra/libGL.so.

My setup steps:

1. create new VM on my Mac and install Ubuntu 14.04
2. install JetPack L4T 2.0 following http://docs.nvidia.com/jetpack-l4t/index.html#developertools/mobile/jetpack/jetpack_l4t/2.0/jetpack_l4t_install.htm
3. build libfreenect2 following https://github.com/OpenKinect/libfreenect2/blob/master/README.md#debianubuntu-1404

I found that libgl1-mesa-dev package overwrite /usr/lib/arm-linux-gnuapihf/libGL.so -> tegra/libGL.so symbolic link. I'm not sure which step introduced the libgl1-mesa-dev package. I guess 'apt-get install mesa-common-dev' in the libfreenect2 build instruction depends the libgl1-mesa-dev.

@bigcmos please check your libGL.so symlink in the directory.

xlz, 2016-01-31:
The overwrite is an old known issue. See the top sticky post https://devtalk.nvidia.com/default/board/162/ Notice On apt-get upgrade: libglx.so Corruption

hanyazou, 2016-01-31:
@xlz http://elinux.org/Jetson_TK1 give an image that 'sudo apt-mark hold xserver-xorg-core' can prevent that issue but can't...

xlz, 2016-01-31:
`/etc/nv_release` contains a list of Tegra proprietary files and their hashes. Compare them and see if any is bad.

hanyazou, 2016-01-31:
I have very similar output message and no image on my TK1.

attached is output of these commands.

$ git checkout v0.1.1
$ rm -rf build
$ mkdir build && cd build
$ cmake ..
$ make
$ ldd ./bin/Protonect
$ dpkg -l libusb-1.0-0-dev
$ uname -a
$ head -1 /etc/nv_tegra_release
$ grep . /sys/bus/usb/devices/*/power/autosuspend
$ lspci -nn
$ lsusb -t
$ sudo DISPLAY=:0 LIBUSB_DEBUG=3 ./bin/Protonect gl
$ dmesg

Protonect says:
>Version: 0.1.1
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!

and dmesg:
>[  678.074259] usb 2-1.1: new SuperSpeed USB device number 5 using tegra-xhci
[  678.086734] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  678.091391] usb 2-1.1: New USB device found, idVendor=045e, idProduct=02c4
[  678.091440] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  678.091479] usb 2-1.1: Product: Xbox NUI Sensor
[  678.091513] usb 2-1.1: Manufacturer: Microsoft
[  678.091546] usb 2-1.1: SerialNumber: 005697142447
[  681.705889] usb 2-1.1: reset SuperSpeed USB device number 5 using tegra-xhci
[  681.716770] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  681.789142] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  682.198233] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  682.221256] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  696.983990] tegra-xhci tegra-xhci: Signal while waiting for configure endpoint command

[tk1_protonect.txt](https://github.com/OpenKinect/libfreenect2/files/111104/tk1_protonect.txt)

xlz, 2016-01-31:
> [Info] [TurboJpegRgbPacketProcessor] avg. time: 84.5806ms -> ~11.823Hz

It is running, is it not?

> sudo DISPLAY=:0 LIBUSB_DEBUG=3 ./bin/Protonect gl

I won't look at a bug report if the user uses sudo to run Protonect.

hanyazou, 2016-01-31:
@xlz Protonect is running until pkill Protonect on other terminal. but I can't see any image on the window.

> I won't look at a bug report if the user uses sudo to run Protonect.

[typescript.txt](https://github.com/OpenKinect/libfreenect2/files/111110/typescript.txt)

hanyazou, 2016-01-31:
@xlz what version of kernel do you use? do you compile upstream source from kernel.org?

>Things to do:
Try newer kernel: http://elinux.org/Jetson_TX1. Maybe it's a driver issue.

xlz, 2016-01-31:
I use Nvidia kernel.

The 0.1.0 release was tested on Jetson by me personally. So why are you getting these here?

Have you maxed CPU and GPU?

xlz, 2016-01-31:
I ran Protonect directly in the X session, not through ssh -X. Have you tried `DISPLAY=:0.0`?

xlz, 2016-01-31:
I can point you to a direction with the logs you provided so far. It is not a USB problem. It's not a kernel problem. It is something wrong with OpenGL as no depth processing was happening.

xlz, 2016-01-31:
I think you need LIBUSB_LOG=4 here.

hanyazou, 2016-01-31:
@xlz now I can see image! Thank you @xlz. You comment was great help.
If I boot the TK1 without Kinect plugged and plug Kinect after boot, it works. I could see images from the device on Protonect window while Protonect says 'skipping xxx packat'. (see attached log) If I boot TK1 plugging Kinect, Protonect does not show any image on the window.

I think this is a kind of kernel and driver issue. I found that the kernel panic randomly (25% or so) on boot if I plugged Kinect to the TK1. (I can't see panic message in detail because the system can't boot.)

[typescript.txt](https://github.com/OpenKinect/libfreenect2/files/111133/typescript.txt)

The log says it can process in 20Hz or so, the image in the window looks a little bit poor performance. I guess 3-5 fps. Do you see 30 fps images on your TK1? 

> [Info] [OpenGLDepthPacketProcessor] avg. time: 49.4618ms -> ~20.2176Hz

xlz, 2016-01-31:
> If I boot the TK1 without Kinect plugged and plug Kinect after boot, it works.

This is a known issue in https://github.com/xlz/libfreenect2/wiki/Jetson-TK1-HOWTO

The visualization is not free. Of course it's slower.

hanyazou, 2016-01-31:
@xlz  I could save my time if I read https://github.com/xlz/libfreenect2/wiki/Jetson-TK1-HOWTO before...

marketto89, 2016-02-18:
Are there any news about libfreenect2 with the Jetson TX1?

I tried to make it work but I had the same `[Debug] [RgbPacketStreamParser] skipping rgb packet!` error lines.

xlz, 2016-02-18:
This is not an error. You see it's a debug message.

CUDA support is merged. The Tegra hardware JPEG decoder support for TK1 is in #576, almost merged. It is probably easy to support TX1's hardware JPEG decoder but I don't have one so I'm unable to tell.

xlz, 2016-02-18:
If you have a TX1, maybe you can try it out and tell us what is needed.

marketto89, 2016-02-19:
Yes, the problem seems related to the missing Tegra hardware JPEG decoder.
An `ldd Protonect | grep jpeg` returns the following line:
`libturbojpeg.so.0 => /usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0 (0xf702e000)`

I tried to look for `/usr/lib/arm-linux-gnueabihf/tegra/libjpeg.so` with no success. 

How can I fix this problem?

xlz, 2016-02-19:
TX1 seems to name libjpeg.so to libnvjpeg.so but their content is the same. What specific errors did you get?

marketto89, 2016-02-19:
Thank you for the information about libnvjpeg.so

Actually I have no errors, but as I said I see a lot of `[Debug] [RgbPacketStreamParser] skipping rgb packet!`.

I also found that `cmake` does not find Tegra JPEG even if I put a symbolic link from `/usr/lib/arm-linux-gnueabihf/libjpeg.so` to `/usr/lib/arm-linux-gnueabihf/libnvjpeg.so`. Is it an error?
I remember that with the TK1 I saw the tegra/libjpeg.so correctly linked with `ldd Protonect`.

xlz, 2016-02-19:
What is the Tegra related output from CMake? It Should be able to find libnvjpeg.so.

xlz, 2016-02-19:
It no longer links to libjpeg because users use other things linked to the standard libjpeg and this  tegra libjpeg creates symbol conflicts. Now it dlopen's at runtime.

marketto89, 2016-02-19:
Here is the CMake output I got: 

[cmake_output.txt](https://github.com/OpenKinect/libfreenect2/files/138070/cmake_output.txt)

xlz, 2016-02-19:
TegraJPEG has to be enabled to provide the native support.

CMake didn't find the tegra headers. It is supposed to extract gst_jpeg.tbz2 to build/gst_jpeg, and there should be nv_headers/jpeglib.h under build/gst_jpeg. Can you take a look there and see why it didn't find it, and provide `CMakeFiles/CMakeError.log`?

marketto89, 2016-02-19:
Well, I can't see the folder you mentioned.

This is the `CMakeFiles/CMakeError.log`: 

[CMakeError.log.txt](https://github.com/OpenKinect/libfreenect2/files/138331/CMakeError.log.txt)

xlz, 2016-02-19:
CMake is supposed to tar xfj gst_jpeg.tbz2 but it didn't. I don't have specific instructions but can you look at cmake --debug or cmake --trace around this line https://github.com/OpenKinect/libfreenect2/blob/master/cmake_modules/FindTegraJPEG.cmake#L51

xlz, 2016-02-19:
The directory is actually supposed to be `build/gstjpeg_src` and `nv_headers/jpeglib.h` under `build/gstjpeg_src`.

xlz, 2016-03-11:
This thread kind of digressed. All TX1 issues to be followed up in #581.

closed 2016-03-11 by xlz