OpenKinect/libfreenect2 #481: Cannot get Protonect viewer

kindsenior, 2015-11-25:
Hello

I have installed libfreenect2 following README and executed Protonect.
The viewer opens but it is black.
Is it the correct behavior?

Linux 14.04  64bit
GK110B [GeForce GTX 780 Ti]
```
$ ./bin/Protonect cpu
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 501181542542
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0c 0f 47 4d 46 31 36 38 2e 31    31332e2e2e2e2e2e474d463136382e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 501181542542
device firmware: 4.3.3912.0.7
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.9716ms -> ~52.7104Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.6442ms -> ~60.0809Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.5814ms -> ~60.3086Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.3204ms -> ~57.7355Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.7327ms -> ~53.3825Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.4102ms -> ~57.4376Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.5067ms -> ~57.121Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.5208ms -> ~60.5299Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.4508ms -> ~54.1982Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.5629ms -> ~56.9383Hz
```

[image]

xlz, 2015-11-25:
`LIBUSB_DEBUG=4 ./Protonect`? And `dmesg`?

kindsenior, 2015-11-25:
[dmesg_log.txt](https://github.com/OpenKinect/libfreenect2/files/43747/dmesg_log.txt)
[protonect_log.txt](https://github.com/OpenKinect/libfreenect2/files/43748/protonect_log.txt)

xlz, 2015-11-25:
From your log, it seems USB is working all right. The strange thing is there was no log messages from the depth processor at all, and in your case there is expected to be a lot because the cpu depth processor is slow.

Can you try these two things?

1. `git checkout v0.1-rc1`, rebuild, and run it - check if this is a regression in recent commits.
2. `gdb ./Protonect`, `run`, Ctrl-C, `thread apply all backtrace` and `bt full` - get a stack strace of all threads and see what they are doing.

kindsenior, 2015-11-27:
1. checkout and rebuild but the black screen was reproduced...
2. The result is below.
```
gdb bin/Protonect 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin/Protonect...done.
(gdb) run
Starting program: /home/k-kojima/git/libfreenect2/build/bin/Protonect 
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Environment variables: LOGFILE=<protonect.log>
Usage: /home/k-kojima/git/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[New Thread 0x7ffff01e1700 (LWP 6026)]
[New Thread 0x7fffef9e0700 (LWP 6027)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 501181542542
[Info] [Freenect2Impl] found 1 devices
[New Thread 0x7fffe614d700 (LWP 6028)]
[New Thread 0x7fffe594c700 (LWP 6029)]
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0c 0f 47 4d 46 31 36 38 2e 31    31332e2e2e2e2e2e474d463136382e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 501181542542
device firmware: 4.3.3912.0.7
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.5896ms -> ~60.2787Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.4127ms -> ~54.3104Hz
^C
Program received signal SIGINT, Interrupt.
pthread_cond_wait@@GLIBC_2.3.2 () at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
185	../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S: そのようなファイルやディレクトリはありません.
(gdb) thread apply all backtrace

[... 55 lines cut ...]
        viewer_enabled = <optimized out>
        listener = {<libfreenect2::FrameListener> = {_vptr.FrameListener = 0x7ffff7dd03f0 <vtable for libfreenect2::SyncMultiFrameListener+16>}, impl_ = 0x95c630}
        undistorted = {timestamp = 6374144, sequence = 0, width = 512, height = 424, bytes_per_pixel = 4, data = 0x7fffe40fe040 "", rawdata = 0x7fffe40fe010 ""}
        executable_name_idx = <optimized out>
        freenect2 = {_vptr.Freenect2 = 0x7ffff7dd1750 <vtable for libfreenect2::Freenect2+16>, impl_ = 0x614bf0}
        registration = 0x7fffe3cd8010
        pipeline = <optimized out>
        frames = {_M_t = {
            _M_impl = {<std::allocator<std::_Rb_tree_node<std::pair<libfreenect2::Frame::Type const, libfreenect2::Frame*> > >> = {<__gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<libfreenect2::Frame::Type const, libfreenect2::Frame*> > >> = {<No data fields>}, <No data fields>}, _M_key_compare = {<std::binary_function<libfreenect2::Frame::Type, libfreenect2::Frame::Type, bool>> = {<No data fields>}, <No data fields>}, _M_header = {_M_color = std::_S_red, _M_parent = 0x0, _M_left = 0x7fffffff9f58, _M_right = 0x7fffffff9f58}, 
              _M_node_count = 0}}}
        registered = {timestamp = 1, sequence = 32767, width = 512, height = 424, bytes_per_pixel = 4, data = 0x7fffe4029040 "", rawdata = 0x7fffe4029010 ""}
        framecount = 0
        viewer = {<WithOpenGLBindings> = {_vptr.WithOpenGLBindings = 0x408b70 <vtable for Viewer+16>, bindings = 0x9b7520}, shouldStop = false, window = 0x90e110, triangle_vbo = 14, triangle_vao = 0, renderShader = {<WithOpenGLBindings> = {
              _vptr.WithOpenGLBindings = 0x408670 <vtable for ShaderProgram+16>, bindings = 0x9b7520}, program = 3, vertex_shader = 1, fragment_shader = 2, 
[... cut ...]
              bindings = 0x9b7520}, bytes_per_pixel = 4, height = 0, width = 0, texture = 0, data = 0x0, size = 0}, ir = {<WithOpenGLBindings> = {_vptr.WithOpenGLBindings = 0x4086f0 <vtable for Texture<ImageFormat<4ul, 33326u, 6403u, 5126u> >+16>, bindings = 0x9b7520}, 
            bytes_per_pixel = 4, height = 0, width = 0, texture = 0, data = 0x0, size = 0}}
        binpath = {static npos = <optimized out>, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x614bb8 "/home/k-kojima/git/libfreenect2/build/bin/"}}
```

xlz, 2015-11-27:
I really cannot see anything wrong. Need your help here.

Can you investigate how `DepthPacketStreamParser::onDataReceived()` is being called in your case ? Here: https://github.com/OpenKinect/libfreenect2/blob/master/src/depth_packet_stream_parser.cpp#L62

Add `LOG_INFO << "some string"` to crucial points in this function and see its code coverage. There are several `LOG_DEBUG` already in there for error cases, but none were printed in your log. 

At least add logging to these two points: two around `processor_->process(packet);`, and one before `if(in_length == 0)` to print data length.

kindsenior, 2015-11-27:
I added LOG_INFO at several points
like https://github.com/kindsenior/libfreenect2.git 's branch add-log-info

It prints as below.
```memcpy``` is printed, but ```footer_found``` isn't.
Because ```wb.length + in_length != wb.capacity + sizeof(DepthSubPacketFooter)``` ,
the necessary process seems not to be executed?

```
$ gdb bin/Protonect
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin/Protonect...done.
(gdb) run cpu
Starting program: /home/k-kojima/git/libfreenect2/build/bin/Protonect cpu
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Environment variables: LOGFILE=<protonect.log>
Usage: /home/k-kojima/git/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[New Thread 0x7ffff01e2700 (LWP 3095)]
[New Thread 0x7fffef991700 (LWP 3096)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 501181542542
[Info] [Freenect2Impl] found 1 devices
[New Thread 0x7fffec8dc700 (LWP 3097)]
[New Thread 0x7fffe7fff700 (LWP 3098)]
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0c 0f 47 4d 46 31 36 38 2e 31    31332e2e2e2e2e2e474d463136382e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[... 44 lines cut ...]
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length==0
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length!=0
[Info] [DepthPacketStreamParser] memcpy
[Info] [DepthPacketStreamParser] onDataReceived()
[Info] [DepthPacketStreamParser] in_length!=0
[Info] [DepthPacketStreamParser] memcpy
```

xlz, 2015-11-27:
Can you print every `in_length` for a while?

kindsenior, 2015-11-28:
Print out of in_length is below.
Print command is executed at first in onDataReceived function.
[in_length_log.txt](https://github.com/OpenKinect/libfreenect2/files/46333/in_length_log.txt)

xlz, 2015-11-28:
Good, now there is clearer understanding of the situation. The problem is in usb transfer.

All the in_length is 11264, and not enough data is gathered for one subpacket before next subpacket starts (when in_length becomes 0, subpacket assembly resets), then there are no subpackets assembled at all. In any case in_length should be 33792. 

@kindsenior Can you post `lspci -nn` and `libusb` here? Also, can you install this kernel http://packages.ubuntu.com/trusty-updates/linux-image-4.2.0-18-generic and try again?

@floe So we have another case of requesting 33792 but transferring 11264 bytes from last time http://www.spinics.net/lists/linux-usb/msg99789.html. Was there any resolution from last time?

floe, 2015-11-28:
No, nothing after that thread. In most cases so far, the solution was to give up and get a different XHCI controller... :-/

floe, 2015-11-28:
@kindsenior what does `lspci` show?

kindsenior, 2015-11-29:
```$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:8cb1]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:8cba]
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I218-V [8086:15a1]
00:1a.0 USB controller [0c03]: Intel Corporation Device [8086:8cad]
00:1b.0 Audio device [0403]: Intel Corporation Device [8086:8ca0]
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:8c90] (rev d0)
00:1c.3 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev d0)
00:1d.0 USB controller [0c03]: Intel Corporation Device [8086:8ca6]
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:8cc4]
00:1f.2 SATA controller [0106]: Intel Corporation Device [8086:8c82]
00:1f.3 SMBus [0c05]: Intel Corporation Device [8086:8ca2]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GK110B [GeForce GTX 780 Ti] [10de:100a] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GK110 HDMI Audio [10de:0e1a] (rev a1)
03:00.0 PCI bridge [0604]: ASMedia Technology Inc. ASM1083/1085 PCIe to PCI Bridge [1b21:1080] (rev 04)
```

Sorry, I couldn't attach the libusb.so to github's comment so posted the url.
https://drive.google.com/file/d/0B1NnnpidNMmgTTdrZWU0TXhHeW8/view?usp=sharing

Install the kernel from now so just a moment.

xlz, 2015-11-29:
@kindsenior Sorry, I meant `lsusb`.

kindsenior, 2015-11-29:
Sorry.....
```
$ lsusb
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:02d8 Microsoft Corp. 
Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 17ef:6009 Lenovo ThinkPad Keyboard with TrackPoint
Bus 003 Device 005: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 004: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

xlz, 2015-11-29:
The libusb you posted is libusb-0.1.so.4.4.4? It's no problem if you attached the wrong version, but anyway libfreenect2.so must be linked with the correct libusb-1.0 1.0.20+. If you haven't done it, you can check it with `ldd ./Protonect | grep libusb`, which should show libusb-1.0.so.0.1.0.

kindsenior, 2015-11-29:
Thank you very much.
Protonect showed 4 views.

I updated kernel from 3.13.0-24-generic to 4.2.0-18-generic.
But my PC has frozen at 4.2.0-18-generic.
So I downgraded to 3.13.0-68-generic and succeeded in Protonect.
At that time, in_length is 33792.

In passing, libusb version seemed to be my mistake.
```
$ ldd ./Protonect | grep libusb
	libusb-1.0.so.0 => /home/k-kojima/git/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007fb968c52000)
```

xlz, 2015-11-29:
I don't understand what happened with you switching kernel version.

closed 2015-12-23 by floe