OpenKinect/libfreenect2 #483: Protonect don't detect my kinect

djo90, 2015-11-26:
Hello,
I followed the install instruction of libfreenect2 in ubuntu. But at the final step, when I wrote the command ./bin/Protenect. It doesn't recognize my kinect even if it's plugged.

djo@djo-K56CM:~/libfreenect2$ ./bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

xlz, 2015-11-26:
`lsusb`?

djo90, 2015-11-26:
the usb3.0 is working fine 

Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 004: ID 13d3:5165 IMC Networks 
Bus 003 Device 003: ID 13d3:3362 IMC Networks 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 045e:02c4 Microsoft Corp. 
Bus 002 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1c4f:0054 SiGma Micro 
Bus 001 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I don't know what I should do, the cmake and make passed good. I need to change some parameters?

xlz, 2015-11-26:
udev rules? https://github.com/OpenKinect/libfreenect2#protonect-complains-about-no-device-connected-or-failure-opening-device?

djo90, 2015-11-26:
thanks it worked :)

closed 2015-11-28 by floe