OpenKinect/libfreenect2 #485: Getting depth of an area on the Color image

hvictor, 2015-11-29:
Dear all, I am using libfreenect2 on the Jetson. 
Suppose I identify a tennis ball with HSV thresholding and Hough circle transforms. Then I have a circular area on the Color frame.
I want to get the depth for those pixels in the big Color frame. So I am integrating Registration::apply in the libfreenect2 modified for Jetson, but I cannot understand a thing:
In a post I read that "undistorted" is the rectified depth map. Is it equivalent to the argument Frame *depth ? If no, could you tell me how to get all necessary parameters, in order to have bigdepth filled correctly? Or an area of it...

xlz, 2015-11-29:
Do the circle detection on the `registered` image (512x424), and the same area in `undistorted` is the depth.

hvictor, 2015-11-29:
Dear xlz, thank you: what is exactly the registered image? does it have colors? how can I get it? 
I searched the API for a Frame::Registered but I didn't find anything!

hvictor, 2015-11-29:
Dear xlz, I think I need to integrate Registration: I followed your post here for using libfreenect2 with Jetson TK1 CUDA acceleration: https://github.com/xlz/libfreenect2/wiki/Jetson-TK1-HOWTO

Is it possible for me to build the registered image there?

There I dont have Registration support: how can I get the registered image in Protonect from that code? Or, how can I use Tegra acceleration in your new libfreenect2 having Registration? 

Thank you infinitely,

Victor

xlz, 2015-11-29:
> You are recommended to use the upstream libfreenect2 directly 

xlz/libfreenect2 does not have registration functions.

The `registered` image is the color image registered to the depth image.

hvictor, 2015-11-29:
Thank you: but how can I enable Tegra acceleration on the upstream libfreenect2?

xlz, 2015-11-29:
You can wait for it to be enabled, or you can port the code yourself.

hvictor, 2015-11-29:
Else if I only want to implement registration in the tegra-capable code
using the existing IR and Camera parameters, then should I use the depth
frame as the depth argument as well as for the undistorted?

xlz, 2015-11-29:
I think you can just copy registration.cpp and registration.h from here into your project, and add the two files into CMakeLists.txt. Registration code seems quite self-contained. I think that's about all you need to backport it. It might not work on first try though.

hvictor, 2015-11-29:
Okkk I was also trying that Yes... could you please, Kindly just send me a
short example of call to apply() registration? I would like to see which
frames you pass as arguments: to me it is not clear the difference between
the depth frame and the other parameter on which the registration is
performed on

Victor

xlz, 2015-11-29:
Example: https://github.com/OpenKinect/libfreenect2/blob/master/examples/Protonect.cpp#L232-L238

Documentation: https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/registration.h#L73-L82

hvictor, 2015-11-29:
Thank you, I will let you know my outcomes

hvictor, 2015-11-29:
Dear Lingzhu,

I managed to compute the registered image by using registered.cpp and .h,
but if I show it, it is all black:

libfreenect2::Registration* registration = new
libfreenect2::Registration(dev->getIrCameraParams(),
dev->getColorCameraParams());
   libfreenect2::Frame undistorted(512, 424, 4), registered(512, 424, 4);

Then:

registration->apply(rgb, depth, &undistorted, &registered);
cv::imshow("REG", cv::Mat(registered.height, registered.width, CV_8UC4,
registered.data));

where rgb, depth are retrieved directly from frames[].

Do you have any suggestion?

xlz, 2015-11-29:
```
const uint32_t *p = reinterpret_cast<const uint32_t *>(registered.data);
std::cerr << p[100*512 + 100] << std::endl;
```
What does this print?

hvictor, 2015-11-29:
It prints always 0.
I have the rgb->width and rgb->height both at 1 ...is this normal?

hvictor, 2015-11-29:
I found out that apply() is refusing some condition on the input
parameters: I let you know shortly which, hold on

2015-11-29 23:56 GMT+01:00 Victor Huerlimann <victor.huerlimann@gmail.com>:

xlz, 2015-11-29:
No, xlz/libfreenect2 uses something special in rgb.data. See this https://github.com/xlz/libfreenect2/blob/jetsontk1/examples/protonect/Protonect.cpp#L89-L93

hvictor, 2015-11-29:
Yes I see:
Yo are interpreting data, which originally is:

data = new unsigned char[width * height * bytes_per_pixel];

i.e. unsigned char *

as: unsigned char **

How should I create a Frame object with the correct data using unsigned
char **pprgba = reinterpret_cast<unsigned char **>(rgb->data); ?

xlz, 2015-11-30:
```
uint8_t *rgb_data = *reinterpret_cast<unsigned char **>(rgb->data);
```

hvictor, 2015-11-30:
Thank you, I tried, didn't do the trick: see comments:

unsigned char **pprgba = reinterpret_cast<unsigned char **>(rgb->data);

    cv::Mat rgba(1080, 1920, CV_8UC4, pprgba[0]);
    cv::Mat bgra(1080, 1920, CV_8UC4);
    cv::cvtColor(rgba, bgra, cv::COLOR_RGBA2BGRA);

    cv::imshow("rgb", bgra); // This is show right

    // Your tip
    uint8_t *rgb_data = *reinterpret_cast<unsigned char **>(rgb->data);

    // New frame for calling Registration::apply()
    libfreenect2::Frame *s = new libfreenect2::Frame(1920, 1080, 4);
    s->data = rgb_data;
    registration->apply(s, depth, &undistorted, &registered);

    const uint32_t *p = reinterpret_cast<const uint32_t *>(registered.data);
    std::cerr << p[100*512 + 100] << std::endl; // But this is always 0

    // And this is black
    cv::imshow("REG", cv::Mat(registered.height, registered.width, CV_8UC4,
registered.data));

hvictor, 2015-11-30:
Dear Lingzhu, I have also checked that _rgb_camera_params in
libfreenect2.cpp are set correctly and completely how is done in the new
version, and also checked: RgbCameraParamsResponse that contains all
required parameters. So I think the problem is still related to my previous
mail, how I am using your suggested uint8_t *rgb_data =
*reinterpret_cast<unsigned char **>(rgb->data);
Could you please, kindly help me to fix this? Am I using it wrong?

Thank you very much,
Victor

2015-11-30 1:13 GMT+01:00 Victor Huerlimann <victor.huerlimann@gmail.com>:

xlz, 2015-11-30:
In fact, I don't know.

Just follow the code and print some more. Where does it start being all zeros and there can you fix it.

hvictor, 2015-12-01:
I found that here (I added a printf) in registration.cpp:

// check if c_off is outside of rgb image
// checking rx/cx is not needed because the color image is much wider then
the depth image
    if(c_off < 0 || c_off >= size_color){
      printf("size_color = %d, C_off = %d = cx + cy * 1920 = %d + %d * 1920
---> outside RGB image\n", size_color, c_off, cx, cy);
      *map_c_off = -1;
      continue;
    }

I have values such as:

size_color = 2073600, C_off = 2269238 = cx + cy * 1920 = 1718 + 1181 * 1920
---> outside RGB image
size_color = 2073600, C_off = 2269241 = cx + cy * 1920 = 1721 + 1181 * 1920
---> outside RGB image
size_color = 2073600, C_off = 2269244 = cx + cy * 1920 = 1724 + 1181 * 1920
---> outside RGB image
size_color = 2073600, C_off = 2269247 = cx + cy * 1920 = 1727 + 1181 * 1920
---> outside RGB image
size_color = 2073600, C_off = 2269250 = cx + cy * 1920 = 1730 + 1181 * 1920
---> outside RGB image

and get c_off set to -1 later. What do you think?

xlz, 2015-12-01:
Is it all like this or is this just the last edge pixels? You can use `Registration::apply( int dx, int dy, float dz, float& cx, float &cy)` to test individual pixels and see if their color is correct.

hvictor, 2015-12-01:
Dear Lingzhu I managed to do it! And I am detecting good a tennis ball in
real time. Are You interested that I share the code with you, so you have
registration enabled with tegra support?

xlz, 2015-12-01:
What was the problem that made it black?

ahundt, 2015-12-02:
Sounds interesting, could this code go up on github somewhere?

xlz, 2015-12-02:
The code of registration is long in the main repo. The user is using my xlz/libfreenect2 fork with Tegra support but with no registration. I didn't maintain the fork because the Tegra support was supposed to be submitted to the main repo after 0.1 quite soon, but I don't know why, more and more stuff is piling up for the 0.1.

hvictor, 2015-12-02:
Hi man
I have found out that having completed the set of rgb camera params did the
trick in the end. I can send you a github link later so

hvictor, 2015-12-02:
Hi, I am cleaning up from my comments and will upload on github or send you
a tar.gz ...Meanwhile could you please help me to understand what call do I
need to perform in order to reproject pixel coordinates to 3D-world
coordinates? I have the real-world depth coordinate z, yes, but I also
would like the x and y in real world

Thanks!

Victor

xlz, 2015-12-03:
You can use `getPointXYZRGB()`.

closed 2016-01-27 by xlz