OpenKinect/libfreenect2 #487: Protonect work fine from the depends folder

djo90, 2015-12-01:
When I execute ./bin/Protonect from libfreenect2 folder, I get error in building opencl but when I execute it from depends folder. It works fine. Is it normal?

Also when I execute rosrun kinect2_bridge kinect2_bridge I got this error
[rospack] Error: package 'kinect2_bridge' not found

How I can solve that?

xlz, 2015-12-01:
It's not normal.

Report kinect2_bridge errors to iai_kinect2. But I bet you didn't do `. devel/setup.bash`.

djo90, 2015-12-01:
I have this file setup.bash with this content inside. what should i modify in it?
#!/usr/bin/env bash
# generated from catkin/cmake/templates/setup.bash.in

CATKIN_SHELL=bash

# source setup.sh from same directory as this file
_CATKIN_SETUP_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)
. "$_CATKIN_SETUP_DIR/setup.sh"

closed 2015-12-01 by djo90