OpenKinect/libfreenect2 #488: It works without displaying a useful data or window.

kerolos, 2015-12-01:
hello,

when i installed everything as it illustrates in tutorial, but when i tested it, i got this result only in the terminal window, furthermore it does not show any video. 
this is my specification of my PC:  op:ubuntu 14.04 LTS, RAM: 16 GB, Processor:Intel® Core™ i7-4820K CPU @ 3.70GHz × 8, and graphic card : GeForce GTX 760/PCIe/SSE2.

thanks in advance

kerolos@Schnittrechner:~/libfreenect2/build/bin$ ./Protonect
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/kerolos/libfreenect2/build/lib/libfreenect2.so)
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/kerolos/libfreenect2/build/lib/libfreenect2.so)
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 039374243547
[Info] [Freenect2Impl] found 1 devices
^Z
[5]+  Angehalten              ./Protonect
kerolos@Schnittrechner:~/libfreenect2/build/bin$ ./Protonect
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/kerolos/libfreenect2/build/lib/libfreenect2.so)
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/kerolos/libfreenect2/build/lib/libfreenect2.so)
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 039374243547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 03 13 01 47 4b 55 30 34 35 2e 31    31332e2e2e2e2e2e474b553034352e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
^Z
[6]+  Angehalten              ./Protonect

xlz, 2015-12-01:
`LIBUSB_DEBUG=4 ./Protonect`? `dmesg`? `gdb ./Protonect; Ctrl-C; bt`?

kerolos, 2015-12-03:
dmesg:
[   86.191532] usb 6-2.1: reset SuperSpeed USB device number 3 using xhci_hcd
[   86.205509] usb 6-2.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[   86.205747] xhci_hcd 0000:07:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8804041ef800
[   86.205754] xhci_hcd 0000:07:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8804041ef840
[   86.205757] xhci_hcd 0000:07:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8804041ef8c0
[   86.205760] xhci_hcd 0000:07:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8804041ef880

LIBUSB_DEBUG=4 ./Protonect:
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[ 1.428699] [00000c0d] libusb: debug [libusb_alloc_transfer] transfer 0xcc2788
[ 1.428705] [00000c0d] libusb: debug [libusb_submit_transfer] transfer 0xcc2788
[ 1.428711] [00000c0d] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 1.428719] [00000c0d] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 1.428757] [00000c0f] libusb: debug [handle_events] poll() returned 1
[ 1.428766] [00000c0f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=0
[ 1.428769] [00000c0f] libusb: debug [handle_control_completion] handling completion status 0
[ 1.428772] [00000c0f] libusb: debug [disarm_timerfd] 
[ 1.428774] [00000c0f] libusb: debug [usbi_handle_transfer_completion] transfer 0xcc2788 has callback 0x7fa895309650
[ 1.428776] [00000c0f] libusb: debug [sync_transfer_cb] actual_length=0
[ 1.428794] [00000c0f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.428797] [00000c0d] libusb: debug [libusb_free_transfer] transfer 0xcc2788
[ 1.428798] [00000c0f] libusb: debug [handle_events] poll() 4 fds with timeout in 100ms
[ 1.428808] [00000c0d] libusb: debug [libusb_alloc_transfer] transfer 0xcc2788
[ 1.428817] [00000c0d] libusb: debug [libusb_submit_transfer] transfer 0xcc2788
[ 1.428824] [00000c0d] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 1.428830] [00000c0d] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 20
[ 1.428839] [00000c0d] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 1.428882] [00000c0f] libusb: debug [handle_events] poll() returned 1
[ 1.428889] [00000c0f] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=20
[ 1.428891] [00000c0f] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 1.428894] [00000c0f] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 1.428896] [00000c0f] libusb: debug [disarm_timerfd] 
[ 1.428898] [00000c0f] libusb: debug [usbi_handle_transfer_completion] transfer 0xcc2788 has callback 0x7fa895309650
[ 1.428900] [00000c0f] libusb: debug [sync_transfer_cb] actual_length=20
[ 1.428916] [00000c0f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.428918] [00000c0d] libusb: debug [libusb_free_transfer] transfer 0xcc2788
[ 1.428924] [00000c0d] libusb: debug [libusb_alloc_transfer] transfer 0xcc2788
[ 1.428931] [00000c0d] libusb: debug [libusb_submit_transfer] transfer 0xcc2788
[ 1.428937] [00000c0d] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 1.428919] [00000c0f] libusb: debug [handle_events] poll() 4 fds with timeout in 100ms
[ 1.428944] [00000c0d] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 512
[ 1.428958] [00000c0d] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 1.428988] [00000c0f] libusb: debug [handle_events] poll() returned 1
[ 1.428993] [00000c0f] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=512
[ 1.428995] [00000c0f] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 1.428997] [00000c0f] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 1.428999] [00000c0f] libusb: debug [disarm_timerfd] 
[ 1.429001] [00000c0f] libusb: debug [usbi_handle_transfer_completion] transfer 0xcc2788 has callback 0x7fa895309650
[ 1.429003] [00000c0f] libusb: debug [sync_transfer_cb] actual_length=512
[ 1.429006] [00000c0f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.429008] [00000c0f] libusb: debug [handle_events] poll() 4 fds with timeout in 100ms
[ 1.429008] [00000c0d] libusb: debug [libusb_free_transfer] transfer 0xcc2788
[ 1.429019] [00000c0d] libusb: debug [libusb_alloc_transfer] transfer 0xcc2788
[ 1.429023] [00000c0d] libusb: debug [libusb_submit_transfer] transfer 0xcc2788
[ 1.429041] [00000c0d] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 1.429047] [00000c0d] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16
[ 1.429054] [00000c0d] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 1.429073] [00000c0f] libusb: debug [handle_events] poll() returned 1
[ 1.429077] [00000c0f] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=16
[ 1.429079] [00000c0f] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 1.429081] [00000c0f] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[... cut ...]
0x0000:  01 26 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e262e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0040:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0050:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e

xlz, 2015-12-03:
> [Debug] [Freenect2DeviceImpl] 1818624 bytes of raw data

This is not supposed to happen.

floe, 2015-12-23:
Still an issue? Otherwise, please close.

closed 2016-01-27 by floe