OpenKinect/libfreenect2 #489: Error LNK1120: 1 resolved external symbol

jaebums, 2015-12-01:
Hello, I have been trying to build the library but have failed over 1 week. I do not have a good experience solving this kind issues partly since I have been using C but not C++. Any comment how I can fix this kind of error?

[image]

xlz, 2015-12-02:
It's a bug. I didn't test it on Windows.

MSVC refuses to resolve the symbol when the return type is different (a different type of pointer). I didn't see this with GCC. I'll create a fix promptly.

xlz, 2015-12-02:
Can you try the above PR 490?

jaebums, 2015-12-02:
0 Warning, 0 Error! Thank you so much!

commit 4162721c (2015-12-03): Merge pull request #490 from xlz/msvc-symbol-resolving

Fix MSVC FTBFS, closes #489
closed 2015-12-03 by floe