OpenKinect/libfreenect2 #491: Compiling Issue (Libfreenect2 updated?)

filipetrocadoferreira, 2015-12-03:
[ 39%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
/home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t, bool, bool, double, double)’:
/home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:393:5: error: incomplete type ‘libfreenect2::DepthPacketProcessor’ used in nested name specifier
     libfreenect2::DepthPacketProcessor::Config config;
     ^
/home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:393:48: error: expected ‘;’ before ‘config’
     libfreenect2::DepthPacketProcessor::Config config;
                                                ^
/home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:394:5: error: ‘config’ was not declared in this scope
     config.EnableBilateralFilter = bilateral_filter;
     ^
/home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:398:46: error: invalid use of incomplete type ‘class libfreenect2::DepthPacketProcessor’
     packetPipeline->getDepthPacketProcessor()->setConfiguration(config);
                                              ^
In file included from /usr/local/include/libfreenect2/libfreenect2.hpp:34:0,
                 from /home/wiigosegundo/catkin/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:44:
/usr/local/include/libfreenect2/packet_pipeline.h:39:7: error: forward declaration of ‘class libfreenect2::DepthPacketProcessor’
 class DepthPacketProcessor;
       ^
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2

filipetrocadoferreira, 2015-12-03:
I made a checkout of libfreenect to a previous commit (21 NOV) and it compiled

RyanGordon, 2015-12-03:
Hello,

iai_kinect2 repository needs to merge in this PR to work with the latest version of libfreenect2: https://github.com/code-iai/iai_kinect2/pull/189

RyanGordon, 2015-12-17:
Looks like the PR was merged last week, @filipetrocadoferreira can you test again and see if this resolves your issue?

sari-sadiya, 2015-12-28:
Hi,

I downloaded and updated the repository today. When I run "catkin_make -DCMAKE_BUILD_TYPE="Release" I get similar errors from "Freenect2Device":

/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:5: error: ‘Config’ is not a member of ‘libfreenect2::Freenect2Device’
     libfreenect2::Freenect2Device::Config config;
     ^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:43: error: expected ‘;’ before ‘config’
     libfreenect2::Freenect2Device::Config config;
                                           ^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:401:5: error: ‘config’ was not declared in this scope
     config.EnableBilateralFilter = bilateral_filter;
     ^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:13: error: ‘class libfreenect2::Freenect2Device’ has no member named ‘setConfiguration’
     device->setConfiguration(config);
             ^
Anyone had similar issues?

xlz, 2016-01-27:
Should no longer be an issue after the 0.1 release.

closed 2016-01-27 by xlz

smarthhy, 2018-07-27:
Have you solved this problem? what should I do?  thanks!

/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:5: error: ‘Config’ is not a member of ‘libfreenect2::Freenect2Device’
libfreenect2::Freenect2Device::Config config;
^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:400:43: error: expected ‘;’ before ‘config’
libfreenect2::Freenect2Device::Config config;
^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:401:5: error: ‘config’ was not declared in this scope
config.EnableBilateralFilter = bilateral_filter;
^
/home/lair/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:405:13: error: ‘class libfreenect2::Freenect2Device’ has no member named ‘setConfiguration’
device->setConfiguration(config);
^