OpenKinect/libfreenect2 #492: store raw depth data

schmiran, 2015-12-03:
Hello, I want to store the raw depth data to a file . somebody has an idea how I do this ? I want the stored raw data to compare them with the data stored by the MS SDK .

xlz, 2015-12-03:
The raw depth data is just 512x424 float, unit: 1mm. I don't know you need anything special to save it.

pookiefoof, 2015-12-03:
Similar question here. I think the raw depth data from Kinect V2 should be 11-bit long, but I browse the code, finding that it is converted to 32-bit float at a very early stage... Any chance to access *raw* depth frames later? Thanks.

floe, 2015-12-03:
The raw phase images are 11 bit, but they don't make any sense by themselves, only after conversion. And the conversion code is modelled on the GPU shaders used by the official SDK, so the results from that stage should basically be identical.

pookiefoof, 2015-12-03:
Thanks for the explanation! Good to know that.

schmiran, 2015-12-04:
Okay thanks, that means with:

libfreenect2::Frame *depth_1 = frames_1[libfreenect2::Frame::Depth];
float *depth_1_data = (float *)depth_1->data;
for (int i = 0; i < size_depth; ++i){
outputStream.write(reinterpret_cast<char*> (&depth_1_data[i]), sizeof(512 * 424));

floe, 2015-12-04:
> it is possible to feed these back into the SDK?
If you mean the Microsoft SDK, then no.

schmiran, 2015-12-04:
Ok, too bad. To get a better overview from the user I wanted to collect data from three KinectV2 with Libfreenect2 and give those to the MS SDK 2.0 to use the tools from SDK. Do you know why this is not possible?

floe, 2015-12-04:
The MS SDK is monolithic binary-only code. So to find out where and how to insert the recorded data, you would have to fully reverse-engineer the entire SDK. That's not entirely impossible, but definitely _very_ difficult. On top of that, the SDK is only designed for one single Kinect v2.

ahundt, 2015-12-04:
Related issue:
https://github.com/OpenKinect/libfreenect2/issues/438

xlz, 2015-12-04:
If you know the format that "MS SDK 2.0" accepts, then you can transform the data to the accepted format and give to the MS SDK.

Nobody here knows the format. Maybe you know, or maybe it's written in some specs.

schmiran, 2015-12-07:
Hi, thanks for the answers. I'll try to understand the structure of .xef FILES. maybe I get the depth data rebuilt.
Another question is: do you know which streams from those 11 displayed in the SDK 2.0 comes from the Kinect v2 and which are calculated retrospectively (in the KinectService maybe)? What exactly is the Body Index and the Body Frame Stream?

xlz, 2016-01-27:
Unfortunately we don't know about details in the Microsoft SDK.

closed 2016-01-27 by xlz