OpenKinect/libfreenect2 #493: LIBUSB Error with OpenNI2 Freenect2Driver

kruss84, 2015-12-03:
Hello,

I am using the Freenect2driver for OpenNI2 from hanyazou (https://github.com/hanyazou/libfreenect2) to implement an large scale application with several kinects v2.

The target operating system is Windows 8.1 with visual studio 2013.
I made some changes in the source code so I could compile the Freenect2driver.
The installation was successful, in operation with one camera, it works without problems. When attempting to start multiple cameras I get the following LIBUSB error in Windows 8.1.

[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[Freenect2DeviceImpl] started
OpenNI2-Freenect2Driver: setStreamProperty: depth-size: 640x480
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1022
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
OpenNI2-Freenect2Driver: deiveOpen: freenect2://1
OpenNI2-Freenect2Driver:     depth-size = 640x480
OpenNI2-Freenect2Driver: Opening device freenect2://1
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
[CommandTransaction::send] sent number of bytes differs from expected number! expected: 20 got: 0
[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
[CommandTransaction::send] sent number of bytes differs from expected number! expected: 20 got: 0
[Freenect2DeviceImpl] ReadData0x14 response
0 bytes of raw data

[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
[CommandTransaction::send] sent number of bytes differs from expected number! expected: 24 got: 0
[Freenect2DeviceImpl] serial number reported by libusb 501733243142 differs from serial number  in device protocol!

When I try to start two cameras with libfreenect2 alone (https://github.com/OpenKinect/libfreenect2/issues/211), there is no error message and the images can be received without any problems in Windows 8.1.

When I start my Openni2 application on Ubuntu 14.04, there it works with two cameras without any problems, but not in windows 8.1

I would appreciate any suggestions or ideas to solve this problem.

floe, 2015-12-03:
USB3 is generally still flaky. I can only give a few generic suggestions, e.g.:
  * try updating the Windows driver for the USB host controller
  * try a different USB3 add-on card

floe, 2015-12-23:
Still an issue? Otherwise, please close.

chipweinberger, 2015-12-29:
Hey Floe,

Just wanted to add some details about these issues as I was having similar ones.

Multiple  Kinects V2s on windows does indeed work. You can fix the above error by Sleeping before opening another device though that did not fix all the USB transfer issues.

Your best bet is to use libfreenect/libusb as a single process for each kinect and doing IPC (as I've noted before). I'm curious why this is so much more robust but it is.

I've also found that I can sometimes get multiple kinects working without the multi process solution. For me this only worked when I had 1 kinect plugged into the mobo, and the other as an expansion card. 

But having multiple kinects plugged into PCIe expansion cards only worked using the multiprocess solution. I tested up to 4 kinects with 3 on PCIe, and 1 on the mobo's built in.

chipweinberger, 2015-12-29:
Maybe this is relevant? Why would multiprocess work 99% of the time but single process fails 100% of the time?

http://libusb.sourceforge.net/api-1.0/mtasync.html

closed 2016-01-27 by kruss84