OpenKinect/libfreenect2 #495: make error ([bin/Protonect], [examples/CMakeFiles/Protonect.dir/all])

OAkyildiz, 2015-12-04:
Hello,

When I run
> libfreenect2/build$ make  
I get these errors:
> ../lib/libfreenect2.so: undefined reference to `libusb_get_ss_endpoint_companion_descriptor'
> ../lib/libfreenect2.so: undefined reference to `libusb_free_ss_endpoint_companion_descriptor'
> ../lib/libfreenect2.so: undefined reference to `libusb_strerror'

FYI, in **FindLibUSB.cmake*,  I've changed 
>**12** SET(ENV{PKG_CONFIG_PATH} "${DEPENDS_DIR}/libusb/lib/pkgconfig")

to 

> SET(ENV{PKG_CONFIG_PATH} "${DEPENDS_DIR}/libusb/lib/pkgconfig:/usr/lib/x86_64-linux-> gnu/pkgconfig:$ENV{PKG_CONFIG_PATH}")

Before that, cmake wouldn't find libusb at all.

Here is the dpkg output:
>Desired=Unknown/Install/Remove/Purge/Hold
>| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
>|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
>||/ Name           Version      Architecture Description
>+++-==============-============-============-=================================
>ii  libusb-1.0-0:a 2:1.0.20-1pp amd64        userspace USB programming library
>ii  libusb-1.0-0-d 2:1.0.20-1pp amd64        userspace USB programming library
>ii  libusb-1.0-doc 2:1.0.20-1pp all          documentation for userspace USB p

Best,

xlz, 2015-12-04:
You have old libusb in `depends/`?

OAkyildiz, 2015-12-04:
The problem was solved when I called the configure in the libusb_src then called
> make clean
> make
> sudo make install

xlz, 2015-12-04:
You should remove `libusb` and `libusb_src` under `depends` if you want to use the system libusb-1.0-0.

xlz, 2015-12-04:
You can definitely see this in your CMake output

```
-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
 -- RPATH set to 
 -- Linking with these libraries: 
  LibUSB_LIBRARY-NOTFOUND
```

**'libusb-1.0>=1.0.20' not found**, it means it found libusb 1.0.19, and this version is not allowed.

OAkyildiz, 2015-12-05:
I have 1.0.20, but I have tiried it without `libusb` and `libusb_src` before, the result was the same. `make clean`  fixed it tough.

closed 2015-12-05 by OAkyildiz