OpenKinect/libfreenect2 #496: Why DepthPacketProcessor is internal ? How to enable it ?

lamogui, 2015-12-04:
Hi
I try to build https://github.com/code-iai/iai_kinect2
But it doesn't build anymore because DepthPacketProcessor has switched to internal.
I read in this commit [https://github.com/OpenKinect/libfreenect2/commit/cdd4f06b97177f2f2e8083c8eba6f6f6f4d11ddb] that something has been done to keep the compatibility but how to enable it properly ?

lamogui, 2015-12-04:
The error is ```  In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t, bool, bool, double, double)’:
/home/uqac/Documents/recherche/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:393:5: error: incomplete type ‘libfreenect2::DepthPacketProcessor’ used in nested name specifier
     libfreenect2::DepthPacketProcessor::Config config;
```

xlz, 2015-12-04:
Use this https://github.com/code-iai/iai_kinect2/pull/189.

xlz, 2015-12-04:
The compatibility breaks something else and was removed. Alternatives were thought of and evaluated but there is no better solution than this.

closed 2015-12-07 by floe