OpenKinect/libfreenect2 #497: Docs now on Github Pages

floe, 2015-12-07:
I've generated the latest docs from master and put them up at https://openkinect.github.io/libfreenect2/ - please browse and check for inconsistencies.

xlz, 2015-12-08:
`cmake .. -DPROJECT_VERSION=0.1; make doc` to add version number next time.

xlz, 2015-12-10:
We should check through the code for proper error checking and add some proper programmatic error reporting mechanism in 0.2. Right now most USB errors lead to segfaults and corruption, and other errors just throw exceptions. None of this should happen in a library. All the LOG_ERROR (and maybe some LOG_WARNING too) messages, which require external intervention for either recovery or termination, should be classified and reported programmatically via something like return values, or getStatus(), or getLastError() and errorString(). The logging utilities are being abused as a post-hoc error reporting channel.

floe, 2015-12-13:
> `cmake .. -DPROJECT_VERSION=0.1; make doc` to add version number next time.
Did that, didn't seem to make any difference? I'm no Doxygen expert, but in the generated Doxyfile, I already have `PROJECT_VERSION=0.1`, so there must be some other reason the version doesn't show up in the generated docs...

xlz, 2015-12-13:
Starting from the generated Doxyfile, I used doxygen 1.8.9.1 and 1.8.6, both of which are able to output the version number. And it was not a CMake problem...

floe, 2015-12-13:
How is it supposed to look like, I think you had another example online somewhere?

xlz, 2015-12-13:
index.html:
```html
<div id="top"><!-- do not remove this div, it is closed by doxygen! -->
<div id="titlearea">
<table cellspacing="0" cellpadding="0">
 <tbody>
 <tr style="height: 56px;">
  <td style="padding-left: 0.5em;">
   <div id="projectname">libfreenect2
   &#160;<span id="projectnumber">0.1</span>
   </div>
  </td>
 </tr>
 </tbody>
</table>
</div>
<!-- end header part -->
<!-- Generated by Doxygen 1.8.9.1 -->
```

floe, 2015-12-14:
Manually running `doxygen` again seems to have fixed this. Updated.

xlz, 2015-12-24:
OK, you may want to put the link somewhere in the front page so people get started to use it.

floe, 2015-12-24:
Ah, yeah, good point. Updated README, closing here.

closed 2015-12-24 by floe