OpenKinect/libfreenect2 #500: multiple Kinect v2, intel nuc, Not enough bandwidth

NicoSilvestri, 2015-12-09:
I'm trying to get multiple kinect v2 to work simultaneously on an intel NUC running Ubuntu 15.10 using libfreenect2. 
I'm working on Protonect example: it works fine with one Kinect.
I modified it duplicating device, listener and so on. 
In this way i'm able to open one device but not the second.
The output of the application is the following:

Environment variables: LOGFILE=<protonect.log>
Usage: /home/mectho/development/libfreenect2/examples-build/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:22 with serial 500603743142
[Info] [Freenect2Impl] found valid Kinect v2 @2:23 with serial 500917643142
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [protocol::UsbControl] failed to set configuration! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=4 .
[Error] [Freenect2Impl] failed to open Kinect v2: @2:23

Typing dmesg on the shell i get this lines:

[ 7324.771252] usb 2-4.1: reset SuperSpeed USB device number 22 using xhci_hcd
[ 7324.867277] usb 2-2.1: reset SuperSpeed USB device number 23 using xhci_hcd
[ 7324.883981] usb 2-2.1: Not enough bandwidth for new device state.
[ 7324.884244] Protonect[7158]: segfault at 0 ip           (null) sp 00007ffcd3c53de8 error 14 in Protonect[400000+b000]

I tryed to modify grub file like suggested in:
https://github.com/OpenKinect/libfreenect2/issues/97
https://github.com/christiankerl/libfreenect2/issues/7

with different values: 
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=64"
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=1000"
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=4000"

but the error persists.

Machine:
CPU: Intel(R) Core(TM) i5-5250U CPU @ 1.60GHz
RAM: 7869 MiB 
kernel: 4.2.0-19-generic

I recently intalled intel-linux-graphics-installer 1.2.1 and runned it.

Am I doing something wrong?

xlz, 2015-12-09:
Not enough bandwidth is reported by the hardware. USBFS buffer size is a software parameter.

You should check the PCI-e bus bandwidth of the Intel NUC. It may be low-end specs. If that is the case, there is nothing software can do.

bellekci, 2015-12-19:
@NicoSilvestri can you share your lspci output? I am also using NUC 5th gen but I am having USB-freeze problem as I have described at #216 even with one kinect. I am curious if your USB controller is also this one:
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)

NicoSilvestri, 2015-12-21:
@xlz Thanks for your reply
@bellekci 
It seems the same:
00:00.0 Host bridge: Intel Corporation Broadwell-U Host Bridge -OPI (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)
00:03.0 Audio device: Intel Corporation Broadwell-U Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)
00:16.0 Communication controller: Intel Corporation Wildcat Point-LP MEI Controller #1 (rev 03)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection (3) I218-V (rev 03)
00:1b.0 Audio device: Intel Corporation Wildcat Point-LP High Definition Audio Controller (rev 03)
00:1c.0 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 (rev e3)
00:1c.3 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #4 (rev e3)
00:1d.0 USB controller: Intel Corporation Wildcat Point-LP USB EHCI Controller (rev 03)
00:1f.0 ISA bridge: Intel Corporation Wildcat Point-LP LPC Controller (rev 03)
00:1f.2 SATA controller: Intel Corporation Wildcat Point-LP SATA Controller [AHCI Mode] (rev 03)
00:1f.3 SMBus: Intel Corporation Wildcat Point-LP SMBus Controller (rev 03)
02:00.0 Network controller: Intel Corporation Wireless 7265 (rev 59)

bellekci, 2015-12-21:
Interesting. Did you keep it running for a while? 
The kinect 2 usb gets disconnectted after some duration in my case.

NicoSilvestri, 2015-12-28:
@bellekci I kept it running for a few minutes and it hasn't any kind of problem.

closed 2016-03-07 by xlz

alexjheck, 2019-02-21:
Likely cause of this issue is that you only have one usb controller on the NUC. To run multiple kinect v2s you must have a controller for each camera. High-end desktops typically only have one controller, I am guessing your NUC is the same. To work with multiple kinects simultaneously, consider buying a PCIe expansion card with multiple controllers. My educated guess says that it will never work on a NUC unless you use an external PCIe expansion box.