OpenKinect/libfreenect2 #502: Error compiling Protonect in qtcreator

TheodoreT, 2015-12-11:
Hi guys, I have followed and finished successfully the procedure for installing libfreenect2 library in windows 8. Moreover, if I run the built Protonect.exe from command line it gives me without problem the output from the 4 views. However, I tried now to compile and run the Protonect example source code through qtcreator and while it compiles if I try to run it, it gives the following error:

>[Info] [Freenect2Impl] enumerating devices...
>[Info] [Freenect2Impl] 12 usb devices connected
>[Info] [Freenect2Impl] found valid Kinect v2 @1:6 with serial 014947350647
>[Info] [Freenect2Impl] found 1 devices
>[Info] [Freenect2DeviceImpl] opening...
>[Info] [Freenect2DeviceImpl] opened
>[Info] [Freenect2DeviceImpl] starting...
>[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
>[Info] [Freenect2DeviceImpl] submitting usb transfers...
>[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
>[Info] [Freenect2DeviceImpl] started
>The program has unexpectedly finished.

does anyone of can understand what is the problem. Thanks.

xlz, 2015-12-11:
> The program has unexpectedly finished.

For this error, you can run Protonect from Visual Studio and catch the error.

I don't see anything wrong in your qtcreator messages. Protonect itself depends on libfreenect2.lib and glfw3dll.lib (if you use OpenGL viewer here).

TheodoreT, 2015-12-11:
Hi @xlz, thanks for the reply. After following your advice and trying to deploy Protonect in Visual Studio line by line it seems that the following line causes the problem:

```c++
std::string serial = freenect2.getDefaultDeviceSerialNumber();
```

running the debugger it gives me the following message:

>Unhandled exception at 0x00007FFD8D8AD164 (msvcr120d.dll) in Protonect.exe: 0xC0000005: >Access violation reading location 0xFFFFFFFFFFFFFFFF.

>	Protonect.exe!std::allocator<char>::deallocate(char * _Ptr, unsigned __int64 __formal) Line 574	C++

[image]

[image]

I do not know does that gives you any clue what might be the problem?

xlz, 2015-12-16:
Unhandled exception at 0x00007FFD8D8AD164 is probably just failure to get the serial number string. I think there is some known bug here that does not parse the string correctly, but it's not the problem for you.

Anyway, you are having usb problems. You can set environment variables LIBUSB_DEBUG=4, and run `Protonect.exe 014947350647` and see what is the error.

TheodoreT, 2015-12-16:
Hi @xlz setting the environment variable LIBUSB_DEBUG=4 as you said I got the following output:

$ ./Protonect.exe 014947350647
hello world!
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\userKinect\Documents\Visual Studio 2013\Projects\Protonect\x64\Debug\Protonect.exe [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[timestamp] [threadID] facility level [function call] <message>

--------------------------------------------------------------------------------
[ 0.000000] [00000edc] libusb: debug [libusb_init] created default context
[ 0.000000] [00000edc] libusb: debug [libusb_init] libusb v1.0.20.11015
[ 0.000000] [00000edc] libusb: debug [windows_init] Windows 8.1 64-bit
[ 0.000000] [00000edc] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.000000] [00000edc] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.000000] [00000edc] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.000000] [00000edc] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [00000edc] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [00000edc] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [00000edc] libusb: debug [windows_init] hires timer available (Frequency: 1558891 Hz)
[ 0.000000] [00000edc] libusb: debug [windows_init] timer thread will run on core #0
[ 0.000000] [00000edc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [00000edc] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [00000edc] libusb: debug [libusb_get_device_list]
[ 0.000000] [00001220] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.000000] [00001220] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.000000] [00001220] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.000000] [00001220] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [31C]
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [2CF]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [3CD]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [3D3]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): usbhub
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [132]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [3B0]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [399]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [11D]
[ 0.000000] [00000edc] libusb: debug [get_api_type] driver(s): usbhub
[ 0.000000] [00000edc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [217]
[ 0.000000] [00000edc] libusb: debug [windows_get_device_list] found existing device for session [132] (0.0)
[ 0.000000] [00000edc] libusb: debug [init_device] got bus number from ancestor #2
[ 0.071147] [00000edc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.071147] [00000edc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.071147] [00000edc] libusb: debug [init_device] (bus: 2, addr: 2, depth: 1, port: 1): '\\.\USB#VID_8087&PID_8001#5&37F67426&0&1'
[ 0.071147] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [32A]
[ 0.071147] [00000edc] libusb: debug [init_device] got bus number from ancestor #3
[ 0.071147] [00000edc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.071147] [00000edc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.071147] [00000edc] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 3): '\\.\USB#VID_046D&PID_C063#6&271BB1DA&0&3'
[ 0.071147] [00000edc] libusb: debug [windows_get_device_list] allocating new device for session [393]
[ 0.071147] [00000edc] libusb: debug [init_device] got bus number from ancestor #2
[ 0.071147] [00000edc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[... cut ...]
[ 0.295474] [00001220] libusb: debug [handle_events] poll() returned 0
[ 0.295474] [00000edc] libusb: debug [libusb_exit]
[ 0.295474] [00000edc] libusb: debug [libusb_exit] destroying default context
[ 0.295474] [00000edc] libusb: warning [libusb_exit] some libusb_devices were leaked
[ 0.295474] [00000edc] libusb: warning [libusb_exit] application left some devices open
[ 0.295474] [00000edc] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.295474] [000012c0] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

do you see something wrong?

TheodoreT, 2015-12-16:
But I was wondering why the Protonect.exe compiled with the library runs without a problem and the customized project from visual studio or qtcreator does not. If it is a usb problem as you say I should had problem in both cases, shouldn't I?

xlz, 2015-12-16:
I thought you reported

> it seems that I am getting the above error also when I run the Protonect.exe example that it has been built with the libfreenect2 library

?

OK, it seems it is not a usb problem. "failure opening device" happened because the default serial number was different than expected. It is still something wrong with std::string. 

I think it is this http://stackoverflow.com/questions/30982538/stdstring-size-changes-between-dll-s-and-access-violation-cause-of-it

TheodoreT, 2015-12-16:
Hi @xlz thanks for the feedback and the link. Just for verification in the beginning I was getting the following error:
``` c++
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
```
only in the customized project. But afterwards in both cases but I guess it might had to do with the problem explained in the link you pointed me out.

Now regarding the actual issue changing from debug mode to release mode in my custom visual studio project seems to solve the problem. However, does that mean that I cannot compile my custom projects in debug mode? Actually if I understood correctly this is happening because libfreenect2 is compiled in release mode right? Will I need to recompile libfreenect2 in debug mode?

xlz, 2015-12-16:
You were using Release mode qt libraries with Debug mode libfreenect2 and that was the problem. If you want to use Debug mode lilbfreenect2, you will need Debug mode qt libraries.

TheodoreT, 2015-12-17:
Hi @xlz thanks for the feedback. Well, problem seems to be solved. Thanks again for the help.

closed 2015-12-17 by TheodoreT