OpenKinect/libfreenect2 #505: Compiling libusb with MinGW?

gppk, 2015-12-15:
I have worked my way through the entire install and compilation procedure for libfreenect and have got to the point where i have successfully compiled the Protonect application. Unfortunately it gives me the
  
    LIB_USB_ERROR_NOT_SUPPORTED 

errors that have been reported by others in other issues.

I believe this is because I have no managed to compile libusb correctly but quite frankly I seem to be struggling. The install cmd supplied is for visual studio and there are no cmake options for libusb. I have not managed to find anything online that was suitably helpful.

So my question is, has anyone been through the procedure for compiling for MinGW successfully and what steps am I missing in order to succeed?

**Some other information of note:**
The protonect application Viewer is unresponsive however in the console it shows:

    [Info] [TurboJpegRgbPacketProcessor] avg. time: 19.2634ms -> ~51.912Hz
    [Info] [TurboJpegRgbPacketProcessor] avg. time: 19.224ms -> ~52.0183Hz

Does this mean that it is indeed running? Or is this something else.

>Computer specs: Win7, USB3, MinGW

Thanks.

xlz, 2015-12-16:
That's right. MinGW has not been supported and is never build tested, because no MinGW user has ever spoken here. 

> LIB_USB_ERROR_NOT_SUPPORTED 

libusbk iso transfer is not built in libusb.

> The install cmd supplied is for visual studio

You can take a look inside and the commands should be easy to apply to mingw. Just ./configure and make?

> Does this mean that it is indeed running?

Running yes, receiving color images but not depth images.

gppk, 2015-12-16:
That's great! Sounds like it's worth persevering for a little while longer to try and build LibUSB correctly.

It's good to know that I am receiving colour images, just need to get access to those pesky depth images now.

In answer to your question 

> just ./configure and make

I believe I need to do something similar to that however those commands aren't available on windows so I need to work out what I need to do instead, I have a feeling it's something to do with MSYS but Im not sure.

Thanks for your response, do you want me to report back if I succeed and get it up and running with MinGW?

xlz, 2015-12-16:
I think MinGW is quite confusing. There are two MinGW's, and several software distributions, including MSYS(2?). I have no idea which one is the latest or well supported, or which one to support. I only remember last time I set up MSYS on Windows it was quite cumbersome. If I were to do this now, I will only consider MXE.

Anyway, MinGW counts as a entirely different platform, so right now there is no documentation on how to build this with it. If you figure it out and get it working, you can submit a PR to update CMake files (a lot of path heuristics about MSVC, none about MinGW) and add a section to README.

libusb would require autotools to build.