OpenKinect/libfreenect2 #506: cannot edit  /sys/module/i915/parameters/enable_cmd_parser file

chevalfoucher, 2015-12-16:
I've tried in many ways to edit this file in a graphic terminal :
sudo su
echo 0 > /sys/module/i915/parameters/enable_cmd_parser
-> bash: /sys/module/i915/parameters/enable_cmd_parser: No such file or directory

or:
sudo vim /sys/module/i915/parameters/enable_cmd_parser
... can t open file for writing
Il never works? Any one knows why?

xlz, 2015-12-16:
Are you using Beignet? Is your hardware Intel CPU? What is the kernel version?

chevalfoucher, 2015-12-16:
Yes, I installed beignet Dev. My hardware is a Intel CPU as well as Intel GPU (intel HD graphics 3000). The version of my linux kernel is 3.13.0-71-generic.
My computer recognises the kinect, but I just see a black screnn when a run Protonect. I have the following log [Debug] [DepthPacketStreamParser] skipping depth packet

xlz, 2015-12-16:
Try at least 3.16 kernel.

chevalfoucher, 2015-12-17:
I updated my kernel to version 4.2 . Now i can set enable_cmd_parser to zero, but I still have 
[Debug] [DepthPacketStreamParser] skipping depth packet

xlz, 2015-12-17:
skipping depth packet is not an error. It just tells you your gpu is too slow to catch up with the data source.

floe, 2015-12-20:
Do you see any output if you run `./Protonect cpu`?

closed 2016-01-27 by floe