OpenKinect/libfreenect2 #508: Access various projections and key matrices

ahundt, 2015-12-17:
There are a few different functions that would be nice to have.

- [color_to_depth](https://github.com/OpenKinect/libfreenect2/blob/master/src/registration.cpp#L85)
- Various transformation matrices
    - IR image x,y to Color x,y and back
    - IR cloud xyz  to color x,y and back
- Getting the intrinsic matrix
    - I know there is Freenect2Device::ColorCameraParams, but for some reason once I put it into a matrix and called `cv::decomposeHomographyMat(H, K, rotations, translations, normals);` from OpenCV, none of the solutions have been projecting correctly for me.
    - There is that interesting predefined constant... does that matter in that case?

xlz, 2015-12-17:
Previous discussion: https://github.com/OpenKinect/libfreenect2/issues/223 https://github.com/OpenKinect/libfreenect2/issues/41

There are some problem extracting these matrices losslessly from the builtin parameters.

ahundt, 2015-12-27:
Here is one example motivating use case that can be used to help determine what is actually needed:

 - Known object detection and pose estimation
 - narrow down the search space via the color image 
    - color_to_depth would be useful here
 - project point cloud models of objects into the frame of the kinect
   - Homography between two images ([example video](https://www.youtube.com/watch?v=WGUtv8ddyc4))
   - Projecting the object model into the frame ([video of broken first attempt that illustrates the idea well](https://www.youtube.com/watch?v=Z2Fg6GEhjpU))

@xlz those previous discussions are definitely relevant, and alternative ways to achieve this aside from the standard transformation matrices are definitely worth considering, particularly if they are more accurate.

Haven't thought about it yet but it is probably also worth determining if color_to_depth can be inverted in a reasonable manner.

floe, 2015-12-27:
My knowledge of multiple-view geometry is a bit rusty, but IIRC, for the special case of the Kinect (one color camera & one depth camera), the exact transformation always depends on the live depth values and therefore can't be expressed as a "classic" transformation matrix, right?

Did you already have a look at the `bigdepth` parameter to `Registration::apply`? (https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a8be3a548ff440e30f5e89a4b053e04fa) This may be useful to address at least some of your issues.

ahundt, 2015-12-27:
- #509, #419 are also potentially relevant issues

 - calibration libraries by  [arpg](https://github.com/arpg) might provide some interesting ideas as well.
    - [Calibu](https://github.com/arpg/Calibu) library
    - [Calibration docs](https://github.com/arpg/Documentation/tree/master/Calibration)
    - [their hardware abstraction](https://github.com/arpg/HAL) for libfreenect2 interface

ahundt, 2015-12-27:
@floe I'm thinking it through and from my understanding there shouldn't be too much of a difference, though I'm still relatively new to this. Both cameras have a field of view and some kind of distortion so unless there is something very different about the depth camera the equations shouldn't be too different, aside from using the depth value directly allowing finding the disparity to be skipped.

It looks like there are some potentially useful resources (papers, etc) in the following [google results](https://www.google.com/search?q=depth+color+intrinsics&oq=depth+color+intrinsics&aqs=chrome..69i57.8159j0j7&sourceid=chrome&es_sm=91&ie=UTF-8#q=time+of+flight+depth+color+intrinsics&search_plus_one=form).

Some of those are for the original Kinect, so of course structured light vs ToF differences apply where appropriate.

floe, 2015-12-27:
You're right, my previous post was slightly incorrect. From one _camera_ space to the other, you can indeed use undistortion + regular transformation matrix. However, to actually map one pixel to the corresponding one in the other image (i.e. from one _image_ space to the other), you need the live depth values.

ahundt, 2015-12-27:
Sorry, my edits made this confusing. For future readers, I originally made a note of the similarity between camera field of views in stereo vs depth + rgb cameras. I then realized there is more complexity to this than the stereo case and edited my post.

xlz, 2015-12-27:
Let me describe the current status:

* The intrinsic parameters of the depth camera are known from the factory builtin parameters.
* The transformation of a 3-D pixel from depth to color camera is known from a builtin formula (`kinect2_depth_to_color`), but it is not in a matrix multiplication form.
* The extrinsic matrix between the depth and color camera is not known.
* The intrinsic parameters of the color camera are not known.

I gave it a go https://github.com/OpenKinect/libfreenect2/issues/223#issuecomment-120308123  trying to figure the extrinsic matrix from `kinect2_depth_to_color` but did not succeed. You can always look at it again.

To use matrix for extrinsic transformation requires either you supply the matrix by external means, or you figure out the extrinsic matrix from `kinect2_depth_to_color`.

floe, 2015-12-27:
> To use matrix for extrinsic transformation requires either you supply the matrix by external means, or you figure out the extrinsic matrix from `kinect2_depth_to_color`.

IIRC `kinect2_depth_to_color` also subsumes the intrinsic parameters of the color camera?

xlz, 2015-12-27:
I think yes, in #223 there is a plot showing that there is some (tiny) distortion produced by `kinect2_depth_to_color`.

xlz, 2015-12-27:
By the way, I found one old proposal:

> I'd say using numeric methods to solve kinect2_color_to_depth() would be sufficient.