OpenKinect/libfreenect2 #509: bigdepth image side black borders

TheodoreT, 2015-12-18:
Hi I was playing a bit with the library and the registration part. I managed successfully to obtain the rgb to dept registration and depth to color registration following @floe advises at #337 and #464. In the big depth image though there are this black borders on the sides of the image as it can be seen below:

[image]

Therefore, I was wondering if it is possible through the library to get the coordinates of the points shown in the bigDepth image or the width in order to extract easier the corresponding ROI from the rgb image?

Otherwise, I can try to do it manually through opencv.

xlz, 2015-12-18:
rgb (1920x1080) is wider than depth (512x424). There is no depth data there.

TheodoreT, 2015-12-18:
Hi @xlz, I know that, as I am saying I was wondering if somehow I can internally take the position of Point1 and Point2 when in registration->apply() function you are doing the remapping. I tried to go through your code but I did not really got that much. How is the scaling done? how is specified from where the depth data start and end (maybe here is where I can get the needed info) in the bigdepth image? Thanks.

floe, 2015-12-20:
The field-of-view of the cameras is different, so there will inevitably be some areas where they don't overlap. Also, there is no direct linear mapping between the two images, as the mapping always depends on the live depth values.

TheodoreT, 2015-12-20:
Hi @floe, I see. Thanks for the feedback. I will try to see if I can do something manually with opencv then.

TheodoreT, 2015-12-23:
something else why in the viewer the depth and bigdepth image acquired from registration are appearing different (see below)?
[image]

with opencv though:
```
cv::Mat(depth->height, depth->width, CV_32FC1, depth->data).copyTo(depthMat);
cv::imshow("depth", depthMat / 4500.0f);
cv::Mat(registeredInv.height, registeredInv.width, CV_32FC1, registeredInv.data).copyTo(registeredInvMat);
cv::imshow("registeredInv", registeredInvMat / 4500.0f);
```
I am getting the same image see below:
[image]

moreover, what is the `int *color_depth_map` variable in the `registration->apply()` function.

Thanks.

xlz, 2015-12-24:
Images are stretched to fit in the viewer.

xlz, 2015-12-24:
color_depth_map -> https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a8be3a548ff440e30f5e89a4b053e04fa

TheodoreT, 2015-12-27:
Hi @xlz thanks for the feedback can you explain a bit more what you mean that the 

>images are stretched to fit in the viewer

trying to show the corresponding image with opencv following the code below:

```
cv::Mat(depth->height, depth->width, CV_32FC1, depth->data).copyTo(depthMat);
cv::Mat(bigdepth.height, bigdepth.width, CV_32FC1, bigdepth.data).copyTo(bigdepthMat);
cv::imshow("depth", depthMat / 4500.0f);
cv::imshow("registeredInv", bigdepthMat / 4500.0f);
```

I am getting  the following result which seems quite what I want:

[image]

However, if I try to save it in the hard disk
```
Mat registeredInv;
bigdepthMat.convertTo(registeredInv, CV_16UC1);
imwrite(filepath + "registeredInv/" + filename +".png", registeredInv);
```
and then try to load it again with the following code:

```
Mat img = imread("../depth.png", CV_LOAD_IMAGE_ANYDEPTH|CV_LOAD_IMAGE_ANYCOLOR);
Mat img2 = imread("../bigdepth.png", CV_LOAD_IMAGE_ANYDEPTH|CV_LOAD_IMAGE_ANYCOLOR);

img.convertTo( img, CV_32FC1);
img2.convertTo( img2, CV_32FC1);

imshow("src", img / 4500.0f);
imshow("src2", img2 / 4500.0f);
```
I am getting the same result as in viewer, i.e. depth image appears with black background while bigdepth appears with white background. So, I am trying to understand what is happening. If I save both images as 32 bit float numbers as .exr files no problem both images afterwards appear with black background as is shown in the image above. However, I would like to avoid .exr files because of the size. Any idea, why is that happening.

xlz, 2015-12-27:
What is your problem again? If you have no problem with the different aspect ratio, the missing data gets visualized as white or black is but an artifact of specific visualizers.

> Depth 	512x424 float, unit: millimeter. Non-positive, NaN, and infinity are invalid or missing data.

https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#a1d1cfd8ffb84e947f82999c682b666a7

TheodoreT, 2015-12-28:
@xlz this answers my question thanks
>the missing data gets visualized as white or black is but an artifact of specific visualizers

xlz, 2016-01-21:
Resolved.

closed 2016-01-21 by xlz

navyasri1, 2016-12-13:
Hello,

I am trying to get color to depth mapping using bigdepth parameter.But I cannot read the values stored in bigdepth. Can you help me resolve this?

my code for declaration,and viewing are as follows

```
libfreenect2::Frame bigDepth(1920,1082,4);
registration->apply(rgb, depth, &undistorted, &registered,&bigDepth);
cv::Mat   bigDepthMat;
cv::Mat(bigDepth.height, bigDepth.width, CV_32FC1, bigDepth.data).copyTo(bigDepthMat);
cv::imshow("registeredInv", bigDepthMat / 4500.0f);

```

sairamkolla, 2016-12-13:
Hi, I think you forgot to pass the `enable_filter` parameter.