OpenKinect/libfreenect2 #511: No response after ./bin/Protonect running

zaidiawan, 2015-12-21:
Hi,
I have been using Protonect before succefully on my lenovo Laptop with i7 processor and AMD Radeon R5 240 GPU. I have to install Protonect again. make was successful without any errors. But when I run Protonect, nothing happens. and nothing is shown up. I re-installed my ubuntu 14.04, the same situation!. What could be the possible problem. After running Protonect not only nothing shows up but also when I shutdown the system, the system can't shutdown. It seems something bad happens inside there. I also used the cpu, gpu and cl options separately. nothing changes.  Have you made any changes to libfreenect2?

Regards,
Zahid

floe, 2015-12-22:
What is the console output from `Protonect`?

xlz, 2016-01-27:
If you have more information, please comment and reopen.

closed 2016-01-27 by xlz