OpenKinect/libfreenect2 #512: Libfreenect2 on Ubuntu 14.04 Nvidia

RichardTrainer2, 2015-12-23:
Hi  

I have been trying to build protonec for 3 days now.  It appears that others others have had similar problems but there are so many differing forks and solutions, that its hard to follow.  Ive tried everything I can find.    

At the moment I get to the point where Cmake fails to find the openGL connection.  I suspect that Install GLFW3 is what has failed, but I am not sure.  

Has anyone made a definative script for installing libfreenect2 on a completely clean new install of Ubuntu?  That would help.  

Im new at Ubuntu so sorry if Im asking 
Best Wishes 
Richard

xlz, 2015-12-24:
The instructions in README is tested over and over again on completely clean new installs of Ubuntu 14.04. If it didn't work for you, you might not have followed the instructions.

RichardTrainer2, 2015-12-25:
Believe me I have read and followed the Readme file many times.  I suspect that the place where Im going wrong is where it says .......

"Install GLFW3, only if OpenGL 3.1 is supported (Odroid XU4 does not support it and you should use `cmake -DENABLE_OPENGL=OFF`). Try `sudo apt-get install libglfw3-dev`, if that is not available (e.g. Ubuntu trusty 14.04), you can download and install deb files from later Ubuntu releases:"   

Could you tell me exactly which file to use?  I have tried dozens and none of them seem to work.  

Best Wishes and happy holidays.

floe, 2015-12-25:
What happens if you run `sudo apt-get install libglfw3-dev`?
Did you run `depends/install_ubuntu.sh` beforehand?

xlz, 2015-12-25:
> you can download and install deb files from later Ubuntu releases:
```
cd libfreenect2/depends
sh install_ubuntu.sh
sudo dpkg -i libglfw3*_3.0.4-1_*.deb
```

It already tell you exactly which file to use.

ahundt, 2015-12-27:
I've followed the instructions successfully in exactly this situation, in case the info helps.

RichardTrainer2, 2015-12-27:
Thanks for the help but Im still not there.  I followed the instructions all the way down to where I cmake 

And I get this as a result .........

richard@richard-desktop:~/Desktop/libfreenect2/build$ sudo cmake ..
-- using tinythread as threading library
-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
-- RPATH set to 
-- Linking with these libraries: 
 LibUSB_LIBRARY-NOTFOUND
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 pthread
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Configurating examples
-- Using in-tree freenect2 target
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/richard/Desktop/libfreenect2

-- Configuring incomplete, errors occurred!
See also "/home/richard/Desktop/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/richard/Desktop/libfreenect2/build/CMakeFiles/CMakeError.log".
richard@richard-desktop:~/Desktop/libfreenect2/build$ 

I have installed LibUSB so I dont know why its not finding it.  Could anyone give me some ideas?
Best Wishes 
Richard

xlz, 2015-12-27:
Why are you using sudo cmake?

RichardTrainer2, 2015-12-28:
Hi  I tried your suggestions and I now get this.............

richard@richard-desktop:~/libfreenect2/build$ sudo apt-add-repository ppa:floe/libusb
 libusb-1.0.20 for Ubuntu 14.04
 More info: https://launchpad.net/~floe/+archive/ubuntu/libusb
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmpg_ap0ahd/secring.gpg' created
gpg: keyring `/tmp/tmpg_ap0ahd/pubring.gpg' created
gpg: requesting key D2817996 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmpg_ap0ahd/trustdb.gpg: trustdb created
gpg: key D2817996: public key "Launchpad libtisch" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
richard@richard-desktop:~/libfreenect2/build$ cmake ..
-- using tinythread as threading library
-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
-- RPATH set to 
-- Linking with these libraries: 
 LibUSB_LIBRARY-NOTFOUND
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 pthread
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Configurating examples
-- Using in-tree freenect2 target
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/richard/libfreenect2

-- Configuring incomplete, errors occurred!
See also "/home/richard/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/richard/libfreenect2/build/CMakeFiles/CMakeError.log".
richard@richard-desktop:~/libfreenect2/build$ 

It doesnt seem to be finding it,  even right after I install it.

floe, 2015-12-28:
OK, something missing in the instructions: run `sudo apt-get update && sudo apt-get upgrade` after `apt-add-repository`, then libusb should be updated to 1.0.20.

floe, 2015-12-28:
Updated the README.

RichardTrainer2, 2015-12-29:
Thanks Florian   I will try again tonight.  Best Wishes Richard

RichardTrainer2, 2015-12-30:
Thanks Everyone  That worked!!!   Im up and running.  
Happy New year    Richard

closed 2015-12-30 by RichardTrainer2