OpenKinect/libfreenect2 #513: Nvidia TX1/TK1 - USB 3.0 Card with 2 Kinects. Will this work?

soulslicer, 2015-12-25:
I'd like to know if it is possible to interface two kinects via a USB 3.0 card on the Nvidia TX1

floe, 2015-12-25:
This is very difficult to answer. Can you find out how the USB3 controller on the TX1 is connected internally, i.e. how much PCI-X bandwidth is available?

xlz, 2015-12-25:
NVIDIA Tegra X1 Datasheet and NVIDIA Jetson TX1 Developer Kit Carrier say there is a PCIe2 x4 interface for additional USB cards. That is 16Gbps bandwidth. It can probably take 4 Kinects. The bottleneck is the decoding though. TurboJPEG decoder definitely cannot handle 2 Kinects. The Tegra hardware JPEG decoder probably is not designed for parallel decoding.

aurelw, 2016-01-29:
With the current jetson hacks implementation on a TK1 board I have to clock the GPU to full performance to be able to decode all frames from a single kinect stream. This only includes depth decoding, the jpegs are streamed uncompressed to a different host. As long as you use the same basic setup I doubt that encoding two kinects will be possible.

TX1 may have enough performance to handle two streams with a more performant implementation, if there is one available.

xlz, 2016-03-23:
TX1 hass near 200Hz jpeg decoder. If USB controller is not an issue, I'd say it may work.

closed 2016-03-23 by xlz