OpenKinect/libfreenect2 #514: LIBUSB_ERROR_NOTSUPPORTED operation not supported unimplemented on this platform

jiaxing-li, 2015-12-25:
when i run protonect.exe, nothing happend, so,i change the protonect code, i successfully show GRB stream, but when i want to show IR and depth image is error.

floe, 2015-12-25:
This bug report is completely useless, we need more details than that. What did you change, what error are you getting, what platform are you on?

xlz, 2015-12-25:
Protonect.exe, so you were using Windows? LIBUSB_ERROR_NOTSUPPORTED, your libusb is not built with winiso patch. How did you build libusb?

jiaxing-li, 2015-12-28:
i'm using win8.1, i use the file "install_libusb_vs2013.cmd" to built libusb. I change the LIBUSB_DEBUG=4 ,and get the error "unsupported API call for 'submit_ios_transfer'<unrecodnized device driver>".

jiaxing-li, 2015-12-28:
i'm using win8.1, i use the file "install_libusb_vs2013.cmd" to built libusb. I change the LIBUSB_DEBUG=4 ,and get the error "unsupported API call for 'submit_ios_transfer'".

I debug libusb and find the error.
the rgb_transfer_pool_ use the USB bulk transfer, but ir_transfer_pool_ use the isochronous transfer.

the libusb is not support the isochronous transfer.  

const struct windows_usb_api_backend usb_api_backend[USB_API_MAX] = {
	{
		USB_API_UNSUPPORTED,
		"Unsupported API",
		NULL,
		0,
		unsupported_init,
		unsupported_exit,
		unsupported_open,
		unsupported_close,
		unsupported_configure_endpoints,
		unsupported_claim_interface,
		unsupported_set_interface_altsetting,
		unsupported_release_interface,
		unsupported_clear_halt,
		unsupported_reset_device,
		unsupported_submit_bulk_transfer,
		unsupported_submit_iso_transfer,
		unsupported_submit_control_transfer,
		unsupported_abort_control,
		unsupported_abort_transfers,
		unsupported_copy_transfer_data,
	}, {
		USB_API_HUB,
		"HUB API",
		hub_driver_names,
		ARRAYSIZE(hub_driver_names),
		unsupported_init,
		unsupported_exit,
		unsupported_open,
		unsupported_close,
		unsupported_configure_endpoints,
		unsupported_claim_interface,
		unsupported_set_interface_altsetting,
		unsupported_release_interface,
		unsupported_clear_halt,
		unsupported_reset_device,
		unsupported_submit_bulk_transfer,
		unsupported_submit_iso_transfer,
		unsupported_submit_control_transfer,
		unsupported_abort_control,
		unsupported_abort_transfers,
		unsupported_copy_transfer_data,
	}, {
		USB_API_COMPOSITE,
		"Composite API",
		composite_driver_names,
		ARRAYSIZE(composite_driver_names),
		composite_init,
		composite_exit,
		composite_open,
		composite_close,
		common_configure_endpoints,
		composite_claim_interface,
		composite_set_interface_altsetting,
		composite_release_interface,
		composite_clear_halt,
		composite_reset_device,
		composite_submit_bulk_transfer,
		composite_submit_iso_transfer,
		composite_submit_control_transfer,
		composite_abort_control,
		composite_abort_transfers,
		composite_copy_transfer_data,
	}, {
		USB_API_WINUSBX,
		"WinUSB-like APIs",
		winusbx_driver_names,
		ARRAYSIZE(winusbx_driver_names),
		winusbx_init,
		winusbx_exit,
		winusbx_open,
		winusbx_close,
		winusbx_configure_endpoints,
		winusbx_claim_interface,
		winusbx_set_interface_altsetting,
		winusbx_release_interface,
		winusbx_clear_halt,
		winusbx_reset_device,
		winusbx_submit_bulk_transfer,
		unsupported_submit_iso_transfer,
		winusbx_submit_control_transfer,
		winusbx_abort_control,
		winusbx_abort_transfers,
		winusbx_copy_transfer_data,
	}, {
		USB_API_HID,
		"HID API",
		hid_driver_names,
		ARRAYSIZE(hid_driver_names),
		hid_init,
		hid_exit,
		hid_open,
		hid_close,
		common_configure_endpoints,
		hid_claim_interface,
		hid_set_interface_altsetting,
		hid_release_interface,
		hid_clear_halt,
		hid_reset_device,
		hid_submit_bulk_transfer,
		unsupported_submit_iso_transfer,
		hid_submit_control_transfer,
		hid_abort_transfers,
		hid_abort_transfers,
		hid_copy_transfer_data,
	},

jiaxing-li, 2015-12-28:
@floe  @xlz 
i'm using win8.1, i use the file "install_libusb_vs2013.cmd" to built libusb. I change the LIBUSB_DEBUG=4 ,and get the error "unsupported API call for 'submit_ios_transfer'".

I debug libusb and find the error.
the rgb_transfer_pool_ use the USB bulk transfer, but ir_transfer_pool_ use the isochronous transfer.

the libusb is not support the isochronous transfer.

const struct windows_usb_api_backend usb_api_backend[USB_API_MAX] = {
{
USB_API_UNSUPPORTED,
"Unsupported API",
NULL,
0,
unsupported_init,
unsupported_exit,
unsupported_open,
unsupported_close,
unsupported_configure_endpoints,
unsupported_claim_interface,
unsupported_set_interface_altsetting,
unsupported_release_interface,
unsupported_clear_halt,
unsupported_reset_device,
unsupported_submit_bulk_transfer,
unsupported_submit_iso_transfer,
unsupported_submit_control_transfer,
unsupported_abort_control,
unsupported_abort_transfers,
unsupported_copy_transfer_data,
}, {
USB_API_HUB,
"HUB API",
hub_driver_names,
ARRAYSIZE(hub_driver_names),
unsupported_init,
unsupported_exit,
unsupported_open,
unsupported_close,
unsupported_configure_endpoints,
unsupported_claim_interface,
unsupported_set_interface_altsetting,
unsupported_release_interface,
unsupported_clear_halt,
unsupported_reset_device,
unsupported_submit_bulk_transfer,
unsupported_submit_iso_transfer,
unsupported_submit_control_transfer,
unsupported_abort_control,
unsupported_abort_transfers,
unsupported_copy_transfer_data,
}, {
USB_API_COMPOSITE,
"Composite API",
composite_driver_names,
ARRAYSIZE(composite_driver_names),
composite_init,
composite_exit,
composite_open,
composite_close,
common_configure_endpoints,
composite_claim_interface,
composite_set_interface_altsetting,
composite_release_interface,
composite_clear_halt,
composite_reset_device,
composite_submit_bulk_transfer,
composite_submit_iso_transfer,
composite_submit_control_transfer,
composite_abort_control,
composite_abort_transfers,
composite_copy_transfer_data,
}, {
USB_API_WINUSBX,
"WinUSB-like APIs",
winusbx_driver_names,
ARRAYSIZE(winusbx_driver_names),
winusbx_init,
winusbx_exit,
winusbx_open,
winusbx_close,
winusbx_configure_endpoints,
winusbx_claim_interface,
winusbx_set_interface_altsetting,
winusbx_release_interface,
winusbx_clear_halt,
winusbx_reset_device,
winusbx_submit_bulk_transfer,
unsupported_submit_iso_transfer,
winusbx_submit_control_transfer,
winusbx_abort_control,
winusbx_abort_transfers,
winusbx_copy_transfer_data,
}, {
USB_API_HID,
"HID API",
hid_driver_names,
ARRAYSIZE(hid_driver_names),
hid_init,
hid_exit,
hid_open,
hid_close,
common_configure_endpoints,
hid_claim_interface,
hid_set_interface_altsetting,
hid_release_interface,
hid_clear_halt,
hid_reset_device,
hid_submit_bulk_transfer,
unsupported_submit_iso_transfer,
hid_submit_control_transfer,
hid_abort_transfers,
hid_abort_transfers,
hid_copy_transfer_data,
},

xlz, 2015-12-28:
Can you try this binary? https://github.com/xlz/libusb/releases/tag/v1.0.20-winiso

jiaxing-li, 2015-12-28:
@xlz Thanks a lot. this libusb is ok. the error is disappeared. thanks again.

floe, 2015-12-28:
@xlz maybe worth linking to that from the README?

closed 2015-12-28 by floe

xlz, 2015-12-28:
I think users should not trust the binary I provide here. Next time if the user is unable to debug why it does not build libusb correctly, I'll point them to the binary.

floe, 2015-12-28:
OK, agreed.

xlz, 2016-01-04:
By the way I think the whole thing will be non-issue once we have binary release for Windows. Discuss in #244