OpenKinect/libfreenect2 #515: point cloud support and/or helpers

ahundt, 2015-12-27:
Possible forms point cloud support could come in:
 - no dependency, but add helper functions that will assist any library with point clouds
    - examples include [opencv_contrib](https://github.com/Itseez/opencv_contrib), [libpointmatcher](https://github.com/ethz-asl/libpointmatcher), [boost.geometry](boost.org/libs/geometry)
 - providing an optional dependency and conversion header for desired libraries

There are a few disparate places where this is being worked on with PCL specifically:
[pcl pull request](https://github.com/PointCloudLibrary/pcl/pull/1410)
[libfreenect2pclgrabber](https://github.com/giacomodabisias/libfreenect2pclgrabber)
[my libfreenect2pclgrabber fork](https://github.com/ahundt/libfreenect2pclgrabber)

floe, 2015-12-27:
We already have a lot of issues with the absolutely unavoidable dependencies (OpenGL, OpenCL, libusb), so I'd avoid going that route. Regarding helper functions, there is `Registration::getPointXYZRGB` (http://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a145c5d0d475a688f4c36ec35b1ac4974), is that what you have in mind?

giacomodabisias, 2015-12-27:
Yeah,
I would also avoid that for now and maintain things separate

xlz, 2015-12-27:
I agree we can add more helpers or variants of Registration::getPointXYZRGB to make it easier to export to PCL.

Also, I plan for a data recorder which saves data in PCLZF format and can be imported to PCL with its own interfaces/tools.

None of these need extra dependency.

The pclgrabber is interesting. I think the purpose of that is mostly realtime pointcloud visualization. That is best done on the PCL side.

hanyazou, 2016-01-03:
JFYI:
I wrote my OpeNI2 driver to use libfreenect2 with PCL. (the pull request is https://github.com/OpenKinect/libfreenect2/pull/518) You can grab point could data from Kinect to PCL because PCL have OpenNI2 grabber already. Pcl_openni2_viewer application in the PCL is a sample realtime point cloud viewer.