OpenKinect/libfreenect2 #517: The device disappear after calling Freenect2DeviceImpl::stop()

hanyazou, 2015-12-30:
I noticed that the device "Xbox NUI Sensor" disappear after a process calling Freenect2DeviceImpl::stop(). I couldn't open the device even if I invoke new process. To recover that, I  must unplug the AC adaptor and the USB cable from the Kinect V2 adaptor to reset that. I must do unplug and plug every time because recent Protonect handles ^C signal to call the stop().

My environment is Kinect sensor which have product ID == 0x02C4 (libfreenect2.hpp says that is "preview"), Mac OS X 10.11.2. and to day's master branch of libfreenect2.

I'm curious to know whether your device show similar behavior because I have only one Kinect and only one Mac.

Someone has same problem? or it's just that my hardware has a problem?

hanyazou, 2015-12-30:
Here is a copy of my terminal output.

bash-3.2$ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 16>
  +-o Root Hub Simulation Simulation@1d000000  <class AppleUSBRootHubDevice, id 0x10000030e, registered, matched, active, busy 0 (138 ms), retain 135>
  | +-o IOUSBHostDevice@1d100000  <class AppleUSBDevice, id 0x10000030f, registered, matched, active, busy 0 (158 ms), retain 140>
  |   +-o IOUSBHostDevice@1d180000  <class AppleUSBDevice, id 0x100000334, registered, matched, active, busy 0 (143 ms), retain 140>
  |     +-o BRCM20702 Hub@1d181000  <class AppleUSBDevice, id 0x10000034e, registered, matched, active, busy 0 (153 ms), retain 396>
  |       +-o Bluetooth USB Host Controller@1d181300  <class AppleUSBDevice, id 0x100000376, registered, matched, active, busy 0 (3345 ms), retain 274>
  +-o Root Hub Simulation Simulation@1a000000  <class AppleUSBRootHubDevice, id 0x100000314, registered, matched, active, busy 0 (95 ms), retain 135>
  | +-o IOUSBHostDevice@1a100000  <class AppleUSBDevice, id 0x100000315, registered, matched, active, busy 0 (141 ms), retain 140>
  |   +-o FaceTime HD Camera (Built-in)@1a110000  <class AppleUSBDevice, id 0x100000830, registered, matched, active, busy 0 (233 ms), retain 277>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x10000062f, registered, matched, active, busy 0 (99 ms), retain 136>
    +-o NuiSensor Adaptor      @14700000  <class AppleUSBDevice, id 0x100006180, registered, matched, active, busy 0 (37 ms), retain 72>
    | +-o Xbox NUI Sensor@14710000  <class AppleUSBDevice, id 0x100009617, registered, matched, active, busy 0 (12 ms), retain 17>
    +-o NuiSensor Adaptor      @14300000  <class AppleUSBDevice, id 0x100006193, registered, matched, active, busy 0 (35 ms), retain 71>
bash-3.2$ ./build/bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:10 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz^@ (CPU)[Intel^@]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 1 packets were lost
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
bash-3.2$ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 16>
  +-o Root Hub Simulation Simulation@1d000000  <class AppleUSBRootHubDevice, id 0x10000030e, registered, matched, active, busy 0 (138 ms), retain 135>
  | +-o IOUSBHostDevice@1d100000  <class AppleUSBDevice, id 0x10000030f, registered, matched, active, busy 0 (158 ms), retain 140>
  |   +-o IOUSBHostDevice@1d180000  <class AppleUSBDevice, id 0x100000334, registered, matched, active, busy 0 (143 ms), retain 140>
  |     +-o BRCM20702 Hub@1d181000  <class AppleUSBDevice, id 0x10000034e, registered, matched, active, busy 0 (153 ms), retain 396>
  |       +-o Bluetooth USB Host Controller@1d181300  <class AppleUSBDevice, id 0x100000376, registered, matched, active, busy 0 (3345 ms), retain 274>
  +-o Root Hub Simulation Simulation@1a000000  <class AppleUSBRootHubDevice, id 0x100000314, registered, matched, active, busy 0 (95 ms), retain 135>
  | +-o IOUSBHostDevice@1a100000  <class AppleUSBDevice, id 0x100000315, registered, matched, active, busy 0 (141 ms), retain 140>
  |   +-o FaceTime HD Camera (Built-in)@1a110000  <class AppleUSBDevice, id 0x100000830, registered, matched, active, busy 0 (233 ms), retain 277>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x10000062f, registered, matched, active, busy 0 (99 ms), retain 136>
    +-o NuiSensor Adaptor      @14700000  <class AppleUSBDevice, id 0x100006180, registered, matched, active, busy 0 (37 ms), retain 71>
    +-o NuiSensor Adaptor      @14300000  <class AppleUSBDevice, id 0x100006193, registered, matched, active, busy 0 (35 ms), retain 71>
bash-3.2$ ./build/bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

xlz, 2015-12-30:
@RyanGordon didn't seem to have this behavior.

hanyazou, 2015-12-30:
I can reproduce this 100% every time I try this sequence with my device.

I've learned in the net unplugging AC adaptor to recognise Kinect v2 while I can't recall the URL. So, I think someone else have similar problem. It might be a problem which early lot of the device have.

xlz, 2015-12-30:
Run `./Protonect` and `pkill -USR1 Protonect` will call `stop()`. Ctrl-C will call `stop()` and `close()`. I expect `close()` to cause the device to disconnect  for a few seconds and reconnect.

What does it do if you `pkill -USR1 Protonect` only?

hanyazou, 2015-12-30:
It seems that the close() does not matter.

on terminal 1
========================================
bash-3.2$ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 16>
  +-o Root Hub Simulation Simulation@1d000000  <class AppleUSBRootHubDevice, id 0x10000030e, registered, matched, active, busy 0 (138 ms), retain 135>
  | +-o IOUSBHostDevice@1d100000  <class AppleUSBDevice, id 0x10000030f, registered, matched, active, busy 0 (159 ms), retain 140>
  |   +-o IOUSBHostDevice@1d180000  <class AppleUSBDevice, id 0x100000334, registered, matched, active, busy 0 (143 ms), retain 140>
  |     +-o BRCM20702 Hub@1d181000  <class AppleUSBDevice, id 0x10000034e, registered, matched, active, busy 0 (153 ms), retain 396>
  |       +-o Bluetooth USB Host Controller@1d181300  <class AppleUSBDevice, id 0x100000376, registered, matched, active, busy 0 (3345 ms), retain 274>
  +-o Root Hub Simulation Simulation@1a000000  <class AppleUSBRootHubDevice, id 0x100000314, registered, matched, active, busy 0 (95 ms), retain 135>
  | +-o IOUSBHostDevice@1a100000  <class AppleUSBDevice, id 0x100000315, registered, matched, active, busy 0 (141 ms), retain 140>
  |   +-o FaceTime HD Camera (Built-in)@1a110000  <class AppleUSBDevice, id 0x100000830, registered, matched, active, busy 0 (234 ms), retain 277>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x10000062f, registered, matched, active, busy 0 (99 ms), retain 137>
    +-o NuiSensor Adaptor      @14700000  <class AppleUSBDevice, id 0x100006180, registered, matched, active, busy 0 (38 ms), retain 72>
    | +-o Xbox NUI Sensor@14710000  <class AppleUSBDevice, id 0x100009af3, registered, matched, active, busy 0 (11 ms), retain 17>
    +-o NuiSensor Adaptor      @14300000  <class AppleUSBDevice, id 0x100006193, registered, matched, active, busy 0 (36 ms), retain 71>
    +-o Android@14200000  <class AppleUSBDevice, id 0x100009a1a, registered, matched, active, busy 0 (10 ms), retain 15>
bash-3.2$ ./build/bin/Protonect
./build/bin/Protonect
Environment variables: LOGFILE=<protonect.log>
Usage: ./build/bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:5 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz^@ (CPU)[Intel^@]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 1 packets were lost
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

on terminal 2
========================================
$ pkill -USR1 Protonect
$ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 16>
  +-o Root Hub Simulation Simulation@1d000000  <class AppleUSBRootHubDevice, id 0x10000030e, registered, matched, active, busy 0 (138 ms), retain 138>
  | +-o IOUSBHostDevice@1d100000  <class AppleUSBDevice, id 0x10000030f, registered, matched, active, busy 0 (159 ms), retain 143>
  |   +-o IOUSBHostDevice@1d180000  <class AppleUSBDevice, id 0x100000334, registered, matched, active, busy 0 (143 ms), retain 143>
  |     +-o BRCM20702 Hub@1d181000  <class AppleUSBDevice, id 0x10000034e, registered, matched, active, busy 0 (153 ms), retain 399>
  |       +-o Bluetooth USB Host Controller@1d181300  <class AppleUSBDevice, id 0x100000376, registered, matched, active, busy 0 (3345 ms), retain 277>
  +-o Root Hub Simulation Simulation@1a000000  <class AppleUSBRootHubDevice, id 0x100000314, registered, matched, active, busy 0 (95 ms), retain 135>
  | +-o IOUSBHostDevice@1a100000  <class AppleUSBDevice, id 0x100000315, registered, matched, active, busy 0 (141 ms), retain 143>
  |   +-o FaceTime HD Camera (Built-in)@1a110000  <class AppleUSBDevice, id 0x100000830, registered, matched, active, busy 0 (234 ms), retain 280>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x10000062f, registered, matched, active, busy 0 (99 ms), retain 137>
[... cut ...]
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [protocol::CommandTransaction] sent number of bytes differs from expected number! expected: 24 got: 0
[Warning] [Freenect2DeviceImpl] serial number reported by libusb 005697142447 differs from serial number  in device protocol!
Protonect(97313,0x7fff77b84000) malloc: *** error for object 0x7f9c927921e0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug

xlz, 2015-12-30:
Is this introduced in some recent commits? If it is, it's probably this commit https://github.com/OpenKinect/libfreenect2/commit/db37864b5d93355c7df78e294ca79aad8328baaa. Can you try bisecting around this commit?

hanyazou, 2015-12-31:
I tried 2db187e "Opencv for good on Aug 17, 2015". That seems to have same problem.
I couldn't try older version because of some compile error. I couldn't execute git bisect because I couldn't find any good version.

bash-3.2$ ../bin/Protonect  cl
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @20:4 with serial 005697142447
[Freenect2Impl] found 1 devices
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz^@ (CPU)[Intel^@]
  1: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[OpenCLDepthPacketProcessor::buildProgram] building OpenCL program...
[OpenCLDepthPacketProcessor::buildProgram] OpenCL program built successfully
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 0b 09 14 47 4d 53 33 30 35 2e 31    31332e2e2e2e2e2e474d533330352e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  1f 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.1.3912.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1020
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[OpenCLDepthPacketProcessor] avg. time: 7.47684ms -> ~133.746Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.4778ms -> ~64.6085Hz
  C-c C-c^C[OpenCLDepthPacketProcessor] avg. time: 5.6253ms -> ~177.768Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.7155ms -> ~63.6315Hz

[OpenCLDepthPacketProcessor] avg. time: 5.588ms -> ~178.955Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.7761ms -> ~63.387Hz
  C-c C-c^C

[OpenCLDepthPacketProcessor] avg. time: 5.47193ms -> ~182.751Hz
[TurboJpegRgbPacketProcessor] avg. time: 15.9086ms -> ~62.8591Hz
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[Freenect2DeviceImpl] stopped
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] already closed, doing nothing
bash-3.2$ bash-3.2$ ioreg -p IOUSB
ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 16>
  +-o Root Hub Simulation Simulation@1d000000  <class AppleUSBRootHubDevice, id 0x10000030e, registered, matched, active, busy 0 (144 ms), retain 135>
  | +-o IOUSBHostDevice@1d100000  <class AppleUSBDevice, id 0x10000030f, registered, matched, active, busy 0 (164 ms), retain 140>
  |   +-o IOUSBHostDevice@1d180000  <class AppleUSBDevice, id 0x100000334, registered, matched, active, busy 0 (148 ms), retain 140>
  |     +-o BRCM20702 Hub@1d181000  <class AppleUSBDevice, id 0x10000034e, registered, matched, active, busy 0 (159 ms), retain 396>
  |       +-o Bluetooth USB Host Controller@1d181300  <class AppleUSBDevice, id 0x100000376, registered, matched, active, busy 0 (3349 ms), retain 274>
  +-o Root Hub Simulation Simulation@1a000000  <class AppleUSBRootHubDevice, id 0x100000314, registered, matched, active, busy 0 (95 ms), retain 135>
  | +-o IOUSBHostDevice@1a100000  <class AppleUSBDevice, id 0x100000315, registered, matched, active, busy 0 (146 ms), retain 140>
  |   +-o FaceTime HD Camera (Built-in)@1a110000  <class AppleUSBDevice, id 0x100000830, registered, matched, active, busy 0 (244 ms), retain 277>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x10000062f, registered, matched, active, busy 0 (99 ms), retain 136>
    +-o NuiSensor Adaptor      @14300000  <class AppleUSBDevice, id 0x100009f3d, registered, matched, active, busy 0 (6 ms), retain 11>
    +-o NuiSensor Adaptor      @14700000  <class AppleUSBDevice, id 0x100009f51, registered, matched, active, busy 0 (4 ms), retain 11>
bash-3.2$ ../bin/Protonect  cl
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found 0 devices
no device connected!

xlz, 2015-12-31:
But you should have it working at one time in the past, or have you never got it properly restarted?

For example, this commit on Jul 13 by you https://github.com/OpenKinect/libfreenect2/commit/5c2392747dcd0fff5dcf473032768f5050f3bb45 should work? There are only a few commits between this and Aug 17.

bellekci, 2015-12-31:
I have the same problem in Ubuntu 15.04 with libusb 1.0.2. 

As a workaround, I reset xhci controller then I can restart.

hanyazou, 2016-01-01:
@xlz Sorry, I didn't care about the problem because I didn't alter protonect and libfreenect2 themselves.
I use to test my driver with NiViewer which comes from OpenNI2. I can stop NiViewer with ^C without the problem. NiViewer does not handle the signal and quit immediately without call stop() and close().

@bellekci when did you buy your kinect sensor and adaptor for windows? What are their model numbers?

here are info from mine:
sensor:
  model 1520 (purchased 2014/9)
  USB product id is 0x02C4 ("preview")
  protonect says "device firmware: 4.0.3912.0"
adaptor:
  model 1637 (purchased 2015/4)

xlz, 2016-01-01:
Which lines here https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L807-L842 effectively make the device disappear?

xlz, 2016-01-01:
@bellekci According to your last reports, your errors were LIBUSB_ERROR_NO_DEVICE because the host controller got busted from DMA errors after running for one hour. So you are reporting a new behavior on your hardware?

hanyazou, 2016-01-01:
@xlz if you call cancel(), the device will disappear.

  LOG_INFO << "disabling usb transfer submission...";
  rgb_transfer_pool_.disableSubmission();
  ir_transfer_pool_.disableSubmission();
  // exit(1);  // works well
  LOG_INFO << "canceling usb transfers...";
  rgb_transfer_pool_.cancel();
  ir_transfer_pool_.cancel();
  exit(1);  // the device will disappear

hanyazou, 2016-01-01:
I made a patch 71519cf73e42aa0ccf0740d942be07aa09584576 to fix this problem. It seems that I should learn how usb_control.cpp works. But this patch solve the problem.

Please look at the patch.

xlz, 2016-01-01:
I'm not sure nuking the parent hub is a good idea.

Can you try moving 
```
  rgb_transfer_pool_.cancel();
  ir_transfer_pool_.cancel();
```
to after `usb_control_.setVideoTransferFunctionState(UsbControl::Disabled);`? If not working, some other places?

hanyazou, 2016-01-03:
@xlz I tried. but it does not work. I want bus analyzer...

0. call exit(1) immediately after ir_transfer_pool_.disableSubmission()
  => the device won't disappear (success rate is greater than 90%)

1. move *_transfer_pool_.cancel() to after usb_contro_.xxx() in the stop().
  => the device will disappear (success rate is 0%)

2. just comment out whole lines in Freenect2DeviceImpl::stop() and close().
  => the device will disappear (0%)
  and the process exit abnormally with "Segmentation fault: 11" or "libc++abi.dylib: Pure virtual function called! Abort trap: 6"

3. comment out usb_event_loop_.stop() and libusb_exit(usb_context_) in ~Freenect2Impl() additionally to 2.
  => same as 2.

4. comment out clearDevices() and clearDeviceEnumeration() in ~Freenect2Impl() additionally to 3.
  => the device will disappear (0%) if the process exit abnormally like 2.
    the device will not disappear or will disappear randomly (success rate 70~80%?) if the process exit without the exception.

xlz, 2016-01-03:
Can you try this:

Add `sleep(10 seconds);` after `xxx.disableSubmission()` and comment out
```
  //rgb_transfer_pool_.cancel();
  //ir_transfer_pool_.cancel();
```
Tell us if the device disappears within the 10 seconds.

The purpose of this is to test if simply not submitting new transfers would make the device disappear. `xxx_pool_.cancel()` just calls some `libusb_cancel_transfer()` and waits for all transfers to expire.

If simply not submitting new transfers makes the device disappear, I tend to consider this a purely hardware problem, because it's entirely legitimate to not submit new transfers.

hanyazou, 2016-01-03:
@xlz do you mean something like this? The device seems to be alive for 10secs and disappear when the process exit. (And device does not disappear after exiting if you comment out the sleep(10).)

terminal 1

    bash-3.2$ git diff | cat
    diff --git a/src/libfreenect2.cpp b/src/libfreenect2.cpp
    index af2188c..01fdaed 100644
    --- a/src/libfreenect2.cpp
    +++ b/src/libfreenect2.cpp
    @@ -907,9 +907,13 @@ bool Freenect2DeviceImpl::stop()
       rgb_transfer_pool_.disableSubmission();
       ir_transfer_pool_.disableSubmission();
    
    -  LOG_INFO << "canceling usb transfers...";
    -  rgb_transfer_pool_.cancel();
    -  ir_transfer_pool_.cancel();
    +  //LOG_INFO << "canceling usb transfers...";
    +  //rgb_transfer_pool_.cancel();
    +  //ir_transfer_pool_.cancel();
    +  LOG_INFO << "sleep 10...";
    +  sleep(10);
    +  LOG_INFO << "exit";
    +  exit(1);
    
       usb_control_.setIrInterfaceState(UsbControl::Disabled);

terminal 2

    bash-3.2$ while sleep 1; do
    > ioreg -p IOUSB
    > done

xlz, 2016-01-03:
Is there any helpful information in `sudo dmesg` and `/var/log/system.log`?

hanyazou, 2016-01-03:
No

    1/3/16 17:31:40  // start Protonect
    1/3/16 17:31:41.000     kernel[0]       021117.332761 Xbox NUI Sensor@14710000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
    1/3/16 17:31:45  // Protonect open the window
    1/3/16 17:31:45.589     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:45.590     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:45.592     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:45.596     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:45.605     appleeventsd[62]        SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:50.631     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:31:50.631     launchservicesd[88]     SecTaskLoadEntitlements failed error=22
    1/3/16 17:32:01  // hit ESC key
    1/3/16 17:32:11  // the process exit

lines start with "//" are added by me.

hanyazou, 2016-01-03:
more lines (but no helpful information)

    LIBUSB_DEBUG=4 ./build/bin/Protonect cl > log.txt 2>&1

[log.txt](https://github.com/OpenKinect/libfreenect2/files/76536/log.txt)

xlz, 2016-01-03:
I don't have your laptop, so most of the exploratory work has to be done by yourself.

Your patch 71519cf "works", but it is heavy handed (resetting the parent hub 16 times? what about other devices under the hub?). I get that this is simulating power cycling the Kinect adaptors. This problem is not seen on other devices. And USB traffic from Microsoft SDK does not touch the Kinect daptors neither.

More information is needed on what happens when the device disappears. Maybe use USB Prober.

hanyazou, 2016-01-03:
Please understand that 'parent hub' in my patch is the 'Kinect adaptor for windows'. That hub has no devices under it other than the Kinect sensor. The resetParentHub() will be called only if there are adaptors have no child Kinect sensors connected. If the adaptor have a child sensor, my patch does not issue any USB traffic. Please try my patch with your Kinect which does not have the problem to find it does not have heavy side-effect.

xlz, 2016-01-03:
I don't have a Mac and I don't have this problem so there is no way for me to test your patch.

The official SDK does not perform this hub resetting routine, which proves this is not something that is required for normal operation.

Another Mac user reported it running well without this issue https://github.com/OpenKinect/libfreenect2/pull/494#issuecomment-163060168, so we can't even use the patch with additional platform checking.

There are two ways to solve this problem. 1. Use some USB debugging tools to find out the root cause; 2. Find out ways to identify affected hardware models, and limit your hub resetting code to those hardware only.

hanyazou, 2016-01-03:
I've bought Windows 8 and tried the SDK with bootcamp. It turned out:
 * The Kinect Configuration Verifier can turn on/off my sensor repeatedly. So this does not seems to a hardware trouble of my sensor and adaptor.
 * Xbox One (game console) can turn on/off my sensor too in menu 'All Settings' > 'Kinect & devices'.
 * Neither SDK nor Xbox One provide firmware update. Current version (4.0.3912.0:7) seems to be latest one. (Xbox One system was updated last month via network.)

> Find out ways to identify affected hardware models

What is the product ID and firmware version of your sensor? Mine is ProductIdPreview = 0x02C4. This was bundled with Xbox One at launch day. This might be differ from the sensor unit of Kinect for Windows v2 kit.

> Besides, the Kinect may be simply not plugged in to the adaptor, if the adaptor hub has no device connected. Have you considered this case?

Yah. That is the problem. If there are such adapters on the bus, this patch takes several tens of seconds to start.

xlz, 2016-01-03:
I meant the affected Mac hardware models (and kernel versions).

hanyazou, 2016-01-04:
@xlz ah, i see... But...

 I found the criminal person. While VMware Fusion virtual machine running in the same Mac, the device will disappear after stop() and close().
I'm using VMware Fusion Version 7.1.3 (3204469). I will upgrade to version 8.

xlz, 2016-01-08:
I think your patch is still interesting. Right now there is a known issue that if the Kinect 2 is plugged in during boot, it will not be detected. Only the adaptors are detected by the OS, and once you replug it in, it is detected. I think this might have something to do with the adaptor setup. But still, power cycling a device needs some good reasons.

hanyazou, 2016-01-09:
I think only "the Kinect may be simply not plugged in to the adaptor" and long timeout is a problem. Do you think there are any other problems?
About long timeout, less retry count or one-shot might be enough for the known issue. One-shot takes less than 1 sec. And we can write parallel power off/on routine which handle several adaptors at once.

xlz, 2016-01-29:
I tested your patch against the condition of "The sensor is sometimes not detected if plugged in during boot". It did not work.

For the problem of "The sensor is sometimes not detected if plugged in during boot" a new issue may be opened in the future.

closed 2016-01-29 by xlz

xlz, 2016-01-31:
Now saying the patch is bad. I tested hub port reset and power cycle first with https://github.com/mvp/uhubctl, and it didn't work, then I tested your patch and it didn't work either, which showed the hub port power cycle is not the solution to this.