OpenKinect/libfreenect2 #519: minor fix for mac OSX 10.8.5

robozo, 2016-01-01:
Protonect failed with GLFW error, "The requested client API version is unavailable".
Turns out that OSX 10.8.5 only supports OpenGL version 3.2, and GLFW asked for 3.3.

So in viewer.cpp and opengl_depth_packet_processor.cpp, changed
glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);  to
glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 2);
things now working.

Although sometimes tricky to get Kinect recognized, have to reboot or plug/unplug
as described in FAQ.  Protonect runs at 40 Hz or so, mid-2012 macbook pro.

xlz, 2016-01-01:
Can you make a pull request?

robozo, 2016-01-02:
Hi, am new to github, didn't know about pull requests etc.
Mostly just wanted to leave a note to save someone some time, in case they run across
the same problem.  Will "close" this now, and maybe do a pull request later
when I have some time to figure out how to do it...

closed 2016-01-02 by robozo