OpenKinect/libfreenect2 #524: Support for calibration/rectification

ahundt, 2016-01-05:
While the hardware provides methods for calibration, it seems the results achievable from calibrating by hand can be much better than the one provided by the hardware. For example note the images in [iai-kinect2](https://github.com/code-iai/iai_kinect2)'s readme.md showing how things improve with the calibration they provide. 

Libraries of interest for calibration:

 - [iai-kinect2](https://github.com/code-iai/iai_kinect2)
 - [arpg calibration](https://github.com/arpg/Documentation/tree/master/Calibration) 

This may not necessarily come in the form of a dependency. It could simply be support for some param file that applies alternative parameters to the driver and notes explaining a process to follow to get those alternative parameters.

floe, 2016-01-05:
Hm, I'm a bit divided about that one - libfreenect2 is mostly supposed to provide an interface to the hardware, everything else is better suited for other libraries. This would be far easier if the factory calibration would conform to regular intrinsic/extrinsic camera parameters, but as it is now, we'd need a separate implementation for external calibration data.

TheodoreT, 2016-01-06:
+1 @ahundt I also think that the option of external calibration will be a nice add on to the library.