OpenKinect/libfreenect2 #525: GLFW error 65543

ruidoquerido, 2016-01-05:
I have an error when I run ./Protonect

[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable.

My gpu is an Intel HD Graphics (MSI cubi) and my kinect is an v2

OpenGL Version string: 3.0 Mesa 10.3.2

 Thanks

floe, 2016-01-05:
Are you running MacOS X 10.8? Then you might want to try the change in #520 and see if that fixes things for you. If yes, then this is a duplicate of #519.

floe, 2016-01-05:
Oh, my bad, now I see you're using Mesa. 10.3 is far too old, you need at least 10.5.

ruidoquerido, 2016-01-07:
i'm running debian jessie.

xlz, 2016-01-08:
According to https://wiki.debian.org/Mesa, OpenGL 3.3 core profile is supported by mesa 10.3.2.

> OpenGL Version string: 3.0 Mesa 10.3.2

This information is incomplete. Please provide the full output of `glxinfo | grep OpenGL`.

anshulvj, 2016-01-11:
I am getting the same error
```
.
.
.
[Debug] [DepthPacketStreamParser] skipping depth packet
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
```

Doesn't matter if I use ```cl```,```cpu``` or  ```gl```...error always ends with ```GLFW error 65543 The requested client API version is unavailable```. With the ```cl, gl``` options I'm also getting "pipeline not found error" but I think I can take care of it.
With the ```cpu``` option, I think it does everything and then fails with the following error:
```
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 007836443847
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
GLFW error 65543 The requested client API version is unavailable
Failed to create opengl window.
```

I checked issue #'s 519, 520 and 525, however it does not solve my problem. I did the nvidia driver purge although that didn't change anything. My output of ```glxinfo | grep OpenGL``` is:

```
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
OpenGL version string: 3.0 Mesa 10.5.9
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
```

xlz, 2016-01-11:
You can use `-noviewer` option. The viewer also has OpenGL version requirement.

xlz, 2016-01-11:
By the way I don't think your OpenGL is set up properly. `llvmpipe` is software rendering, you should have hardware acceleration.

anshulvj, 2016-01-11:
I followed all the instructions at ```http://www.freedesktop.org/wiki/Software/Beignet/```
(well, and some instructions from ```http://arrayfire.com/opencl-on-intel-hd-iris-graphics-on-linux/``` when the former did not work). The instructions at freedesktop included installing llvm.

The problem is, ```clinfo``` gives error:
```
anshul@anshul-omegainfinity:~/libfreenect2/build$ clinfo
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
Device open failed, aborting...
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
open("/dev/dri/card1", O_RDWR) failed: No such file or directory
open("/dev/dri/card2", O_RDWR) failed: No such file or directory
open("/dev/dri/card3", O_RDWR) failed: No such file or directory
open("/dev/dri/card4", O_RDWR) failed: No such file or directory
open("/dev/dri/card5", O_RDWR) failed: No such file or directory
open("/dev/dri/card6", O_RDWR) failed: No such file or directory
open("/dev/dri/card7", O_RDWR) failed: No such file or directory
open("/dev/dri/card8", O_RDWR) failed: No such file or directory
open("/dev/dri/card9", O_RDWR) failed: No such file or directory
open("/dev/dri/card10", O_RDWR) failed: No such file or directory
open("/dev/dri/card11", O_RDWR) failed: No such file or directory
open("/dev/dri/card12", O_RDWR) failed: No such file or directory
open("/dev/dri/card13", O_RDWR) failed: No such file or directory
open("/dev/dri/card14", O_RDWR) failed: No such file or directory
open("/dev/dri/card15", O_RDWR) failed: No such file or directory
Device open failed, aborting...
X server found. dri2 connection failed! 
Trying to open directly...Device open failed
Inconsistency detected by ld.so: dl-close.c: 762: _dl_close: Assertion `map->l_init_called' failed!
```
and the ```utests_run``` also fails glamorously! Problem is, all these months I was working on small minicomputers; a couple of hours ago, I installed Ubuntu 14.04 on a powerful PC. Here are the details:
```
anshul@anshul-omegainfinity:~/libfreenect2/build$ lspci | grep VGA
01:00.0 VGA compatible controller: NVIDIA Corporation GM107 [GeForce GTX 750 Ti] (rev a2)
```
And It seems I'm using the noveau driver because I purged nvidia drivers because my desktop wasn't showing up:
```
anshul@anshul-omegainfinity:~/libfreenect2/build$ lshw -c video | grep driver
WARNING: you should run this program as super-user.
       configuration: driver=nouveau latency=0
```

I'm willing to overwrite everything as this is a fresh ubuntu installation. However, should I follow just the OpenCL instructions at [http://wiki.tiker.net/OpenCLHowTo](http://wiki.tiker.net/OpenCLHowTo), or any other resources you would like to recommend?

xlz, 2016-01-12:
`/dev/dri/card0 not authenticated` this says your dri devices are somehow not set up with proper permissions, or something. Look at my setup
```
xlz@zotac-zboxei750:~$ ls -l /dev/dri/card0 
crw-rw---- 1 root video 226, 0 Jan  8 23:35 /dev/dri/card0
xlz@zotac-zboxei750:~$ id
uid=1000(xlz) gid=1000(xlz) groups=1000(xlz),6(disk),7(lp),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),46(plugdev),108(netdev),113(kvm),114(wireshark)
```
Maybe the permissions are right but you are not in the `video` group.

I don't have a definite guide for Nvidia hardware. But I prefer using nvidia drivers only and blacklisting nouveau if possible.

Also, you do have an Intel GPU, and you can use it. Nvidia graphics should be irrelevant here unless it breaks other things.

xlz, 2016-01-12:
Btw I don't think you need to follow any instructions on http://www.freedesktop.org/wiki/Software/Beignet/. See our README.md, you can install beignet-dev package from ppa. The freedesktop site is just for troubleshooting. Building this from source is prone to errors.

anshulvj, 2016-01-12:
I installed beignet as given on your readme, and reinstalled libfreenect2. Now when I run Protonect, it gives me the following (same) error:

```
anshul@anshul-omegainfinity:~/libfreenect2/build$ id
uid=1000(anshul) gid=1000(anshul) groups=1000(anshul),4(adm),24(cdrom),27(sudo),30(dip),44(video),46(plugdev),108(lpadmin),124(sambashare)

anshul@anshul-omegainfinity:~/libfreenect2/build$ ls -l /dev/dri/
total 0
crw-rw----+ 1 root video 226,   0 Jan 11 17:54 card0
crw-rw----  1 root video 226,  64 Jan 11 17:54 controlD64
crw-rw----  1 root video 226, 128 Jan 11 17:54 renderD128

anshul@anshul-omegainfinity:~/libfreenect2/build$ sudo ./bin/Protonect cl
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
X server found. dri2 connection failed! 
Protonect: /home/anshul/Downloads/beignet/src/intel/intel_driver.c:146: intel_driver_init: Assertion `res' failed.

anshul@anshul-omegainfinity:~/libfreenect2/build$ sudo ./bin/Protonect gl
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
X server found. dri2 connection failed! 
Protonect: /home/anshul/Downloads/beignet/src/intel/intel_driver.c:146: intel_driver_init: Assertion `res' failed.

anshul@anshul-omegainfinity:~/libfreenect2/build$ sudo ./bin/Protonect cpu
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
X server found. dri2 connection failed! 
Protonect: /home/anshul/Downloads/beignet/src/intel/intel_driver.c:146: intel_driver_init: Assertion `res' failed.
anshul@anshul-omegainfinity:~/libfreenect2/build$ 
```

xlz, 2016-01-12:
I think the same error happens with `clinfo`. I guess you have to research the setup for Intel/Nvidia dual graphics until `clinfo` works.

anshulvj, 2016-01-14:
I gave up trying to get it running on Ubuntu (after giving it one whole day). I switched instead, to Windows 8.1 on my PC. I spend like half a day or so trying to get the dependencies installed (Windows sucks in this regard!!). But finally I got the Protonect example running on Windows. Only thing I would like to point out is that I had to do libusb installation stuff manually; i.e. I couldn't just run the ```install_libusb_vs2013.cmd``` directly from the git bash shell. I had to make the following changes:
1) ```msbuild msvc\libusb_dll_2013.vcxproj /p:Platform=x64 /p:Configuration=%CONFIG% /target:Rebuild``` command won't run. It gives ```MSBUILD : error MSB1008: Only one project can be specified```, no matter how I change the switches/project names. This I had to manually compile with Visual Studio.
2) Also, the libusb-1.0.pdb does not get generated in Release config (only the .dll and .lib does). So I had to recompile with Debug config to place them in the X64/dll folder. Maybe the ```install_libusb_vs2013.cmd``` can be updated accordingly (?).

xlz, 2016-01-14:
I have released a libusb binary   https://github.com/OpenKinect/libfreenect2/issues/514#issuecomment-167481916.

Also, I have released a whole libfreenect2 binary https://github.com/OpenKinect/libfreenect2/issues/244#issuecomment-170185037. I think you can just use the binary release.

> MSBUILD : error MSB1008: Only one project can be specified

I have no idea about this one. Never seen it.

> libusb-1.0.pdb

Do you really need it? I never needed it. Setting environment variables LIBUSB_DEBUG dumps a lot of information.

anshulvj, 2016-01-14:
Thanks, I'll check them out.

Well. the .pdb requirement was given in the ```install_libusb_vs2013.cmd``` last time. I'm not sure if we need it if we can dump debug information as you said.

xlz, 2016-01-14:
Okay, maybe pdbs are not produced with vs2013? It has no effect overall though. Just ignore it.

xlz, 2016-01-14:
If you find the binary release working, please leave a comment to show the testing effort.

xlz, 2016-02-02:
This bug is mixed up with Intel, Intel/Nvidia, and Windows. If any of you still have the problem, open a new issue.

closed 2016-02-02 by xlz