OpenKinect/libfreenect2 #527: restart the kinect

zfxw0206, 2016-01-06:
I have a kinect, but sometimes, it may die. So I am thinking if I could write a code to detect it, if the kinect die, I will kill the kinect driver and restart it.

Does anybody have an idea about this?

xlz, 2016-01-08:
If it "dies", you should report an issue. But considering the style of this issue you are writing in, I recommend you perform complete collection of relevant information before submitting an issue.

Error reporting in the library will be improved in 0.2.

zfxw0206, 2016-01-11:
I am sorry, I may not state it clearly. The situation is my robot is using kinect to see something, but sometime the kinect may died. I can accept it died, but I want to write a code to detect kinect, if it died, the system will restart the kinect automatically.

In my mind, I will write a code to subscribe the topic, once the kinect died, I won't get the topic, then I will send "1" to a ".txt" file. I also write a ".bash" file, which is used to detect the ".txt" file, once I get "1", it will run "roslaunch kinect_bridge.launch" again.

Do you think this is doable and how could I make the ".bash" file run background?

xlz, 2016-01-11:
You can create a new thread in your program and in that thread set a timer of 1 second. In the data receiving thread, reset the timer each time a new frame is received. Once the timer times out, restart.

zfxw0206, 2016-01-14:
Thank you for your idea, but I have another question. How to restart kinect2_bridge.launch? Can we realize it by using cpp code?

xlz, 2016-01-14:
There are too many ways to restart kinect2_bridge.launch. You can.

Here is an example:
```bash
while true; do
  roslaunch kinect2_bridge.launch
done
```

zfxw0206, 2016-01-15:
This is a shell script, right?

xlz, 2016-01-15:
Yes

zfxw0206, 2016-01-20:
Ok, I did it. Thank you very much!

xlz, 2016-01-21:
Then this issue is considered resolved.

closed 2016-01-21 by xlz