OpenKinect/libfreenect2 #528: Saving images/depth data from kinnect 2 into .jpg file or similar

noobzone, 2016-01-06:
Hello. I'm an IT student from Portugal and I'm developing an application with kinnect 2 and libfreenect2 that should save (or insert images into a queue) to further analysis with an algorithm. The kinnect will have a top-down view of the room and the results of the analysis will be sent to a remote server with the inclusion of the serial number of the kinnect. If this seems strange to you but I was proposed with an internship to finish my degree and this was the project they gave me. The solution has to run in linux and be developed only on C++ as well. (new challenge as I only know C and C#)

Now the real question:

After searching all the files for some clue as to what to save the image on, I got nothing. I got the console to open and show me I get frames (as in the screenshot). Any sort of help on saving files would be wonderful.
[image]

Best regards.
Noobzone

floe, 2016-01-07:
libfreenect2 doesn't have such functionality built in, you have to implement it yourself.

noobzone, 2016-01-07:
Thanks for your reply. I managed to figure out what the line "libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];" does (as I said. no experience or knowledge in C++ ) and correct me if I'm wrong but "rgb -> Data" gives me all the bytes of a frame necessary to construct an image right? Also other attributes of the object give me the bytes per pixel (4- Alpha, Red, Green, Blue) as well as width and height. With that and the right library I'm capable of constructing an image from a frame using TimeStamp as a name correct? Once again thank you for your help 

Regards,
Noobzone

xlz, 2016-01-08:
You can find all the relevant information in the API reference openkinect.github.io/libfreenect2/.

> With that and the right library I'm capable of constructing an image from a frame using TimeStamp as a name correct

Yes.

closed 2016-01-18 by floe