OpenKinect/libfreenect2 #529: Installing at 13.04

wilianson, 2016-01-07:
Hi there, I need to install libfreenect2 at Ubuntu 13.04, because some restrictions about the hardware we are using, and when I am installing libusb, I need to use the version which is not available for my Ubuntu version, what Can I do?

floe, 2016-01-07:
I think you can directly download the .deb files from the PPA page on Launchpad and install them with dpkg. No guarantees, though.

de-vri-es, 2016-01-07:
You would be better off compiling the needed libusb from source. Installing packages from different distributions or versions of your distribution could end in tears.

xlz, 2016-01-08:
Ubuntu 13.04?

> Support for Ubuntu 13.04 officially ended on 27 January 2014

I think you are better off reinstalling Ubuntu 14.04.

floe, 2016-01-18:
If you really need to use that unsupported Ubuntu version, then you need to compile libusb-1.0.20 yourself as suggested by @de-vri-es. Closing.

closed 2016-01-18 by floe