OpenKinect/libfreenect2 #531: "Error" C4146 while running protonect

rmkemker, 2016-01-08:
I am going through protonect example with Visual Studio 2015 (Windows 8.1) and I am coming up with one error :

>c:\libfreenect2-master\build\install\include\libfreenect2\frame_listener.hpp(98): error C4146: unary minus operator applied to unsigned type, result still unsigned

I understand that C4146 is a Level 2 Warning; but even when I disable all warnings, this comes up and stops me from successfully building my project.  From debugging, it seems to come from trying to include the header files.  Any help would be appreciated.  Thank-you!

xlz, 2016-01-08:
Why is a warning becoming an error? How did you configure MSVC?

rmkemker, 2016-01-08:
After checking, I realized that warnings as errors was turned off, but SDL was turned on.  I didn't know what that was prior to Googling it, and I realized that was the setting that made it throw the C4146 as an error (vs a warning).  I disabled SDL and it builds correctly.  Sorry for the trouble...thank-you!

closed 2016-01-10 by floe